networkupstools/nut #355: Tripp-lite  - unhandled ASN 0x5 recieved

dmhicken, 2017-01-05:
I may have run into the same or similar problem to this issue:  Tripp-Lite SNMPWEBCARD segfault/errors with snmp-ups driver  #171

I have a Tripp-Lite SU10000RT3UPM with the TLNET card and I get the following errors:

[tripplite]
         driver = snmp-ups
         port = 10.0.1.156
         community = public
         snmp_version = v1
         pollfreq = 15

root@xymon:/lib/nut# ./snmp-ups -a tripplite -D
Network UPS Tools - Generic SNMP UPS driver 0.47 (2.4.3)
   0.000000     debug level is '1'
   0.000341     SNMP UPS driver : entering upsdrv_initups()
   0.001207     load_mib2nut: trying apcc mib
   0.021787     load_mib2nut: trying mge mib
   0.061714     load_mib2nut: trying netvision mib
   0.101741     load_mib2nut: trying pw mib
   0.141689     load_mib2nut: trying aphel_genesisII mib
   0.181715     load_mib2nut: trying aphel_revelation mib
   0.221699     load_mib2nut: trying raritan mib
   0.261758     load_mib2nut: trying baytech mib
   0.301745     load_mib2nut: trying cpqpower mib
   0.341674     load_mib2nut: trying ietf mib
   0.388987     load_mib2nut: using ietf mib
   0.428118     Detected SU10000RT3UPM    on host 10.0.1.156 (mib: ietf 1.3)
   0.428144     SNMP UPS driver : entering upsdrv_initinfo()
   0.428170     getting data: ups.mfr (1.3.6.1.2.1.33.1.1.1.0)
   0.468039     entering su_setinfo(ups.mfr)
   0.468059     getting data: ups.model (1.3.6.1.2.1.33.1.1.2.0)
   0.508177     entering su_setinfo(ups.model)
   0.508191     getting data: ups.firmware (1.3.6.1.2.1.33.1.1.3.0)
   0.548032     entering su_setinfo(ups.firmware)
   0.548050     getting data: ups.firmware.aux (1.3.6.1.2.1.33.1.1.4.0)
   0.568241     entering su_setinfo(ups.firmware.aux)
   0.568255     getting data: ups.serial (1.3.6.1.2.1.33.1.1.5.0)
   0.588102     entering su_setinfo(ups.serial)
   0.588116     getting data: ups.status (1.3.6.1.2.1.33.1.4.1.0)
   0.608023     su_find_infoval: found OL (value: 3)
   0.608034     getting data: ups.status (1.3.6.1.2.1.33.1.2.1.0)
   0.628140     su_find_infoval: no matching INFO_* value for this OID value (2)
   0.628151     getting data: ups.status (1.3.6.1.2.1.33.1.2.1.0)
   0.648202     su_find_infoval: no matching INFO_* value for this OID value (2)
   0.648212     getting data: ups.status (1.3.6.1.2.1.33.1.2.1.0)
   0.668217     su_find_infoval: found  (value: 2)
   0.668227     getting data: ups.test.result (1.3.6.1.2.1.33.1.7.4)
   0.688382     entering su_setinfo(ups.test.result)
   0.688396     getting data: ups.test.result (1.3.6.1.2.1.33.1.7.3)
   0.708134     su_find_infoval: found Done and passed (value: 1)
   0.708145     entering su_setinfo(ups.test.result)
   0.708153     getting data: battery.charge (1.3.6.1.2.1.33.1.2.4.0)
   0.728027     entering su_setinfo(battery.charge)
   0.728040     getting data: battery.runtime (1.3.6.1.2.1.33.1.2.3.0)
   0.748035     entering su_setinfo(battery.runtime)
   0.748047     getting data: battery.runtime.low (1.3.6.1.2.1.33.1.9.7.0)
   0.768186     [tripplite] unhandled ASN 0x5 recieved from 1.3.6.1.2.1.33.1.9.7.0
   0.768196     getting data: battery.voltage (1.3.6.1.2.1.33.1.2.5.0)
   0.788324     entering su_setinfo(battery.voltage)
   0.788336     getting data: battery.current (1.3.6.1.2.1.33.1.2.6.0)
   0.808026     [tripplite] unhandled ASN 0x5 recieved from 1.3.6.1.2.1.33.1.2.6.0
   0.808037     getting data: battery.temperature (1.3.6.1.2.1.33.1.2.7.0)
   0.828177     entering su_setinfo(battery.temperature)
   0.828190     getting data: output.phases (1.3.6.1.2.1.33.1.4.3.0)
   0.848112     setvar 1.3.6.1.2.1.33.1.4.3.0
   0.848122     entering su_setinfo(output.phases)
   0.848129     getting data: output.frequency (1.3.6.1.2.1.33.1.4.2.0)
   0.868197     entering su_setinfo(output.frequency)
   0.868208     Check outphases
   0.868212     outphases is set
[... 129 lines cut ...]
   1.328175     entering su_setinfo(output.voltage)
   1.328185     getting data: output.current (1.3.6.1.2.1.33.1.4.4.1.3.1)
   1.348171     entering su_setinfo(output.current)
   1.348182     getting data: output.power (1.3.6.1.2.1.33.1.4.4.1.4.1)
   1.368238     entering su_setinfo(output.power)
   1.368249     getting data: ups.load (1.3.6.1.2.1.33.1.4.4.1.5.1)
   1.388164     entering su_setinfo(ups.load)
   1.388174     getting data: input.phases (1.3.6.1.2.1.33.1.3.2.0)
   1.408220     setvar 1.3.6.1.2.1.33.1.3.2.0
   1.408231     entering su_setinfo(input.phases)
   1.408236     getting data: input.frequency (1.3.6.1.2.1.33.1.3.3.1.2.1)
   1.428096     entering su_setinfo(input.frequency)
   1.428107     getting data: input.voltage (1.3.6.1.2.1.33.1.3.3.1.3.1)
   1.448366     entering su_setinfo(input.voltage)
   1.448403     getting data: input.quality (1.3.6.1.2.1.33.1.3.1.0)
   1.468321     entering su_setinfo(input.quality)
   1.468337     SU_CMD_MASK => 1.3.6.1.2.1.33.1.8.2
   1.468341     SU_CMD_MASK => 1.3.6.1.2.1.33.1.7.1
   1.468344     SU_CMD_MASK => 1.3.6.1.2.1.33.1.7.1
   1.468418     SNMP UPS driver : entering upsdrv_updateinfo()

    How would I try the new driver you updated for issue #171?

Best regards, 
Dave

clepple, 2017-01-06:
What OS and distribution are you using?

Some distributions have newer versions of packages such as NUT in a "backports" section. On others, you can download the source to a recent version of NUT, and configure it to match the installed paths. Details vary widely between distributions.

dmhicken, 2017-01-06:
It was a pre packaged linux vm

Linux version 2.6.32-5-686 (Debian 2.6.32-31)
Dies this answer the question?
Best regards, Dave

clepple, 2017-01-06:
> Linux version 2.6.32-5-686 (Debian 2.6.32-31)

Is it possible to upgrade the VM? There are newer versions of NUT in the supported Debian repositories:

https://packages.debian.org/search?keywords=nut <https://packages.debian.org/search?keywords=nut>

You could also point the upsmon configuration in that VM to a newer system.

It is still an option to install the development packages to rebuild NUT in that VM, but that might not be the easiest approach.

LXXero, 2018-07-18:
I had this issue when using protocol v2c, if you use v1, read only, it doesn't happen, but of course it's read-only. Although, I tried out the write features of the v2c, and it didn't work anyway. I suspect if we resolve issue 309 and everyone is using the latest firmware, these problems will all go away and we could probably get some of the snmpset/write-commands working as well.
https://github.com/networkupstools/nut/issues/309