networkupstools/nut #359: APC Smart-UPS C1500 data

gr0mit1, 2017-01-11:
Hi
I've just bought a new APC SmartUPS C1500 and the data I'm getting has a lot of parameters missing.  Specifically I'm not getting load, line voltage (should be around 240 volts) or output voltage.

Any ideas please?  Also, if you need any diagnostics to get more infor let me know and I can run them.

Last login: Wed Jan 11 09:00:49 2017 from 192.168.100.74
root@ibworth-pi:~# /usr/local/ups/bin/upsc farnham
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.runtime: 17820
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 27.3
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion 
device.model: Smart-UPS C 1500
device.serial: 3S1632X07908  
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4-177-g7427a32
driver.version.data: APC HID 0.96
driver.version.internal: 0.42
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.firmware: UPS 10.0 / ID=1005
ups.mfr: American Power Conversion 
ups.mfr.date: 2016/08/17
ups.model: Smart-UPS C 1500
ups.productid: 0003
ups.serial: 3S1632X07908  
ups.status: OL
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.vendorid: 051d

clepple, 2017-01-12:
It is possible that your UPS has a MODBUS interface, which is not yet implemented in NUT (see issue #139)

curlyel, 2018-03-22:
Same issue here. Just the basic parameters as listed by @gr0mit1 for the Smart-UPS C1500. Has there been some progress in supporting it?

curlyel, 2019-09-17:
... one year later ;-)
I'm still not having a working solution which provides me more than the very basic information from my UPS.

The only software I've tried so far which provides the internal sensor readings is _APC PowerChute Business_. This proves, that the UPS has the measurements and that they are available via usb (modbus? or _whatever_?):
[image]

Recently, I reached out to APC support to ask, if the `Smart-UPS C 1500`  is supporting modbus. They did not really understand and said: "_PowerChute Business_ is the only software supported" :-(

Have to admit, that I gave `apcupsd `a try ;-)
They claim to have a working `modbus-usb` support. I got it compiled with the modbus-usb option but did not got the UPS working with the `MODBUS UPS driver`.

Any idea how to proceed?

clepple, 2019-09-18:
@curlyel I wish I had a better answer, but I have had a Smart-UPS C 1000 sitting around for over a year, and as I mentioned to Gene on the nut-upsuser email list the other day, it's hard to argue with "it just works". (Developing a NUT APC Modbus driver would probably mean some network downtime, and it is currently powering the NUT Buildbot server. Moving it is certainly an option, but inertia...)

That said, I am somewhat surprised that the `apcupsd` driver didn't work. If you are using packages, it is possible that the `apcupsd` package has different USB rules than NUT, so you might need to remove NUT completely (verifying that the uninstall scripts have stopped the NUT driver), unplug the USB cable, (re-)install `apcupsd`, and try again.

Another thing that can help is to see what `lsusb -vvv -d 051d:` returns. I'm especially interested in the part after `Report Descriptors:` (though the whole output would be useful). You may need to start and stop the `usbhid-ups` driver once to detach the kernel HID driver so that it doesn't just say `** UNAVAILABLE **`.

curlyel, 2019-09-18:
Thanks @clepple for looking into it!
> is to see what `lsusb -vvv -d 051d:` returns

O.k. that's an easy one:

```
# lsusb -vvv -d 051d:

Bus 004 Device 003: ID 051d:0003 American Power Conversion UPS
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x051d American Power Conversion
  idProduct          0x0003 UPS
  bcdDevice            1.06
  iManufacturer           1 American Power Conversion
  iProduct                2 Smart-UPS C 1500 FW:UPS 10.0 / ID=1005
  iSerial                 3 <cleared>
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     515
          Report Descriptor: (length is 515)
            Item(Global): Usage Page, data= [ 0x84 ] 132
                            Power Device Page
            Item(Local ): Usage, data= [ 0x04 ] 4
                            UPS
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Local ): Usage, data= [ 0x24 ] 36
                            Power Summary
            Item(Main  ): Collection, data= [ 0x02 ] 2
                            Logical
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x01 ] 1
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report ID, data= [ 0x01 ] 1
            Item(Local ): Usage, data= [ 0xfe ] 254
[... 435 lines cut ...]
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval              20
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval              10
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
#
```

clepple, 2019-09-18:
The `ModbusRTUTx` Usage ID (FF86:FD) is in the descriptor under Report ID 0x89:

```
            Item(Global): Usage Page, data= [ 0x86 0xff ] 65414
                            (null)
            Item(Global): Report ID, data= [ 0x89 ] 137
            Item(Local ): Usage, data= [ 0xfd ] 253
                            (null)
```

From `MPAO-98KJ7F_R1_EN.pdf` ("Application Note 176: Modbus implementation in Smart-UPS"; available from APC.com, and R0 from apcupsd source tree):

>All APC Smart-UPS that support the Modbus communication interface ship with this feature disabled. To enable the Modbus communication functionality, in the front control console of the UPS, go to Advanced Menu mode, under Configuration and enable Modbus.

curlyel, 2019-09-18:
> enable the Modbus communication functionality, in the front control console of the UPS, go to Advanced Menu mode

Yes, I found this paper too. Unfortunately, my UPS does not have the option to enter a "Advanced Menu". The frontpanel LCD has some symbols only and a numeric line and the "Configuration Menu" is limited to setting of `sensitivity, output voltage, green-mode` and `self-test`. 

No advanced mode :-(
See manual page below...

>but I have had a Smart-UPS C 1000 sitting around

The front panel looks identical. Maybe you can try it yourself. Probably you know some trick ;-)

Could it be that somehow _PowerChute_ can enable the `modbus mode` via the USB to gather the additional informations. Or is something among the `Report ID`'s listed above which is not "known" yet to `usbhid-ups` driver and therefore ignored? If so, can this be added to `apc-hid.c`?

That's the page from the manual:
[image]

curlyel, 2019-09-18:
The newer SMC1500I models with "Cloud Connect" seem to have the config option on front panel for modbus _although_ having the limited symbol/numeric display. _And:_ Not necessary to enable anymore. It's on by default now. :-(

clepple, 2019-09-21:
> Could it be that somehow PowerChute can enable the modbus mode via the USB to gather the additional informations.

Certainly possible. I have not used PowerChute recently, so I don't have much information there.

> Or is something among the Report ID's listed above which is not "known" yet to usbhid-ups driver and therefore ignored? If so, can this be added to apc-hid.c?

You can find the list of HID descriptor items not known by stopping NUT, manually starting the driver in debug mode, and looking for messages as the driver iterates through the HID items. It will probably need at least debug level 3, e.g. `/lib/nut/usbhid-ups -DDD -a ups`

I think we got most of them that we know, though, because 500 bytes is shorter than the descriptors for an UPS that provides "everything".

Going the other way, these are the two known HID paths to `ups.load` on APC:

drivers/apc-hid.c:  { "ups.load", 0, 0, "UPS.Output.PercentLoad", NULL, "%.1f", 0, NULL },
drivers/apc-hid.c:  { "ups.load", 0, 0, "UPS.PowerConverter.PercentLoad", NULL, "%.0f", 0, NULL },

`PercentLoad` is page 0x84, usage 0x35. I am not seeing a usage 0x35 in the `lsusb` output, even in the vendor pages (xx `0xff`).

fastcat, 2020-11-30:
FWIW, while the manual for My Smart-UPS C 1500 matches what @curlyel posted above (which matches https://download.schneider-electric.com/files?p_File_Name=AHUG-9D8BRK_R4_EN.pdf&p_Doc_Ref=SPD_AHUG-9D8BRK_EN&p_enDocType=User%20guide), the _actual_ configuration menu on it matches a different manual (which is listed for the same product model lol): https://download.schneider-electric.com/files?p_File_Name=AHUG-9ZSBHT_R3_EN.pdf&p_Doc_Ref=SPD_AHUG-9ZSBHT_EN&p_enDocType=User%20guide ... I'm guessing this is a firmware or model revision issue somewhere.

This _does_ have an entry to enable the ModBus protocol, but, contrary to the manual which says it's default enabled, mine was in the disabled state.

Saentist, 2022-06-21:
https://youtu.be/feCh_2NXf5U
interesting solution for this case

jimklimov, 2022-09-12:
Coult be related to #1453

EetuRasilainen, 2022-12-27:
The older APC SMC series does not support Modbus. AFAIK only the most recent SMC devices with _SmartConnect_ may also be controlled trough Modbus.

As you have noticed the SMC devices only display rudimentary data through USB-HID. For full control these devices need to be controlled through the MicroLink protocol, which is what APC's proprietary PowerChute software is using for communication as well.

Unfortunately MicroLink is also not supported by NUT yet (see #1426).

TrevorPH, 2026-10-03:
I'm happy to report that the SMC1500I is now supported by the new apcmicrolink driver that is in nut when built from git master branch.