networkupstools/nut #361: nut-scanner fails to find libusb-compat libraries on openbsd without symlinking

lf-, 2017-01-15:
```
# pkg_info -L libusb-compat
Information for inst:libusb-compat-0.1.5p0

Files:
/usr/local/bin/libusb-config
/usr/local/include/usb.h
/usr/local/lib/libusb.a
/usr/local/lib/libusb.la
/usr/local/lib/libusb.so.10.0
/usr/local/lib/pkgconfig/libusb.pc
```

Since nut-scanner is looking only for `libusb-0.1.so`, it doesn't find it. With a symlink from `/usr/local/lib/libusb-0.1.so` to `/usr/local/lib/libusb.so.10.0`, it works correctly.

jimklimov, 2017-01-16:
Thanks, good catch. There is support for loops looking for alternate SONAMEs to load in other cases (not sure if it made way to upstream branch though), but anyhow should be relatively easy to fix.

jimklimov, 2017-01-18:
Found the fix which looks relevant, posted PR #362 - @lf- please see if it helps in your case?

clepple, 2017-01-19:
Since the ports tree may not be expecting to rerun autoconf, a `make dist` tarball of commit fe711672 (from the PR Jim mentioned) is available here: http://buildbot.networkupstools.org/~buildbot/docker-debian-jessie/snapshot/pull_362_ltdl/rfe7116748c2bc40994a1c4c8a99dcc1fcf0118d9-695/nut-v2.7.4-210-gfe711672.7.4.1.tar.gz

jimklimov, 2017-01-24:
@lf- : hi, did you have a chance to give this code a spin?

lf-, 2017-01-28:
No, sadly my router is too underpowered to build stuff in a reasonable time frame. The only other alternative is to spin up an openbsd VM on hyper-v (which it doesn't work on :/).

jimklimov, 2017-01-28:
Well, you can leave it building overnight ;)

And dump HyperV for e.g. Virtualbox =) (but if you really do - mind the step of disabling HyperV modules so they do not own CPU virtualization extensions before another hypervizing solution starts - this is detailed on the internet).

In practice, I gather that the symlinks works for you; the code worked for us in another LTDL use-case and now it is merged to common upstream/master...

Still, confirmation of a working solution for original problem would be nice before closing the issue.

closed 2017-03-04 by lf-

jimklimov, 2017-03-04:
@lf- : why was this closed? No longer a problem (e.g. tested fixed by that commit?) or something else?