networkupstools/nut #365: Later variable list for some UPS

NUTonAPCuser, 2017-01-23:
Hi there,
I'm running NUT 2.7.4 on at the moment two OMV installations, and going to do a third, all with different APC UPSs, and found, that at least some of the variables list for the UPSs, that I can find at NUT website are not fitting those, that I get from OMV plugins.
As lists, that I get, are some more significant as lists published at NUT website for same UPS, I'd offer mine for publishing or at least double-checking.
If you are interested, just let me know.
Best regards,
Johannes

clepple, 2017-01-24:
We are happy to add additional entries. Note that some of them have not yet been added to the main website, but are posted here every so often: http://buildbot.networkupstools.org/~buildbot/docker-debian-jessie/docs/latest/ddl/

Of the items mentioned at the bottom of the [hardware compatibility list](http://networkupstools.org/stable-hcl.html#footnotes), the most important are whether `ups.status` tracks online and battery status, and whether shutdown works. (Plus any OMV-specific settings and versions.)

NUTonAPCuser, 2017-01-24:
Hi Charles,

thanks for your reply.

Find attached a 7zip archive containing updated variables list of the 
two APC UPSes, that I use at my home. They are not APC's latest, but 
still interesting for home and small enterprise usage. BTW the info at 
buildbot... for those devices is same as on main website.

During the next days or maybe the next week, I'll be able to cross-check 
two other APC devices plus one Mustek. I'll report as well.

Whether shutdown works - well, I will try it out with my UPSes at home. 
Only at company, it is not quite as easy... On the other hand it must be 
done somewhen, anyway. So why not now? ;-)

Best regards,
Johannes

clepple, 2017-01-25:
You may need to post the files as attachments on the GitHub website. I didn't see an archive.

Also, I think we are mainly concerned with new and different results (which is how I interpreted "at least some of the variables list for the UPSs, that I can find at NUT website are not fitting those, that I get from OMV plugins"). We might ask for confirmation that things still work after a major change, but I don't think 2.7.4 included anything major related to APC variables.

NUTonAPCuser, 2017-01-25:
Guess archive was wrong format (7z) - website would not accept it, too, just ZIP.

Ok, your concerns may be right basically, and so I'm looking forward to read out the Mustek, as this device not in your reports yet. On the other hand, related to APC, there are some interesting differences in variables lists from the 2.0.x reports at buildbot to 2.7.4 that I'm using. They may be mainly for older products like the RS series - AFAIS the precedessor of Back-UPS-Pro series - which are still hot on second hand market. Just take a look at BR500I file in archive, and decide for your own.

Best regards,
Johannes

[New_readings_of_APC_UPS.zip](https://github.com/networkupstools/nut/files/730858/New_readings_of_APC_UPS.zip)

jimklimov, 2017-01-25:
@NUTonAPCuser : Johannes, please note that to test NUT shutdowns in a controlled environment (where you can plug the UPSes back in the wall before it is too late), you can edit the `upsmon.conf` file, specifically the line which defines your `SHUTDOWNCMD`. By default it triggers a Unix-like system graceful shutdown, but you can override it to anything - such as a custom script to log data (`echo` to file, be systematic via `logger`, post on the `wall`, etc.) so you avoid the actual shutdown.

Of course, you also have to be careful about the UPS not really running out of juice (e.g. recently calibrated batteries would be useful) and/or have the power redundant (two or more UPSes, N+1 PSUs on servers, ATS for things like switches with just a single power input).

Hope this helps,
Jim Klimov

NUTonAPCuser, 2017-01-26:
@jimklimov : Jim, I know about things like shutdowncmd etc. which I can edit in upsmon.conf. Only, my intension for this thread is to make people aware of additional variables listed by newer versions of NUT, that may be interesting to check e,g, state of accu and power line. (For which former versions of NUT just do not listed enough of variables...)

Best regards,
Johannes

aquette, 2017-03-02:
hi @NUTonAPCuser 
thanks for your report!
Just to ensure my understanding: you just want to provide updated reports for these units, using NUT 2.7.2? Or have you identified any regression or issues in the data you see now?

As far as I understand, it case 1 (provide updated dumps), which is very much appreciated.
In which case, could you please send instead the standard upsc/upsrw/upscmd outputs, which would make our life easier to integrate into the DDL?
Note that there is a script to help you there [tools/nut-ddl-dump.sh](https://github.com/networkupstools/nut/blob/master/tools/nut-ddl-dump.sh)

closed 2017-05-22 by clepple