networkupstools/nut #372: Documentation Issues

Geraden07, 2017-01-27:
(For the TL;DR, see last section)

### The Issue
I've found three issues in the documentation I feel need to be addressed.

1. In the `upscli_add_host_cert` description, there is no warning that the call takes no action if using OpenSSL. This call will only *do* anything if compiled using NSS.
2. In the `upscli_connect` description, there is mention of the UPSCLI_CONN_TRYSSL and UPSCLI_CONN_REQSSL flags, but no mention of the UPSCLI_CONN_CERTVERIF flag added in version 2.7.
3. In `upscli_init` description, it makes the **erroneous** claim that if compiled with OpenSSL, the parameter certpath should be the path to a .pem file. This is wrong for reasons discussed below.

### Regarding `upscli_init`
The `upscli_init` capath ultimately gets passed to OpenSSL's `SSL_CTX_load_verify_locations` which has parameters for cafile and capath (see man page: https://www.openssl.org/docs/man1.0.2/ssl/SSL_CTX_load_verify_locations.html ).

However, the `upscli_init` code passes to the OpenSSL **capath**, *not* the cafile. As a result, what is required is *not a file but a directory containing* the certificate files.

Further more, the directory pointed to by capath needs to have the certificates named in a specific manner and will not just load .pem files. The .pem format is the right file format, but links should be created with the certificate's hash concatenated with a ".0" file extension, as this is how OpenSSL requires they be named. If two certificates happen to hash to the same value, the ".0" can be incremented to ".1" and so on as needed.

For example, if a file ca.pem needs to be loaded, the capath should point to the directory where it can be found, and it must be either linked or renamed to `a1b2c3e4.0` where "a1b2c3e4" is the hash of the certificate. If this conflicts with any other certificate in that directory with that same hash, we could name/link the file as `a1b2c3e4.1` or `a1b2c3e4.2` instead.

The command to get the hash that OpenSSL expects is:
```
openssl x509 -hash -noout -in ca.pem
```

So to create a link to a ca.pem file, the following bash command can be used:
```
ln -s ca.pem ./$(openssl x509 -hash -noout -in ca.pem).0
```
This assumes .0 but as mentioned before, one can increment that to ".1" or more as needed to avoid naming conflicts.

Of note as well, there exists the c_rehash utility provided by the openssl-perl package that will take a directory as an argument and iteratively link all certificates found there using the convention outlined above. For example:
```
c_rehash /path/to/certs/folder
```

### TL;DR
The long and short of it is I've discovered three issues with the documentation. One where a warning should be added, another where a new flag recently added in the code should be documented, and finally where the documentation is flat out *wrong* and needs to be corrected.

I've created this issue and so I can link it to the pull request I will submit containing fixes to the documentation.

clepple, 2017-01-27:
Thanks for putting this together. #373 looks good to me as well, but before we close this issue, can someone check to see if this issue also affects the documentation of the configuration files, such as upsd.conf and upsmon.conf?

Geraden07, 2017-02-05:
In the file `upsmon.conf.txt` on line 335 - 339, it correctly describes that the capath should be a directory containing the certificates looked up by their hash.

The upsd.conf.txt is different though because it's the server side. I'm only familiar with the code in the upsclient.c and so I can't speak as to whether the server side documentation is inaccurate or not.

jimklimov, 2017-02-07:
I believe currently the client-side documentation on this matter has been reviewed (upsmon) and updated (upscli), so PR #373 is done.

The issue stays open for now, as someone should do at least a revision for the server side. Right? :)