networkupstools/nut #382: POWEROFF_WAIT option has no effect with systemd

bigon, 2017-01-28:
sHi,

When using systemd, it seems that `POWEROFF_WAIT` has no effects.

The /lib/systemd/system-shutdown/nutshutdown script is just executing `upsdrvctl shutdown` and that's all.

This means that his feature is not working under systemd.

Shouldn't that script also check if the `POWEROFF_WAIT` is set and just wait like the power-off action of the LSB initscript was doing?

I'm not sure if these scripts are also bounded to the 90s timeout of systemd

See: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=835634

jimklimov, 2017-01-28:
I think this is not quite an issue (or rather one not solvable at this level), and a similar one is in effect for Solaris 10+ (including many OpenSolaris/illumos related distributions with SMF) where a binary rather than scripts control the actual lifecycle and timeouts and shutdown of a system.

In earlier distros driven purely by initscripts, waiting in the last init-script or in `rc.local` or equivalent might allow to avoid power-races. Now when a different system oversees OS shutdowns and aborts services that "take long" to become gracefully disabled (and where service management is done in parallel so you can not ubiquitously ensure that yours will be the last one standing; in case of Solaris and descendants one can set up any or even no timeout on start/stop of a service, but this does not hold for OS shutdowns), there is little one can do inside the sandbox of init scripts and service manifests.

My best (theoretical) take on the issue was that if the particular end-users do want power-race avoidance (not all do), they should enable some kernel module provided as part of NUT (TODO!) and the module as part of the inner-circle of the kernel might refuse to let the system turn off. I am not sure that this will work at all, or on all interesting kernels (and the module implementations will probably be either ifdef or fully separate for different OSes out there), but it seems the best way to solve this entanglement.

bigon, 2017-01-28:
I've added https://github.com/bigon/nut/commit/f3e7f32e6d09d0e9f6eb6f459f5def57993d8548 I think that should fix this bug

`systemctl reboot --force --force` will reboot the machine even if the shutdown was requested (I tested this)

bigon, 2017-01-28:
Looking at the `systemd-halt.service(8)` manpage, I can see:

>Immediately before executing the actual system halt/poweroff/reboot/kexec systemd-shutdown will run all executables in /lib/systemd/system-shutdown/ and pass one arguments to them: either "halt",
>"poweroff", "reboot" or "kexec", depending on the chosen action. All executables in this directory are executed in parallel, and execution of the action is not continued before all executables
>finished.

So it seems that the script is not limited to the usual timeout, but is still run in parallel with the other scripts. This requires more testing of course.

On debian, I see that there are only 3 scripts (including the nut one) installed by a package in that directory:

> apcupsd: /lib/systemd/system-shutdown/apcupsd_shutdown
> mdadm: /lib/systemd/system-shutdown/mdadm.shutdown
> nut-server: /lib/systemd/system-shutdown/nutshutdown

Only the mdadm one _could_ be an issue here if the machine is killed by the UPS before the process has returned. But this is already the case ATM anyway.

PR #384 (2017-03-13): Add support for POWEROFF_WAIT parameter
Use "systemctl reboot --force --force" to reboot the machine has not
been shutdown after the POWEROFF_WAIT delay.

Closes: #382
commit 49cd54a7 (2017-03-13): Add support for POWEROFF_WAIT parameter

Use "systemctl reboot --force --force" to reboot the machine has not
been shutdown after the POWEROFF_WAIT delay.

Closes: #382
commit 423a2f04 (2020-10-24): Add support for POWEROFF_WAIT parameter

Use "systemctl reboot --force --force" to reboot the machine has not
been shutdown after the POWEROFF_WAIT delay.

Closes: #382
closed 2017-03-13 by aquette