networkupstools/nut #383: upsdrvctl shutdown doesn't wait that the ups is shutdown

bigon, 2017-01-28:
Hi,

After issuing `upsdrvctl shutdown` the command confirms that it has initiated the ups shutdown and it then immediately returns and doesn't wait for the ups to shutdown.

This is a bit a problem as the machine will continue to shutdown and will not wait for the power to be removed.

With my eaton ellipse pro, it takes around 10s for the ups to shutdown

jimklimov, 2017-01-28:
Well, on a system with multiple UPSes you might want to request shutdown from many devices, so blocking on the first one might be a problem (not that this idea prohibits one from adding a flag to wait).

A UPS may also be configured to delay after the command - specifically for the purpose of letting the fed device finish its graceful shutdown routine (umount, remount ro, sync, poweroff, etc).

If your system if fed by this UPS, it will lose a power input when the UPS does go down; on a single-PSU system this will be the hard ungraceful shutdown event, on others it may count into a formula for upsmon and MINSUPPLIES. In this case the power will be removed regardless of if the machine waits for it or not ;)

On an admin workstation fed from some other source you might send the command, and UPS can die without telling you that "now I'm gonna do it!"

In short, for this issue I fail to see an actual practical problem. It may touch on power-race avoidance (e.g. "we initiate shut down when power is critical, so be almost off and sleep long, then reboot : if wall-power gets lost, so be it; if not - we get back up"), but that may be better solved elsewhere.

bigon, 2017-01-29:
OK, I was indeed confused

closed 2017-01-29 by bigon