networkupstools/nut #385: Ellipse 750 using USBHD-UPS driver: Update of ups.timer.{shutdown,start} somtimes fails

ledav-net, 2017-01-31:
Hi,

I wrote a little watchdog script to update regularly the values of ups.timer.{shutdown,start} and noticed that sometimes the registers are not well updated. I needed to workaround this by updating the registers again and again until well done...

Is someone already experienced such issue ?

I'm using an old Ellipse 750 ups with USBHID-UPS driver (same problem using the serial device with mge-shut driver) and NUT 2.7.4 on a CentOS 5.11 release.

Here is the output of my script:

! Timers off/on = 297/320 (before)
! Timers off/on = 597/2684355180
! Timers off/on = 598/320400
! Timers off/on = 598/185260
! Timers off/on = 599/143990
! Timers off/on = 598/620
Tue Jan 31 11:12:53 CET 2017 ups-watchdog UPS scheduled to power off/on in 598/620 seconds
!! Timers off/on = 299/320 (before)
! Timers off/on = 67109462/620
! Timers off/on = 597/620
Tue Jan 31 12:48:39 CET 2017 ups-watchdog UPS scheduled to power off/on in 597/620 seconds

The (before) is the value read from the ups before update. Then the value is read again to ensure all is well updated.

I'm using upsc the get the values and upscmd to set them...

Here is a upsc output:
$ upsc mgeups
battery.charge: 100
battery.charge.low: 50
battery.runtime: 1331
battery.type: PbAc
device.mfr: MGE OPS SYSTEMS
device.model: Ellipse 750
device.serial: BDCK350HC
device.type: ups
driver.name: usbhid-ups
driver.parameter.lowbatt: 50
driver.parameter.ondelay: 6
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ups/ellipse
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: MGE HID 1.39
driver.version.internal: 0.41
input.transfer.high: 264
input.transfer.low: 184
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 2
outlet.1.status: on
outlet.1.switchable: no
outlet.desc: Main Outlet
outlet.id: 1
outlet.switchable: no
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 6
ups.load: 16
ups.mfr: MGE OPS SYSTEMS
ups.model: Ellipse 750
ups.power.nominal: 750
ups.productid: ffff
ups.serial: BDCK350HC
ups.status: OL
ups.timer.shutdown: 372
ups.timer.start: 400
ups.vendorid: 0463

Thanks in advance for your feedback.

Regards,
David.

ledav-net, 2017-04-19:
usb 2-2: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 161 rq 1 len 4 ret -110

wazoox, 2018-06-25:
Same problem here with Eaton UPS. We have two problem: MGE/Eaton UPS USB implementation sucks, and nut communication error management sucks. In fact, nut 2.7.2 works only for 5 minutes or so, then after the first USB error fails to monitor the UPS until I restart the nut-driver service manually. Nut 2.7.4 monitors the UPS OK, but constantly spams /var/log/messages with 

` usb 3-1: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 161 rq 1 len 6 ret -71`