networkupstools/nut #391: Beeps when a UPS is disconnected

vporton, 2017-02-10:
My UPS is in repair. The UPS system on my Debian Linux beeps regularly.

Please make it not to beep if UPS is totally disconnected, automatically.

clepple, 2017-02-11:
This is a feature. (What if someone accidentally disconnected the cable, and a power failure occurred?)

You can turn off the `WALL` flag in the `NOTIFYMSG` section of `upsmon.conf`.

aquette, 2017-03-02:
hi @vporton 
Thanks for your report!
Additionnaly to @clepple comment, you can
* simply ignore the bad (loss of) communication, as per http://networkupstools.org/docs/man/upsmon.conf.html:
NOTIFYFLAG COMMBAD IGNORE
* stop NUT while the UPS is in repair and restart it after.

Please keep in mind that the point you have raised is a potential HW issue (as stated by @clepple) that needs user/admin review and action. There is nothing NUT can do automatically on its own, without putting the protected system in a dangerous and harmful position.

I will therefore close this ticket.
If you feel that there still be something NUT can do, please feel free to reopen this ticket and provide us with more info, so that we can think about a suitable solution. (Ex: one thing that comes to mind is that your system is protected by 2 UPSs, among which 1 is in maintenance for few minutes / hours. Your may want to have a command to put this specific UPS in "maintenance mode" and disable monitoring for a transient period...)

closed 2017-03-02 by aquette