networkupstools/nut #409: USB driver EA200LED/EA2150 LED 1500VA

Soyasi, 2017-03-20:
The UPS is this http://www.saicanarias.com/productos/ups/ea_200_led.html#02

I found this issue https://github.com/networkupstools/nut/issues/307

```
nut-scanner -NU
Scanning USB bus.
[nutdev1]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        product = "MEC0003"
        bus = "005"
```

```
/usr/lib/ups/driver/nutdrv_qx -a myups -DDD        
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '3'
   0.000248     upsdrv_initups...
   0.005983     Checking device (1D6B/0003) (006/001)
   0.006006     Failed to open device, skipping. (Permission denied)
   0.006010     Checking device (0001/0000) (005/005)
   0.013708     - VendorID: 0001
   0.013715     - ProductID: 0000
   0.013718     - Manufacturer: unknown
   0.013720     - Product: MEC0003
   0.013722     - Serial Number: unknown
   0.013725     - Bus: 005
   0.013727     - Device release number: 0100
   0.013730     Trying to match device
   0.013754     Device matches
   0.013773     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.013782     send: QGS
   0.013784     read: QGS
   0.013791     qx_process_answer: short reply (input.voltage)
   0.013794     send: QGS
   0.013796     read: QGS
   0.013799     qx_process_answer: short reply (input.voltage)
   0.013802     send: QGS
   0.013804     read: QGS
   0.013806     qx_process_answer: short reply (input.voltage)
   0.013810     send: M
   0.013812     read: M
   0.013817     voltronic_qs_protocol: invalid protocol [M]
   0.013820     send: M
   0.013822     read: M
   0.013824     voltronic_qs_protocol: invalid protocol [M]
   0.013827     send: M
   0.013829     read: M
   0.013832     voltronic_qs_protocol: invalid protocol [M]
   0.013837     send: M
   0.013847     read: M
   0.013852     voltronic_qs_hex_protocol: invalid protocol [M]
   0.013856     send: M
   0.013858     read: M
   0.013862     voltronic_qs_hex_protocol: invalid protocol [M]
   0.013870     send: M
   0.013872     read: M
   0.013874     voltronic_qs_hex_protocol: invalid protocol [M]
   0.013878     send: QS
   0.013880     read: QS
   0.013884     qx_process_answer: short reply (input.voltage)
   0.013887     send: QS
   0.013889     read: QS
   0.013892     qx_process_answer: short reply (input.voltage)
   0.013894     send: QS
   0.013897     read: QS
[... 323 lines cut ...]
   1.182710     received 10 (85)
   1.182718     read: UPS No Ack
   1.190721     received 10 (85)
   1.190733     read: UPS No Ack
   1.198893     received 10 (85)
   1.198900     read: UPS No Ack
   1.206716     received 10 (85)
   1.206723     read: UPS No Ack
   1.214885     received 10 (85)
   1.214892     read: UPS No Ack
   1.222723     received 10 (85)
   1.222733     read: UPS No Ack
   1.230707     received 10 (85)
   1.230714     read: UPS No Ack
   1.238707     received 10 (85)
   1.238714     read: UPS No Ack
   1.238718     qx_process_answer: short reply (input.voltage)
   1.238722     Device not supported!
   1.238725     Device not supported!
```

> lsusb -s 005:005
> Bus 005 Device 005: ID 0001:0000 Fry's Electronics

Could you suggest me something else to try ?

clepple, 2017-03-21:
What if you add the following to `ups.conf`?

```
    protocol="megatec"
    subdriver="fabula"
```

(You can remove `bus="005"` if you only have one UPS on the system - that number might change later if you move the USB cable to another port.)

Soyasi, 2017-03-21:
```
[myups]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        product = "MEC0003"
        bus = "005"
        protocol="megatec"
        subdriver="fabula"
```result:
```
/usr/lib/ups/driver/nutdrv_qx -a myups -DDD
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '3'
   0.000313     upsdrv_initups...
   0.006518     Checking device (1D6B/0003) (006/001)
   0.006543     Failed to open device, skipping. (Permission denied)
   0.006548     Checking device (0001/0000) (005/005)
   0.008850     - VendorID: 0001
   0.008860     - ProductID: 0000
   0.008863     - Manufacturer: unknown
   0.008866     - Product: MEC0003
   0.008869     - Serial Number: unknown
   0.008872     - Bus: 005
   0.008875     - Device release number: 0100
   0.008878     Trying to match device
   0.008909     Device matches
   0.008934     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.008945     Skipping protocol Voltronic 0.06
   0.008951     Skipping protocol Voltronic-QS 0.07
   0.008956     Skipping protocol Voltronic-QS-Hex 0.10
   0.008961     Skipping protocol Mustek 0.07
   0.008966     Skipping protocol Megatec/old 0.07
   0.008971     Skipping protocol BestUPS 0.06
   0.008977     Skipping protocol Mecer 0.07
   0.008985     send: Q1
   0.011883     read: UPS No Ack
   0.011900     qx_process_answer: short reply (input.voltage)
   0.011905     send: Q1
   0.014777     read: UPS No Ack
   0.014787     qx_process_answer: short reply (input.voltage)
   0.014792     send: Q1
   0.017772     read: UPS No Ack
   0.017782     qx_process_answer: short reply (input.voltage)
   0.017787     Skipping protocol Zinto 0.06
   0.017791     Skipping protocol Q1 0.07
   0.017794     Device not supported!
   0.017797     Device not supported!
```

Without the bus:

[myups]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        product = "MEC0003"
        protocol="megatec"
        subdriver="fabula"
```
/usr/lib/ups/driver/nutdrv_qx -a myups -DDD
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '3'
   0.000388     upsdrv_initups...
   0.006597     Checking device (1D6B/0003) (006/001)
   0.006620     Failed to open device, skipping. (Permission denied)
   0.006624     Checking device (0001/0000) (005/005)
   0.009201     - VendorID: 0001
   0.009223     - ProductID: 0000
   0.009226     - Manufacturer: unknown
   0.009229     - Product: MEC0003
   0.009232     - Serial Number: unknown
   0.009235     - Bus: 005
   0.009238     - Device release number: 0100
   0.009240     Trying to match device
   0.009277     Device matches
   0.009311     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.009323     Skipping protocol Voltronic 0.06
   0.009331     Skipping protocol Voltronic-QS 0.07
   0.009336     Skipping protocol Voltronic-QS-Hex 0.10
   0.009345     Skipping protocol Mustek 0.07
   0.009348     Skipping protocol Megatec/old 0.07
   0.009354     Skipping protocol BestUPS 0.06
   0.009360     Skipping protocol Mecer 0.07
   0.009367     send: Q1
   0.011616     read: UPS No Ack
   0.011650     qx_process_answer: short reply (input.voltage)
   0.011656     send: Q1
   0.014586     read: UPS No Ack
   0.014608     qx_process_answer: short reply (input.voltage)
   0.014614     send: Q1
   0.017920     read: UPS No Ack
   0.017944     qx_process_answer: short reply (input.voltage)
   0.017954     Skipping protocol Zinto 0.06
   0.017960     Skipping protocol Q1 0.07
   0.017964     Device not supported!
   0.017969     Device not supported!
```
Thank you for trying to help me!!!

zykh, 2017-03-21:
Not sure if yours is the same problem of #307 (and others in NUT's lists).
In their case, if I recall correctly, the problem was that, using libusb (recte: libusb 0.x or libusb-compat), we can't reproduce the exact sequence of commands which appears to be needed to get valid and up-to-date values from the device. But now that we have a libusb 1.x branch, I think that, starting from it, we could implement something that should work for them.
Assuming you have the original manufacturer-provided monitoring program, a capture of it 'talking' to the device could help diagnosing the problem.

aquette, 2017-03-22:
FYI:
* I've just updated the libusb-1.0 branch against the master
* the issue in #307, and for which the fix was reported in the libusb-1.0 issue ( #300) was apparently hitting only Debian users with libusb 0.1, not the compat version of libusb 1.0

clepple, 2017-03-22:
> the issue in #307, and for which the fix was reported in the libusb-1.0 issue ( #300) was apparently hitting only Debian users with libusb 0.1, not the compat version of libusb 1.0

and presumably Ubuntu?

zykh, 2017-03-23:
Probably because I'm half asleep, but I'm not sure I understand the correlation between #300 and #307.

aquette, 2017-03-23:
I may also be off my shoes on too many topics but...
#307 is the one you pointed
#300 is the one mentioning the regression on libusb 0.1 and timeouts Vs the libusb 1.0 port
Maybe no correlation but complement... ;)

Soyasi, 2017-03-25:
In case it is something I leave the packages of my system:
search -i libusb
El repositorio Packman Repository no está actualizado. Puede ejecutar zypper refresh como usuario Root para actualizarlo.
Cargando datos del repositorio...
Leyendo los paquetes instalados...

E | Nombre              | Resumen                                                           | Tipo   
--+---------------------+-------------------------------------------------------------------+--------
i | libusb-0_1-4        | libusb-1.0 Compatibility Library for libusb-0.1                   | paquete
i | libusb-0_1-4-32bit  | libusb-1.0 Compatibility Library for libusb-0.1                   | paquete
i | libusb-1_0-0        | Biblioteca USB                                                    | paquete
i | libusb-1_0-0-32bit  | Biblioteca USB                                                    | paquete
i | libusb-1_0-devel    | Biblioteca USB                                                    | paquete
i | libusb-compat-devel | libusb-1.0 Compatibility Layer for libusb-0.1                     | paquete
i | libusbmuxd4         | A client library to multiplex connections from and to iOS devices | paquete
uname -r  4.11.0-rc3-1.g5799355-default
NAME="openSUSE Leap"
VERSION="42.2"

aquette, 2017-04-04:
@Soyasi : just to clarify a bit: nut-scanner only checks for USB vendor/product IDs
Which does not imply the bagazillion variation of the Qx protocol...
Have you tried with other "protocol" and "subdriver"?
@zykh is the Qx master here, so prefer to follow his recommendations ;)

Soyasi, 2017-04-04:
He probado con "nutdrv_qx" todos sus protocolos y sus subdrivers te dejo los resultados.
I have tested with "nutdrv_qx" all its protocols and its subdrivers I leave the results.

[EA2150_LED_1500VA.txt](https://github.com/networkupstools/nut/files/893387/EA2150_LED_1500VA.txt)

Joserron, 2018-12-17:
Hola @Soyasi , estoy teniendo problemas a la hora de configurar el UPS en el PFsense.
Es un MEC0003 3000kva y sigo tus instrucciones, al menos las del log que pegas, y no me lo acepta, ¿podrías indicarme cómo lo hiciste?, ¿configuraste el ups.conf a mano?
Saludos y gracias por adelantado