networkupstools/nut #410: Improvement idea: add pthread or forking support to (selected) NUT drivers for massive monitoring

jimklimov, 2017-03-22:
Our setup monitors dozens of devices using `netxml-ups`, `snmp-ups`, or the in-development `snmp-ups-dmf` drivers. Each copy of the respective NUT driver takes up some virtual and resident memory, and even if it is just in the range of 1-2Mbytes RSS (and about 4-5Mb virtual) on our appliance, these numbers add up considerably as there are many independent copies running (shared memory reported by `top` is under 1Mb for each).

There is a working hypothesis that some more resources can be conserved by (optionally?) causing the one driver process to thread or fork off, so most of the code and data in memory is shared and just the tracking of unique dstates and similar variables would differ. Thus, requesting a new (2nd+) instance of a driver to start, would instead pass a signal to the currently running master-instance to do some magic if needed (e.g. load new DMF data if not yet loaded by a previous driver in a way that later launched drivers can use this data instantly) and fork/thread a copy to run the loop tracking a single device.

End-user behavior -- e.g. starting via driver binary directly, via upsdrvctl, as well as service wrappings -- should remain the same regardless of whether we run independent or forked/threaded driver processes (for service wrappings note that the driver process initially launched by request to monitor a device would cause another process to do the work and itself would die off, this must be handled somehow). One possibly relaxed behavioral point might be that theoretically drivers can run with credentials of different user accounts - this might be doable for forked copies of a driver too, but likely not for threads of a single process.

The option can be triggered e.g. by a combination of default global setting in `ups.conf` and a custom setting for a specific section in the file (so handling both the case of "on for all, off for some" and vice versa) as selected by the end-user who manages the deployment.

This issue is logged to solicit, collect and track ideas and opinions in this area, and perhaps an eventual PoC to test the idea in practice.

While it is not likely for a deployment with directly-connected UPSes/ePDUs to benefit much from this approach (simply because there are usually few serial/USB ports reliably available, and not so many devices within a few meters reach from the box), this can prove useful for any sort of networked drivers (snmp, netxml, ipmi, powerman, mqtt, etc.) If the feature is made optional, support for it might as well be added in the `main.c` common code.

clepple, 2017-03-22:
Check /proc/<pid>/maps first. In the past, top has taken shortcuts on reporting shared memory due to the time needed to calculate that number.

A lot of the tables in NUT (snmp-ups, in particular) could be made "const" to prevent their pages from getting copied if an adjacent variable changes. However, this is only a few KB here and there. A new thread is going to have many more KB for its stack alone.

For an appliance, it might be better to consider an embedded libc.

I will not argue that certain drivers might benefit from this, but I would want to see hard numbers before considering this for the core driver code.

jimklimov, 2017-03-22:
Hmmm, ensuring const tables and other data might be a good starting point, thanks for the idea. 

Anyhow, I did not plan to force-push core code or anything, just to stir up discussion and get helpful inputs like yours. After all, reliably and responsibly testing whether the change was beneficial or harmful to the footprint is also important, so thanks for these pointers too :)