networkupstools/nut #415: upsmon: add ALARM support

aquette, 2017-04-04:
As per http://lists.alioth.debian.org/pipermail/nut-upsuser/2017-April/010591.html

Add support for the ALARM flag in ups.status to upsmon

Development tracked in https://github.com/networkupstools/nut/tree/upsmon_alarm

Luo2Jie2, 2017-04-04:
I think it would be better to handle the general case first, by implementing a fall-through <notify type> of "UNKNOWN" so that all status changes, no matter how weird, can be caught.  Such a catch-all <notify type> would also have caught the "ALARM" in the specific case raised in the mailing list. 

Later, specific new <notify type> values can be added after a more thorough discussion of the real needs.

Roger

aquette, 2017-04-06:
thanks for all your comments @Luo2Jie2 .
ALARM support is to me a good step, which was really missing, and which covers the base status.
However, I agree that this filters out "OTHER" (better wording than "UNKNOWN", no?) status changes.
We may consider that for a 2nd step.

Calling to @clepple , @zykh , @jimklimov for comments...

Luo2Jie2, 2017-04-06:
"OTHER" is certainly a better name than "UNKNOWN" for a catch-all.   Roger

clepple, 2017-04-08:
I don't want to sound like I am nit-picking on the name, but if we were to implement a catch-all, wouldn't it be better to make it match *everything* rather than things not explicitly named? Then it should be possible to have the user-provided notification script do the filtering, without worrying if the notification name has changed at some point.

Also, as of revision f7de20aee, there is an [unused variable warning in Buildbot](http://buildbot.networkupstools.org/public/nut/builders/Debian-x64-gcc/builds/759/steps/compile/logs/warnings%20%281%29).

aquette, 2017-11-17:
postponing to 2.7.6 to have time for the right considerations on this point...

jimklimov, 2024-11-01:
NOTE: #2657 / #2658 might have achieved this goal (TODO: Check what was discussed/desired here vs. what was done there)

PR #2709 (2024-12-06): Update upsmon ALARM support and related themes from issue #415
Closes: #415
Follow-up to: #2658
Touches on: #2708
commit 7f314af3 (2024-12-06): clients/upsmon.{c,h}, docs, conf: apply suggestions from issue #415 discussion adding to PR #2658 changes

Previously merged-in the couple of commits from issue
https://github.com/networkupstools/nut/issues/415 proposal at
https://github.com/networkupstools/nut/tree/upsmon_alarm
for posterity, now cherry-picking some changes and further
discussion ideas from mailing list thread at
http://lists.alioth.debian.org/pipermail/nut-upsuser/2017-April/010591.html

Closes: #415
closed 2024-12-06 by jimklimov