networkupstools/nut #416: Add support for Liebert GXT2U w/ SNMP/web Card

autumnwalker, 2017-04-05:
Hi All,

I have acquired a Liebert GXT2U UPS with a "voyager" SNMP/web card installed (note, this is NOT the same as the is-webcard). Trying to configure NUT using snmp-ups driver and receive the following output:

```
Network UPS Tools - Generic SNMP UPS driver 0.72 (2.7.3)
No matching MIB found for sysOID '.1.3.6.1.4.1.476.1.1.1.14'!
Please report it to NUT developers, with an 'upsc' output for your device.
Going back to the classic MIB detection method.
No supported device detected
```

Any chance of adding support for that MIB? It's available online to download or I can run something locally (I have access to the card and can walk the device).

jimklimov, 2017-04-06:
Hi. Generally, there are several approaches to such issues :)

One is that vendors should implement the standard generic IETF MIB for power devices (the "mibs=ietf" setting in the ups.conf section for the device, or equivalent "-x" flag while testing via command-line). If this card does expose that standard entry point, it may suffice at least for basic monitoring of power status, voltages, etc. - though nothing advanced and specific to the vendor and/or device capabilities, probably not management via upscmd/upsrw either.

The other option with currently standard NUT upstream/master is to copy and adapt one of the `drivers/*-mib.c` files to define a uniquely-named mapping table for the vendor MIB that you have, mention it in the list of structure references in snmp-ups.c, and recompile the driver. Each `*-mib.c` is just a C file with a large array of a structure to map SNMP OIDs to NUT variables.

For the latter approach, it would help you to download the vendor MIB to install into your net-snmp configuration directory, and snmpwalk the vendor entry point, so you'd have the commented results (what data is actually exposed and how it's named according to the MIB); also the comments usually present in the MIB text file are helpful to figure out the nuances.

Poking @aquette just in case he has more nuances to share (there was IIRC a helper script to automate some of this new-mapping work).

Unfortunately, the current NUT codebase requires to recompile the snmp-ups driver to incorporate any changes to such SNMP-NUT mappings. There is a DMF branch to support same mappings dynamically via loadable XML files, but the branch is large and review to integrate it into trunk has not yet been completed.

Also note that by your screenshots, your system uses NUT 2.7.3 that may miss quite a bit of improvements made in the current Git HEAD - so consider updating by compiling your own NUT (at least, if IETF does not suffice for you and you have to compile new driver anyway) :)

aquette, 2017-04-06:
@autumnwalker the script mentioned by @jimklimov creates the base implementation files (xxx-mib.c/h) and makes it easier to implement and complete such files, as long as you have the MIB available to resove numeric OIDs into textual (human friendly ones):
* https://github.com/networkupstools/nut/blob/master/scripts/subdriver/gen-snmp-subdriver.sh
* http://networkupstools.org/docs/developer-guide.chunked/ar01s04.html#snmp-subdrivers

autumnwalker, 2017-04-06:
Thanks guys! Looks like I am behind a version. I am trying to get this running on my unRAID box using a plugin that was developed there (https://github.com/Ambrotos/plugins/tree/master/nut). Perhaps I should take a step back and stand up vanilla NUT on a VM and get that working first.

I looked at the snmp-subdrivers documentation, but didn't dig into the actual usage where the plugin I am using doesn't have gen-snmp-subdriver.sh with it and I wasn't sure how to go about updating it specifically.

I'll fire up a plain Ubuntu VM with NUT tonight and see if I can get that to speak to my UPS. I will report back with progress (and questions).

aquette, 2017-04-06:
note @autumnwalker that you can simply get the gen-snmp-subdriver.sh by using
wget https://raw.githubusercontent.com/networkupstools/nut/master/scripts/subdriver/gen-snmp-subdriver.sh
Keep in mind to put the needed MIB(s) in the same dir to have it working optimally

aquette, 2019-08-09:
@autumnwalker  any update on this? If possible, the minimum would be to provide a complete SNMP walk of the unit, so that I can process this and bootstrap a subdriver.

autumnwalker, 2019-09-08:
Hey @aquette - admittedly I'm not familiar with SNMP. I loaded up the mib provided by Liebert and did a walk. Below are the results; is this what you are looking for?

lcUpsIdentManufacturer.0
Liebert
lcUpsIdentModel.0
GXT2-1500RT120
lcUpsIdentSoftwareVersion.0
2.7.3
lcUpsIdentSpecific.0
.iso.org.dod.internet.private.enterprises.emerson.liebert Corp.liebertUps.luExtensions.luUPStationGxt
lcUpsIdentFirmwareVersion.0
GXT2MR14
lcUpsIdentSerialNumber.0
0219100012AF051
lcUpsIdentManufactureDate.0
05JUL02
lcUpsBatTimeRemaining.0
43
lcUpsBatVoltage.0
54
lcUpsBatCapacity.0
51
lcUpsInputFrequency.0
60
lcUpsInputNumLines.0
1
lcUpsInputLine.1
1
lcUpsInputVoltage.1
120
lcUpsOutputFrequency.0
60
lcUpsOutputLoad.0
30
lcUpsOutputNumLines.0
1
lcUpsOutputLine.1
1
lcUpsOutputVoltage.1
120
lcUpsOutputCurrent.1
3
lcUpsOutputWatts.0
324
lcUpsInverterStatus.0
on(2)
lcUpsAlarms.0
0
lcUpsTestBattery.0
unknown(1)
lcUpsTestBatteryStatus.0
passed(2)
lcUpsControlOutputOffDelay.0
0
lcUpsControlOutputOnDelay.0
0
lcUpsControlOutputOffTrapDelay.0
0
lcUpsControlOutputOnTrapDelay.0
0
lcUpsControlUnixShutdownDelay.0
0
lcUpsControlUnixShutdownTrapDelay.0
0
lcUpsControlCancelCommands.0
unknown(1)
lcUpsControlRebootAgentDelay.0
0
lcUpsNominalOutputVoltage.0
120
lcUpsNominalInputVoltage.0
120
lcUpsNominalOutputFreq.0
60
lcUpsNominalInputFreq.0
60
lcUpsNominalOutputVaRating.0
1500
lcUpsNominalOutputWattsRating.0
1050
lcUpsOnBypass.0
no(3)
lcUpsBypassFrequency.0
60
lcUpsBypassNumLines.0
1
lcUpsBypassLine.1
1
lcUpsBypassVoltage.1
120
lcUpsConfigType.0
online(2)
lcUpsConfigBypassInstalled.0
yes(2)
lcUpsConfigModuleCount.0
0
lcUpsConfigAudibleStatus.0
enabled(2)
lcUpsConfigLowBattTime.0
2
lcUpsConfigAutoRestart.0
0
.1.3.6.1.4.1.476.1.1.3.1.1.0
00 00 01 01 01 02 00 00 00 00
.1.3.6.1.4.1.476.1.1.3.1.2.0
fe 00 00 a4 00 00 8c 00 c0 f8 00 e0 80 00 00 c0 00 e0 c0 00 00 ff 00 00 cf 00 00 00 00 00 00 00 00 00 00 00 f0 00 c0 ee 00 00
Request Failed: Get Response PDU received from 192.168.2.21 Error Indication in response: There is no such variable name in this mib. Errindex: 1

autumnwalker, 2019-09-08:
Would it be easier to just see the MIB from Liebert? [liebert-mib.zip](https://github.com/networkupstools/nut/files/3588173/liebert-mib.zip)

autumnwalker, 2019-10-26:
I spun up a Ubuntu box and installed NUT via APT. I pulled down gen-snmp-subdriver.sh and had it generate a .mib using mode 1. Now I'm lost - I don't have /drivers or snmp-ups.c in order to continue. 

For C-style integration, do not forget to:
* bump DRIVER_VERSION in snmp-ups.c (add "0.01")
* copy "liebert-voyager.mib-mib.h" and "liebert-voyager.mib-mib.c" to "../../drivers"
* add #include "liebert-voyager.mib-mib.h" to drivers/snmp-ups.c
* add &liebert-voyager.mib to drivers/snmp-ups.c:mib2nut[] list,
* add liebert-voyager.mib-mib.c to snmp_ups_SOURCES in drivers/Makefile.am
* add liebert-voyager.mib-mib.h to dist_noinst_HEADERS in drivers/Makefile.am
* "./autogen.sh && ./configure && make" from the top level directory

Ultimately - do I have to pave / nuke and build NUT from source in order to do this?

autumnwalker, 2019-10-28:
So, I found [this article](https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu) and followed the steps to get the dev files on my box. I followed the instructions there combined with the "C-style" integration instructions from above and I still get: 

Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic SNMP UPS driver 1.12 (2.7.4-734-g1b217f46)
No matching MIB found for sysOID '.1.3.6.1.4.1.476.1.1.1.14'!
Please report it to NUT developers, with an 'upsc' output for your device.
Going back to the classic MIB detection method.
No supported device detected
Driver failed to start (exit status=1)

aquette, 2019-11-06:
hi @autumnwalker 
bad timing, notification received while moving on vacation :dagger: 
could you please post a zip with the generated files.
Using these files, and doing the integration mentioned by the script implies indeed to modify sources and recompile, which may not be trivial.

autumnwalker, 2019-11-06:
Hi @aquette, no problem at all! I hope your move went well and you had a relaxing vacation!

Zip file is here with the .c and .h file generated by the subdriver script.

[liebert_voyager.zip](https://github.com/networkupstools/nut/files/3814400/liebert_voyager.zip)

aquette, 2019-11-14:
hi @autumnwalker 
finally got the few seconds to look at. thanks for the archive!
question: how would you be able to test? Git (branch or master), binary, ...?
I'd like some checks from you, since I don't have such HW underhand, despite the sample values in your provided dump.

autumnwalker, 2019-11-14:
Hey @aquette! I can test however you'd like me to. I can spin up a VM and install however it would work best.

The target install is with the NUT plugin for Unraid [https://forums.unraid.net/topic/60217-plugin-nut-v2-network-ups-tools/](url), but I have been doing my testing in a Ubuntu 18.04 VM with the Ubuntu NUT package + dev files. I can nuke / pave and pull from Git or go from binaries if you'd like.

aquette, 2019-11-15:
hi @autumnwalker 
so, going the git path, faster for me.

I've made a 1rst stab, available on the *liebert-voyager* branch.
To test it:
* git clone as usual
* switch to that branch (git checkout liebert-voyager)
* compile as usual

Few comments:
* this is a first stab. Basic info for upsc, no command nor settings yet!
* there is nothing to fill ups.status. This is probably provided through the IETF MIB.
Could you please send back an SNMP walk of ".1.3.6.1.2.1.33"
* I'm interested in your review, at least of the drivers.list info:
https://github.com/networkupstools/nut/compare/liebert-voyager?expand=1
* I'm also interested in an upsc

autumnwalker, 2019-11-15:
Thanks @aquette!

I tried to build NUT from scratch using the source on Git; did the following:

- Created a fresh Ubuntu 18.04 server container (LXC) and upgraded base packages.
- Installed libsnmp-dev, build-essential, autoconf, automake, and libtool.
- Created a user / group called ups. 
- Cloned and checked out the liebert-voyager branch.
- Ran ./autogen.sh
- Ran ./configure --with-user=ups --with-group=ups --with-snmp

Get the following error:
```
checking for pkg-config... no
./configure: line 5251: syntax error near unexpected token `dummy_PKG_CONFIG,'
./configure: line 5251: `         PKG_CHECK_MODULES(dummy_PKG_CONFIG, pkg-config,'
```

Doing an SNMP walk of ".1.3.6.1.2.1.33" returns nothing. Issued command:
```
snmpwalk -c <community> -v1 <IP> .1.3.6.1.2.1.33
```
My preliminary review of of your changes for drivers.list (and other files in compare) make sense. My only caveat is the "Voyager SNMP/web Card" name is the best that Liebert could determine based on the markings / serials on the card - so nobody is going to tell us otherwise I guess!

I'm going to try and install the Ubuntu package for NUT and install the liebert-voyager branch over it using the instructions on the Wiki - I'll see if that will allow me to get upsc for you.

clepple, 2019-11-16:
@autumnwalker You got most of the build steps, but steps 1 and 3 on the [wiki instructions](https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu) grab all of the dependencies like `pkg-config` automatically.

aquette, 2019-11-19:
@autumnwalker  thanks for the preliminary feedback. Please follow @clepple advice on compiling.

Beside from this, I'm still puzzled by your feedback on the missing IETF MIB info (empty results of snmpwalk on .1.3.6.1.2.1.33) and the lack of other OIDs to feed ups.status (at least OL, OB, LB).
The only thing I saw is the BYPASS status

Once good, or beside from this, could you make an additional test:
* having your UPS running from Main, do a "snmpget ... .1.3.6.1.4.1.476.1.1.1.1.5.1.0"
* then same running on battery
and provide back the results? 

According to these results, we may have to move on step 2 more quickly, and implement alarms handling to get other status bits.

autumnwalker, 2019-11-20:
Compiled via wiki and did a basic nut.conf (netserver) and ups.conf. Ran upsdrvctl start and got the following:

>root@nut01:/etc/nut# upsdrvctl start
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic SNMP UPS driver 1.12 (2.7.4-736-gf9fa0f63)
Detected GXT2-1500RT120                 on host 192.168.2.21 (mib: liebert_voyager 0.1)

Ran upsd and got the following:

>root@nut01:/etc/nut# upsd
Network UPS Tools upsd 2.7.4
fopen /var/run/nut/upsd.pid: No such file or directory
listening on 127.0.0.1 port 3493
listening on ::1 port 3493
Connected to UPS [GXT2U]: snmp-ups-GXT2U

Ran upsc and got the following:

>root@nut01:/etc/nut# upsc GXT2U
Init SSL without certificate database
battery.charge: 19
battery.voltage: 54
device.firmware: GXT2MR14        
device.firmware.aux: 2.7.3 
device.mfr: Liebert
device.model: GXT2-1500RT120                
device.serial: 0219100012AF051     
device.type: ups
driver.name: snmp-ups
driver.parameter.pollinterval: 2
driver.parameter.port: 192.168.2.21
driver.parameter.synchronous: no
driver.version: 2.7.4-736-gf9fa0f63
driver.version.data: liebert_voyager MIB 0.1
driver.version.internal: 1.12
input.frequency: 60
input.frequency.nominal: 60
input.phases: 1
input.voltage: 122
input.voltage.nominal: 120
output.current: 3
output.frequency: 60
output.frequency.nominal: 60
output.phases: 1
output.realpower: 363
output.voltage: 120
output.voltage.nominal: 120
ups.load: 34
ups.power.nominal: 1500
ups.realpower.nominal: 1050
ups.status: 
ups.test.result: 2

I also tried snmpget -c <community> -v1 <IP> .1.3.6.1.4.1.476.1.1.1.1.5.1.0 and got:

>SNMPv2-SMI::enterprises.476.1.1.1.1.5.1.0 = INTEGER: 2

I will have to wait until I get home to unplug the UPS and run the snmpget command again.

Looking at "battery.charge: 19" - looks incorrect. Battery should be 100 unless I am misinterpreting the "19".

Huge progress. Thank you for all the effort @aquette and @clepple!

aquette, 2019-11-21:
thx for the feedback @autumnwalker , indeed progressing ;)
that also confirms partially some of the ups.status points. This will btw cause me some implementation troubles for the "OL" status at least.
For the others, when doing your on battery test, could you please also snmpwalk on ".1.3.6.1.4.1.476.1.1.1.1.6"
as for the battery and 19, you're indeed right, it's not the percentage but the runtime (in minutes). Fixed now (and published as number of seconds in battery.runtime) along with added the battery remaining charge (as battery.charge).

autumnwalker, 2019-11-21:
Thanks again @aquette. Pulled, recompiled. Looks like the data is correct!

>root@nut01:~/nut/drivers# upsc GXT2U@192.168.2.12
Init SSL without certificate database
battery.charge: 100
battery.runtime: 1140
battery.voltage: 54
device.firmware: GXT2MR14        
device.firmware.aux: 2.7.3 
device.mfr: Liebert
device.model: GXT2-1500RT120                
device.serial: 0219100012AF051     
device.type: ups
driver.name: snmp-ups
driver.parameter.pollinterval: 2
driver.parameter.port: 192.168.2.21
driver.parameter.synchronous: no
driver.version: 2.7.4-737-gde5c8642
driver.version.data: liebert_voyager MIB 0.1
driver.version.internal: 1.12
input.frequency: 60
input.frequency.nominal: 60
input.phases: 1
input.voltage: 121
input.voltage.nominal: 120
output.current: 3
output.frequency: 60
output.frequency.nominal: 60
output.phases: 1
output.realpower: 373
output.voltage: 120
output.voltage.nominal: 120
ups.load: 35
ups.power.nominal: 1500
ups.realpower.nominal: 1050
ups.status: 
ups.test.result: 2

I wasn't able to get to the UPS last night, but I'll try to run the onbatt tests / walks tonight.

autumnwalker, 2019-11-22:
Ran the on battery tests. I ran them each without specifying the MIB and also with specifying (-m) the Liebert MIB. Interestingly, the first command returned the same result while powered and while on battery.

Powered:
>root@nut01:~# snmpget -c [community] -v1 [ip] .1.3.6.1.4.1.476.1.1.1.1.5.1.0
SNMPv2-SMI::enterprises.476.1.1.1.1.5.1.0 = INTEGER: 2

>root@nut01:~# snmpget -c [community] -v1 [ip] .1.3.6.1.4.1.476.1.1.1.1.5.1.0 -m LIEBERT-UPSTATION-GXT-UPS-MIB
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsInverterStatus.0 = INTEGER: on(2)

>root@nut01:~# snmpwalk -c [community] -v1 [ip] .1.3.6.1.4.1.476.1.1.1.1.6
SNMPv2-SMI::enterprises.476.1.1.1.1.6.1.0 = Gauge32: 0

>root@nut01:~# snmpwalk -c [community] -v1 [ip] .1.3.6.1.4.1.476.1.1.1.1.6 -m LIEBERT-UPSTATION-GXT-UPS-MIB
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarms.0 = Gauge32: 0

On Battery:

>root@nut01:/etc/snmp# snmpget -c [community] -v1 [ip] .1.3.6.1.4.1.476.1.1.1.1.5.1.0
SNMPv2-SMI::enterprises.476.1.1.1.1.5.1.0 = INTEGER: 2

>root@nut01:/etc/snmp# snmpget -c [community] -v1 [ip] .1.3.6.1.4.1.476.1.1.1.1.5.1.0 -m LIEBERT-UPSTATION-GXT-UPS-MIB
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsInverterStatus.0 = INTEGER: on(2)

>root@nut01:/etc/snmp# snmpwalk -c [community] -v1 [ip] .1.3.6.1.4.1.476.1.1.1.1.6
SNMPv2-SMI::enterprises.476.1.1.1.1.6.1.0 = Gauge32: 4
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.1.3 = INTEGER: 3
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.1.10 = INTEGER: 10
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.1.31 = INTEGER: 31
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.1.35 = INTEGER: 35
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.2.3 = OID: SNMPv2-SMI::enterprises.476.1.1.1.1.6.3.3
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.2.10 = OID: SNMPv2-SMI::enterprises.476.1.1.1.1.6.3.10
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.2.31 = OID: SNMPv2-SMI::enterprises.476.1.1.1.1.6.3.13
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.2.35 = OID: SNMPv2-SMI::enterprises.476.1.1.1.1.6.3.17
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.3.3 = Timeticks: (644584600) 74 days, 14:30:46.00
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.3.10 = Timeticks: (644584615) 74 days, 14:30:46.15
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.3.31 = Timeticks: (644584640) 74 days, 14:30:46.40
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.3.35 = Timeticks: (644584655) 74 days, 14:30:46.55

>root@nut01:/etc/snmp# snmpwalk -c [community] -v1 [ip] .1.3.6.1.4.1.476.1.1.1.1.6 -m LIEBERT-UPSTATION-GXT-UPS-MIB
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarms.0 = Gauge32: 4
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmId.3 = INTEGER: 3
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmId.10 = INTEGER: 10
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmId.31 = INTEGER: 31
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmId.35 = INTEGER: 35
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmDescr.3 = OID: LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmUtilFailed
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmDescr.10 = OID: LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmOnBattery
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmDescr.31 = OID: LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmInputFreqError
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmDescr.35 = OID: LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsBadBypassPower
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmTime.3 = Timeticks: (644584600) 74 days, 14:30:46.00
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmTime.10 = Timeticks: (644584615) 74 days, 14:30:46.15
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmTime.31 = Timeticks: (644584640) 74 days, 14:30:46.40
LIEBERT-UPSTATION-GXT-UPS-MIB::lcUpsAlarmTime.35 = Timeticks: (644584655) 74 days, 14:30:46.55

autumnwalker, 2020-01-21:
Hey @aquette, anything else you need from me here? I'm unclear on next steps. Thanks!

niallm90, 2020-05-05:
I also have an older Liebert GTX with the SNMP web card and can confirm this works. Would be keen to see this merged, can I be of any help?

autumnwalker, 2020-05-05:
@niallm90 this isn't ready for PROD yet. NUT still does not understand the status of the UPS (i.e. on battery / on mains).

niallm90, 2020-05-05:
Oh, face palm. I'll see if I can get any info form my unit today to help. Unfortunate my batteries are shot so I may not be able to even run the UPS off them with no load.

niallm90, 2020-06-07:
Took me a while to get some new batteries sorted but I've finally been able to and have a bit of a hack around. As far as I can tell their is no nice way to get the status with this UPS. I think I have worked out the logic to get the mode its in, however it doesn’t match the way any of the other SNMP sub-drivers work.

As far as I can tell:
- `OL` is the default for the UPS and has no explicit OID in the MIB. It can be inferred by a lack of the `lcUpsAlarmOnBattery` alarm, `lcUpsAlarmUpsOff` alarm and the `lcUpsInverter` state of `on(2)`.
- `OB` is indicated by the alarm `lcUpsAlarmOnBattery`
- `LB` is indicated by the alarm `lcUpsAlarmLowBatteryWarning`
- `BYPASS` is indicated by `lcUpsOnBypass` state of `on(2)`
- `OFF` is indicated by the alarm `lcUpsAlarmUpsOff`

I'm keen to see support for this UPS added and let me know if there is anything I can do to speed this up.

autumnwalker, 2020-12-16:
Just re-racked my lab (moved houses) and rigged up my GXT2U again and thought about revisiting this. @niallm90 thank you for putting together that logic. It makes sense to me and I can see those same parameters from mine.

@aquette can you help us build the logic into the driver? Is that even possible?

aquette, 2020-12-16:
I'll try to take a look tomorrow, but will need a lot of reactivity from your side to test. Possibly attach snmpwalk for both OB and OL so that I can snmpsimd on these

autumnwalker, 2020-12-16:
Thank you @aquette! Let me know what to do and I will run it here.

I can SNMP walk the conditions that @niallm90 indicated for OL and OB. All right now almost return no response as my UPS is on mains now, not battery. lcUpsInverter shows on(2).

Are you trying to build the mib for simulation? The MIB I'm using is here: http://www.oidview.com/mibs/476/LIEBERT-UPS-MIB.html

aquette, 2020-12-17:
@autumnwalker would you be able to test from git, using the branch liebert-voyager?

autumnwalker, 2020-12-17:
Running on latest code from branch liebert-voyager.

Running on mains:
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.6.3.10 - no value
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.6.3.1 - no value
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.13.1 - SNMPv2-SMI::enterprises.476.1.1.1.1.13.1.0 = INTEGER: 3
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.6.3.12 - no value

I will run on battery shortly and report back.

autumnwalker, 2020-12-17:
On battery:
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.6.3.10 - no value
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.6.3.1 - no value
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.13.1 - SNMPv2-SMI::enterprises.476.1.1.1.1.13.1.0 = INTEGER: 3
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.6.3.12 - no value

Same results.

aquette, 2020-12-17:
I've just pushed some mods, blind mode though (not even had time to try compiling)
plz try and report back, thx

autumnwalker, 2020-12-17:
Not seeing a change in the branch.

aquette, 2020-12-17:
Pushed on my personal repos

aquette, 2020-12-18:
now fixed and pushed again on nut repos:
https://github.com/networkupstools/nut/tree/liebert-voyager

autumnwalker, 2020-12-18:
Thank you @aquette!

Ran git pull
Ran ./autogen.sh
Ran ./configure --with-user=ups --with-group=ups --with-snmp
Ran systemctl restart nut-server

Running on mains:
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.6.3.10 - no value
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.6.3.1 - no value
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.13.1 - SNMPv2-SMI::enterprises.476.1.1.1.1.13.1.0 = INTEGER: 3
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.6.3.12 - no value

Running on battery:
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.6.3.10 - no value
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.6.3.1 - no value
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.13.1 - SNMPv2-SMI::enterprises.476.1.1.1.1.13.1.0 = INTEGER: 3
snmpwalk -c [community] -v1 [IP] .1.3.6.1.4.1.476.1.1.1.1.6.3.12 - no value

I also tried:
snmpwalk -c [community] -v1 [ip] .1.3.6.1.4.1.476.1.1.1.1.6
SNMPv2-SMI::enterprises.476.1.1.1.1.6.1.0 = Gauge32: 4
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.1.3 = INTEGER: 3
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.1.10 = INTEGER: 10
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.1.31 = INTEGER: 31
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.1.35 = INTEGER: 35
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.2.3 = OID: SNMPv2-SMI::enterprises.476.1.1.1.1.6.3.3
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.2.10 = OID: SNMPv2-SMI::enterprises.476.1.1.1.1.6.3.10
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.2.31 = OID: SNMPv2-SMI::enterprises.476.1.1.1.1.6.3.13
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.2.35 = OID: SNMPv2-SMI::enterprises.476.1.1.1.1.6.3.17
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.3.3 = Timeticks: (48592060) 5 days, 14:58:40.60
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.3.10 = Timeticks: (48592075) 5 days, 14:58:40.75
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.3.31 = Timeticks: (48592100) 5 days, 14:58:41.00
SNMPv2-SMI::enterprises.476.1.1.1.1.6.2.1.3.35 = Timeticks: (48592115) 5 days, 14:58:41.15

I wonder if I have the right code compiled / reboot / etc. properly? Results are all the same.

autumnwalker, 2020-12-18:
It just occurred to me that doing the walks means nothing with the updated code. 

Ran upsc and ups.status is blank.

upsc GXT2U@[local ip]
Init SSL without certificate database
battery.charge: 51
battery.runtime: 11280
battery.voltage: 53
device.firmware: GXT2MR14        
device.firmware.aux: 2.7.3 
device.mfr: Liebert
device.model: GXT2-1500RT120                
device.serial: 0219100012AF051     
device.type: ups
driver.name: snmp-ups
driver.parameter.pollinterval: 2
driver.parameter.port: 192.168.2.21
driver.parameter.synchronous: no
driver.version: 2.7.4-737-gde5c8642
driver.version.data: liebert_voyager MIB 0.1
driver.version.internal: 1.12
input.frequency: 60
input.frequency.nominal: 60
input.phases: 1
input.voltage: 113
input.voltage.nominal: 120
output.current: 4
output.frequency: 60
output.frequency.nominal: 60
output.phases: 1
output.realpower: 490
output.voltage: 120
output.voltage.nominal: 120
ups.load: 46
ups.power.nominal: 1500
ups.realpower.nominal: 1050
ups.status: 
ups.test.result: 2

clepple, 2020-12-20:
@autumnwalker commit de5c8642 (from the end of `driver.version`, after `2.7.4-##-g`) is from 2019 - you likely want a866637578 (from Dec 17, 2020) or better yet, 487c667c43 (latest on `liebert-voyager` branch, from Dec 18, 2020).

If you have pulled the branch with those commits, you'll need to rebuild and reinstall the `snmp-ups` driver (`make` in `drivers/` then `make install` as root), then restart the driver (`upsdrvctl restart` or equivalent init/systemctl incantation; probably `systemctl restart nut-server`?).

aquette, 2020-12-20:
Note to @clepple the data dump mode of the drivers is very convenient for such case (in tree test):
./drivers/snmp-ups -s test -d2 -x port=<up addr> -x <other params like community with each -x>

autumnwalker, 2020-12-20:
@clepple - I ran make and make install as root in drivers/ and restarted - upsc is reporting the same driver.version. I can see that files are updated today though (e.g. snmp_ups-liebert-voyager-mib.o). I can also see that liebert-voyager-mib.c was updated on the 18th (when I did git pull).

@aquette apologies. Where do you want me to add -DDD?

I might need some ELI5 here.

clepple, 2020-12-20:
@autumnwalker my mistake, I guess we are missing a dependency to automatically update the version when you run `make` from `drivers` (I am usually building from one directory up).

If I understand Arno's comment, he's saying to stop the rest of the NUT machinery (`systemctl stop nut-server` and possibly `systemctl stop nut-client`) and just test the driver without installing, using the standalone command he mentioned. (The systemd service scripts tell `upsdrvctl` to start the driver in a similar fashion, but for that, the driver needs to be installed to the same locations as the .debs. This new test mechanism sidesteps that, and lets you run the driver straight from where it is built with `make`, and doesn't require `upsd`. On the other hand, without `upsd`, you can't use `upsc`, but apparently the test mode prints similar output to `upsc`.) Something like this:

`path/to/nut/drivers/snmp-ups -s test -d2 -DDD -x port=<ip> -x community=<community>`

clepple, 2020-12-20:
@aquette while I was mentally walking through the dependency chain for fixing `driver.version` in this case (#938), it occurred to me that you could also just bump the version in the sub-driver when you make a change.

autumnwalker, 2020-12-21:
Running as root.

root@nut01:~/nut# ./drivers/snmp-ups -s test -d2 -DDD -x port=192.168.2.12 -x community=private
Network UPS Tools - Generic SNMP UPS driver 1.12 (2.7.4-737-gde5c8642)
   0.000000     [D1] debug level is '3'
   0.000982     Can't chdir to /var/run/nut: Permission denied

Even here - it seems to be running the de5c864 driver version. When I ran git pull from the nut directory (liebert-voyager branch) shouldn't that have grabbed the updated driver?

jimklimov, 2020-12-21:
Are you sure git also successfully checked out that new branch?

autumnwalker, 2020-12-21:
@jimklimov 

root@nut01:~/nut# git branch
* liebert-voyager
  master
  show-current

clepple, 2020-12-21:
> Even here - it seems to be running the de5c864 driver version. When I ran git pull from the nut directory (liebert-voyager branch) shouldn't that have grabbed the updated driver?

@autumnwalker It sounds like you might have the latest (you can confirm with `git show`, which prints the git version hash for the source), but the build rule to update the `driver.version` string is incomplete. A `make clean` from the top nut directory would fix it on the next `make`, but that is overkill - hence the new issue (#938).

aquette, 2020-12-21:
Try
./drivers/snmp-ups -s test -d2 -x root  -DDD -x
 port=192.168.2.12 -x community=private
Prefer the zip the result, more readable ;)

clepple, 2020-12-21:
Maybe `-u root` instead of `-x root`?

I thought the pathnames and usernames specified in [the wiki](https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu) should have taken care of any permissions issues, especially when starting the driver as root.

autumnwalker, 2020-12-22:
Running now. @clepple had to run with -u root per your suggestion.

@aquette what do you mean "zip the result"? Do you mean [command] > file.txt and then zip it / upload here?

autumnwalker, 2020-12-23:
Here is the result. https://pastebin.com/pzhLfXd8

Appears to still be running the old driver.

clepple, 2020-12-26:
@aquette downloaded from pastebin and posted here for future reference.

[GXT2U on mains.txt.gz](https://github.com/networkupstools/nut/files/5743440/GXT2U.on.mains.txt.gz)

@autumnwalker due to the newly uncovered bug I mentioned a few days ago (#938), you won't see `driver.version` update automatically after a `git pull` unless you run `make clean` from the top NUT source directory before rebuilding. Even if someone updates `driver.version.data` (MIB-specific version) that won't automatically fix `driver.version`, but at least that would show some difference. I don't think Arno has had a chance to push new code to do that yet.

autumnwalker, 2020-12-27:
Thanks @clepple. Should I run make clean on this? Would it make any difference? Should I run this on battery? Anything else I can do to help?

aquette, 2020-12-29:
Can someone please extract the interesting oids from the driver output?
.1.3.6.1.4.1.476.1.1.1.1.6.3.10
.1.3.6.1.4.1.476.1.1.1.1.6.3.1
.1.3.6.1.4.1.476.1.1.1.1.13.1
.1.3.6.1.4.1.476.1.1.1.1.6.3.12

autumnwalker, 2020-12-30:
@aquette if you can let me know what commands to run I'll do it.

aquette, 2020-12-30:
It's just about opening the driver output and do a grep with the above OIDs. I just got a phone underhand... vacation time 😁

autumnwalker, 2021-01-07:
@aquette I think this is what you're looking for. 

   1.632402	[D1] get_and_process_data: ups.status (.1.3.6.1.4.1.476.1.1.1.1.6.3.10)
   1.632417	[D2] su_ups_get: ups.status .1.3.6.1.4.1.476.1.1.1.1.6.3.10
   1.632432	[D3] Entering nut_snmp_get_int()
   1.632447	[D3] nut_snmp_get(.1.3.6.1.4.1.476.1.1.1.1.6.3.10)
   1.632462	[D3] nut_snmp_walk(.1.3.6.1.4.1.476.1.1.1.1.6.3.10)
   1.710993	[D2] [test] nut_snmp_walk: .1.3.6.1.4.1.476.1.1.1.1.6.3.10: (noSuchName) There is no such variable name in this MIB.
   1.711025	[D2] => Failed

   1.711041	[D1] get_and_process_data: ups.status (.1.3.6.1.4.1.476.1.1.1.1.6.3.1)
   1.711056	[D2] su_ups_get: ups.status .1.3.6.1.4.1.476.1.1.1.1.6.3.1
   1.711071	[D3] Entering nut_snmp_get_int()
   1.711086	[D3] nut_snmp_get(.1.3.6.1.4.1.476.1.1.1.1.6.3.1)
   1.711101	[D3] nut_snmp_walk(.1.3.6.1.4.1.476.1.1.1.1.6.3.1)
   1.790134	[D2] [test] nut_snmp_walk: .1.3.6.1.4.1.476.1.1.1.1.6.3.1: (noSuchName) There is no such variable name in this MIB.
   1.790165	[D2] => Failed

   1.790182	[D1] get_and_process_data: ups.status (.1.3.6.1.4.1.476.1.1.1.1.13.1)
   1.790197	[D2] su_ups_get: ups.status .1.3.6.1.4.1.476.1.1.1.1.13.1
   1.790212	[D3] Entering nut_snmp_get_int()
   1.790227	[D3] nut_snmp_get(.1.3.6.1.4.1.476.1.1.1.1.13.1)
   1.790242	[D3] nut_snmp_walk(.1.3.6.1.4.1.476.1.1.1.1.13.1)
   1.874189	[D2] [test] nut_snmp_walk: .1.3.6.1.4.1.476.1.1.1.1.13.1: (noSuchName) There is no such variable name in this MIB.
   1.874220	[D2] => Failed

   1.874236	[D1] get_and_process_data: ups.status (.1.3.6.1.4.1.476.1.1.1.1.6.3.12)
   1.874252	[D2] su_ups_get: ups.status .1.3.6.1.4.1.476.1.1.1.1.6.3.12
   1.874267	[D3] Entering nut_snmp_get_int()
   1.874284	[D3] nut_snmp_get(.1.3.6.1.4.1.476.1.1.1.1.6.3.12)
   1.874299	[D3] nut_snmp_walk(.1.3.6.1.4.1.476.1.1.1.1.6.3.12)
   2.954250	[D2] [test] nut_snmp_walk: .1.3.6.1.4.1.476.1.1.1.1.6.3.12: (noSuchName) There is no such variable name in this MIB.
   2.954283	[D2] => Failed

rkage, 2021-04-12:
Hi - what's the latest on this? I can continue validation of the driver if necessary.

autumnwalker, 2021-10-01:
Revisiting this one - I've been trying to hobble along with the Liebert Multilink software, but it doesn't support headless Linux environments.

Really hoping we can get this driver across the line. Anything else I can do to provide data here?

tastewar, 2025-01-07:
Oh boy. I've been doing research about how to add my (pretty ancient at this point) Liebert UPS into my Home Assistant installation via NUT and came upon this thread. Seems like support for the SNMP card isn't there yet, and not much interest. Is that a fair assessment?

[2 more replies]