networkupstools/nut #420: Support PiModules UPS PIco HV3.0A for Raspberry Pi A+/B+/2B/3B and Zero (W)

renne, 2017-04-09:
Please add support for the [PiModules UPS PIco HV3.0A](http://pimodules.com/index-50.html).

It needs triggering the watchdog with a [train pulse on GPIO 22/27](https://github.com/modmypi/PiModules/blob/master/code/python/upspico/picofssd/scripts/picofssd) via sysfs GPIO and returns power state via the GPIOs. Further settings and information can be obtained via [I²C registers](https://www.modmypi.com/download/pimodules/_28_03_2017-register-list.docx).

jimklimov, 2023-04-17:
Sounds like a fun project, but unfortunately needs someone with hardware to step up and implement and test it. So far nobody picked it up, and the links in this issue went defunct...

jimklimov, 2023-04-17:
That above said, now there is more in-tree precedent about both GPIO and I2C drivers, including a PiJuice HAT driver added by #730 - at least to look and be inspired by for new work.