networkupstools/nut #424: [PATCH] Detect correct systemdsystemshutdowndir

clepple, 2017-04-19:
from [Stanislav Brabec on nut-upsdev](http://lists.alioth.debian.org/pipermail/nut-upsdev/2017-April/007268.html):

```
nut hardcodes systemdsystemshutdowndir to
${libdir}/systemd/system-shutdown. This is not valid:
- Executable helpers are installed to ${libexecdir}, which is different
  from ${libdir} on bi-arch systems.
- Even if nut will use ${libexecdir}, it could be different from
  systemd libexecdir.

That is why it is better to check systemd.pc for the correct path.
That path is stored as systemdutildir. In older systemd versions,
systemdutildir does not exist yet, and we need to use
systemdsystemunitdir can be used.

---
 configure.ac | 7 ++++++-
 1 file changed, 6 insertions(+), 1 deletion(-)

diff --git a/configure.ac b/configure.ac
index 6f8a40c4..f9a298d9 100644
--- a/configure.ac
+++ b/configure.ac
@@ -1197,7 +1197,12 @@ dnl Override installation directory, with the local installation
 dnl prefix. This is needed for 'distcheck*' targets, otherwise
 dnl files will try to get intalled to the actual system directories
 if test -n "${systemdsystemunitdir}"; then
-	systemdsystemshutdowndir="${libdir}/systemd/system-shutdown"
+	systemdutildir=`$PKG_CONFIG --variable=systemdutildir systemd`
+	if test -n "${systemdutildir}"; then
+		systemdsystemshutdowndir="${systemdutildir}/system-shutdown"
+	else
+		systemdsystemshutdowndir="${systemdsystemunitdir}/system-shutdown"
+	fi
 	AC_MSG_RESULT(using ${systemdsystemunitdir})
 else
 	AC_MSG_RESULT(no)
-- 
2.12.2

-- 
Best Regards / S pozdravem,

Stanislav Brabec
software developer
```

jimklimov, 2017-04-20:
Seems reasonable

clepple, 2017-04-20:
pushed branch systemdsystemshutdowndir for CI testing (using first commit from @stanislav-brabec in #425)

clepple, 2017-04-21:
Unfortunately, our Buildbot server is down, but it looks like it failed on Ubuntu wily+xenial and on Debian jessie. "make distcheck" seems to try to install directly to /lib instead of the staging directory:

 /bin/mkdir -p '/lib/systemd/system-shutdown'
 /usr/bin/install -c nutshutdown '/lib/systemd/system-shutdown'
/usr/bin/install: cannot create regular file '/lib/systemd/system-shutdown/nutshutdown': Permission denied
Makefile:426: recipe for target 'install-systemdsystemshutdownSCRIPTS' failed
make[5]: Leaving directory '/data/buildbot/slave/Ubuntu-xenial-x64/build/nut-v2.7.4-372-g7caa1612.7.4.1/_build/sub/scripts/systemd'
make[5]: *** [install-systemdsystemshutdownSCRIPTS] Error 1
Makefile:537: recipe for target 'install-am' failed
make[4]: Leaving directory '/data/buildbot/slave/Ubuntu-xenial-x64/build/nut-v2.7.4-372-g7caa1612.7.4.1/_build/sub/scripts/systemd'
make[4]: *** [install-am] Error 2
Makefile:471: recipe for target 'install-recursive' failed
make[3]: Leaving directory '/data/buildbot/slave/Ubuntu-xenial-x64/build/nut-v2.7.4-372-g7caa1612.7.4.1/_build/sub/scripts'
make[3]: *** [install-recursive] Error 1
Makefile:533: recipe for target 'install-recursive' failed
make[2]: Leaving directory '/data/buildbot/slave/Ubuntu-xenial-x64/build/nut-v2.7.4-372-g7caa1612.7.4.1/_build/sub'
make[2]: *** [install-recursive] Error 1
Makefile:739: recipe for target 'distcheck' failed
make[1]: Leaving directory '/data/buildbot/slave/Ubuntu-xenial-x64/build'
make[1]: *** [distcheck] Error 1
Makefile:936: recipe for target 'distcheck-light' failed
make: *** [distcheck-light] Error 2

jimklimov, 2017-04-21:
I have a "gut feeling" that the previous use of `libdir` ensured installation into `$(DESTDIR)@libdir@` somewhere in autotools... At least I do not quickly see anything special about it in NUT sources.

jimklimov, 2017-04-21:
At least can confirm the report from @clepple - for my local debian checkout of the branch, a `make distcheck` failed similarly, trying to write into OS `/lib/...` paths.

jimklimov, 2017-04-21:
And this is where it gets strange :)

Going into that `dist`ed build directory, and running a `make install DESTDIR=...` it installs `nutshutdown` script as expected, BUT then reinstalls the service unit files into `$(DESTDIR)$(DESTDIR)/lib/system/system` (prepends the passed value of DESTDIR to the one used by the build).

Without an explicit DESTDIR in the arguments, it fails to install `nutshutdown` into the OS path, but puts the unit files where they should be - `$(DESTDIR)/lib/system/system`.

Note that the expanded verbatim value of `dist`-used `DESTDIR` is already present in the generated `Makefile`, which explains the prepending and apparent duplication of the value I passed to `make` (for the unit files). 

And in the generated Makefile, here's the difference (confirming the "gut feeling" above and one of few evils of autotools):

````
systemdsystemshutdowndir = /lib/systemd/system-shutdown
systemdsystemunitdir = ${prefix}/lib/systemd/system
````

For units that work, the variable is expanded later on. For script that does not work, it is already verbatim (and was an expandable `${libdir}` earlier) :\

jimklimov, 2017-04-21:
I looked at `auglensdir` support as a similar case, where the (default) path is effectively hardcoded by the `configure.ac` script. In the corresponding `Makefile` it already has the curly-braced `${prefix}` instead of the incoming `/usr` - so I guess autotools took care of mapping it somehow. And since the OS `/lib` is not among paths that autotools cares about, it is not mapped.

But so far it is guesswork and speculation on my side :)

jimklimov, 2017-04-21:
Ok, and further drilling into recipes - the `Makefile.am` defines certain settings for `distcheck` to use among `DISTCHECK_CONFIGURE_FLAGS` including (and incorrectly a bit) the `--with-augeas-lenses-dir='$${prefix}/usr/share/augeas/lenses'` piece.

I think the systemdshutdowndir just warrants a complete separate variable and toggle similar to systemdunitdr support :)

clepple, 2017-11-26:
Closing since #426 was merged to master; please re-open if the original issue was not fixed.

closed 2017-11-26 by clepple