networkupstools/nut #429: openssl 1.1 support

The-Compiler, 2017-04-28:
Running `./configure --with-openssl` with openssl 1.1 (which [is backwards-incompatible](https://www.openssl.org/news/openssl-1.1.0-notes.html)) shows:

```
checking for OpenSSL version via pkg-config... 1.1.0e found
checking for OpenSSL cflags... 
checking for OpenSSL ldflags... -lssl -lcrypto
checking for openssl/ssl.h... yes
checking for SSL_library_init... no
configure: error: "OpenSSL not found, required for SSL support"
```

See the [full config.log](http://paste.the-compiler.org/view/a30b62bb) for more details.

jimklimov, 2017-05-01:
I wonder if as a short-term solution, we an solicit the old API (whether this works), e.g. from those changelog notes :

````
* Deprecated interfaces can now be disabled at build time either relative to the latest release via the "no-deprecated" Configure argument, or via the "--api=1.1.0|1.0.0|0.9.8" option.
* Application software can be compiled with -DOPENSSL_API_COMPAT=version to ensure that features deprecated in that version are not exposed.
````

The-Compiler, 2017-05-09:
That sounds like you can opt out of stuff, but probably you won't be able to opt-in to things which were already dropped when building OpenSSL.

clepple, 2017-11-28:
@The-Compiler although I haven't had a chance to check, it seems like #504 should help.

jimklimov, 2020-03-23:
+1 : See also #504. In fact, does it resolve this issue?..

The-Compiler, 2020-04-09:
Sorry for the (very) late answer - I can confirm #504 fixes this. Thanks everyone! Any chance for a new release (v2.7.4 has been released four years ago)?

closed 2020-04-09 by The-Compiler