networkupstools/nut #437: CPS: "battery.mfr.date" value is bogus

clepple, 2017-06-04:
For Cyber Power Systems UPSes, `battery.mfr.date` is pulled from `UPS.PowerSummary.iOEMInformation`, and that is usually either `CPS` or `Cyber Power Systems` (according to the DDL entries we currently have).

http://lists.alioth.debian.org/pipermail/nut-upsuser/2017-May/010662.html among others.

alexruffell, 2021-10-30:
This bug has been open since 2017... the problem persists as my CyberPower PR1000RTXL2U also shows:

`battery.mfr.date: CPS`

flashydave, 2021-11-13:
Also the same on CP900EPFCLCD
`battery.mfr.date: CPS`

Brunty, 2022-09-01:
It's the same on my BR1000ELCD

jimklimov, 2022-09-02:
Looking at the spec https://www.usb.org/sites/default/files/pdcv11.pdf (page 38) it seems there is a field for what NUT would know as `"UPS.Battery.ManufacturerDate"` (according to the few precedents in NUT sources):

> 4.2.6 Battery Settings
> ManufacturerDate The date the pack was manufactured in a packed integer. The date is packed in the following fashion: (year – 1980)*512 + month*32 + day.

Curiously, some drivers recognize further data like UPS Manufacture date (per PowerSummary), or battery replacement date (though probably a typo in `powercom-hid.c`), which is not in that spec. As well as other drivers (`openups`) that use `UPS.PowerSummary.iOEMInformation` for `battery.mfr.date`.

I'll post a PR to prefer mapping that value if available - hoping you would be able to build NUT from that branch to see if it really works ;)

jimklimov, 2022-09-05:
@Brunty ? @flashydave ? @alexruffell ? Can someone of you please build https://github.com/jimklimov/nut/tree/issue-437 and run `./usbhid-ups -a yourupsname -d1` (with main driver service stopped) to see if it picks up a reasonable date value?

alexruffell, 2022-09-05:
@jimklimov unfortunately I do not know how to do that

flashydave, 2022-09-05:
I will try but I dont think the alt fields have much that is useful on CPS UPS'es.
lets see...

flashydave, 2022-09-06:
I can confirm there is no change with the new compiled driver on CPS UPS.

battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 4830
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CP900EPFCLCD
device.serial: 000000000000
device.type: ups
driver.flag.pollonly: enabled
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.8.0-500-gefa9cff
driver.version.data: CyberPower HID 0.8
driver.version.internal: 0.49
driver.version.usb: libusb-1.0.19 (API: 0x1000103)
input.transfer.high: 260
input.transfer.low: 170
input.voltage: 244.0
input.voltage.nominal: 230
output.voltage: 244.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 5
ups.mfr: CPS
ups.model: CP900EPFCLCD
ups.productid: 0501
ups.realpower.nominal: 540
ups.serial: 000000000000
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764

regards
Dave

jimklimov, 2022-09-06:
@alexruffell : For a quick test, without installing the custom build into your OS, it may suffice that you:

````
:; git clone -b issue-437 https://github.com/jimklimov/nut
:; cd nut
:; ./ci_build.sh
# ...wait some time for build/test to complete...

# Stop existing NUT driver
# Get a data dump from new driver build:
:; ./drivers/usbhid-ups -a yourupsname -d1
# ...or if it complains about access:
:; sudo ./drivers/usbhid-ups -a yourupsname -d1 -x user=root
````

To perform the build, you may need to install tools and/or prerequisites (notably libusb-devel) as listed in `docs/config-prereqs.txt` for your OS type.

jimklimov, 2022-09-06:
@flashydave : thanks! Can you also run the driver with debug (IIRC level 2 would suffice, `-DD`) to print out the data it sees in the Report Descriptor for your device?

In a different issue #1512 and short how-to in #1618 there was a troubleshooting session for another CPS issue with `output.voltage` (curiously yours seems to be in the okay ballpark), where the HEX printout of the USB HID Report Descriptor (debug level 3) is parsed and interpreted by tools in detail. In that discussion, the device also reports some non-standard values like `UPS.ff010004.ff010024.ff0100d0` which the driver (nor parsers) apparently do not know (yet) how to interpret. Maybe one of those hides the battery date?..

flashydave, 2022-09-06:
As below:

   0.000000	[D3] do_global_args: var='maxretry' val='3'
   0.011514	[D3] do_global_args: var='pollinternal' val='10'
   0.015897	[D3] main_arg: var='driver' val='usbhid-ups'
   0.018885	[D3] main_arg: var='port' val='auto'
   0.022760	[D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.023155	[D3] main_arg: var='desc' val='CP900EPFCLCD'
   0.027552	[D3] main_arg: var='pollonly' val='<null>'
   0.034728	[D5] send_to_all: SETINFO driver.flag.pollonly "enabled"
   0.038958	[D1] debug level is '5'
   0.070122	[D5] send_to_all: SETINFO device.type "ups"
   0.077544	[D2] Initializing an USB-connected UPS with library libusb-1.0.19 (API: 0x1000103) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.081160	[D1] upsdrv_initups (non-SHUT)...
   0.179161	[D2] Checking device 1 of 2 (0764/0501)
   0.209856	[D2] - VendorID: 0764
   0.215905	[D2] - ProductID: 0501
   0.218792	[D2] - Manufacturer: CPS
   0.224220	[D2] - Product: CP900EPFCLCD
   0.226054	[D2] - Serial Number: 000000000000
   0.226518	[D2] - Bus: 001
   0.226934	[D2] - Device: unknown
   0.227168	[D2] - Device release number: 0001
   0.227756	[D2] Trying to match device
   0.237472	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.237942	[D3] match_function_regex: matching a device...
   0.238203	[D2] Device matches
   0.238838	[D2] Reading first configuration descriptor
   0.239232	[D3] libusb_kernel_driver_active() returned 0
   0.255002	[D2] Claimed interface 0 successfully
   0.255718	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.259993	[D3] HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 b9 01
   0.262754	[D3] HID descriptor length (method 1) 441
   0.263015	[D4] i=0, extra[i]=09, extra[i+1]=21
   0.263678	[D3] HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 b9 01
   0.264006	[D3] HID descriptor length (method 2) 441
   0.264243	[D2] HID descriptor length 441
   0.278770	[D2] Report Descriptor size = 441
   0.284516	[D3] Report Descriptor: (441 bytes) => 05 84 09 04 a1 01 09 24 a1 00 85 01 09 fe
   0.286764	[D3]  75 08 95 01 15 00 26 ff 00 b1 23 85 02 09 ff b1 23 85 1b 06 01 ff 09 d0 b1
   0.287246	[D3]  22 85 03 05 85 09 89 b1 23 85 04 09 8f b1 23 85 05 09 8b b1 23 85 06 09 2c
   0.295028	[D3]  b1 23 85 07 75 08 95 06 25 64 09 83 09 8d 09 8e 09 8c 09 29 09 67 b1 23 85
   0.295929	[D3]  08 75 08 95 01 65 00 09 66 81 23 09 66 b1 a3 09 68 75 10 27 ff ff 00 00 66
   0.301513	[D3]  01 10 81 23 09 68 b1 a3 09 2a 26 58 02 81 23 09 2a b1 a2 85 09 75 08 26 ff
   0.305839	[D3]  00 05 84 09 40 67 21 d1 f0 00 55 06 b1 23 85 0a 09 30 b1 a3 09 02 a1 02 65
   0.312838	[D3]  00 55 00 85 0b 75 01 95 06 25 01 05 85 09 d0 09 44 09 45 09 42 09 46 09 43
   0.315837	[D3]  81 23 09 d0 09 44 09 45 09 42 09 46 09 43 b1 a3 75 02 95 01 81 01 b1 01 c0
   0.318833	[D3]  85 0c 05 84 09 5a 75 08 15 01 25 03 b1 a2 09 5a 81 22 85 0d 09 fd 15 00 26
   0.322967	[D3]  ff 00 b1 23 c0 05 84 09 1a a1 00 85 0e 05 84 09 40 75 08 67 21 d1 f0 00 55
   0.323775	[D3]  07 b1 23 85 0f 75 10 09 30 b1 a3 85 10 09 53 16 aa 00 26 b4 00 b1 a2 09 53
   0.324254	[D3]  81 23 09 54 16 04 01 26 0e 01 b1 a2 09 54 81 23 c0 09 1c a1 00 85 12 09 30
   0.333219	[D3]  b1 a3 75 08 15 00 26 ff 00 65 00 55 00 85 13 09 35 b1 a3 85 14 09 58 25 06
   0.334068	[D3]  b1 a2 09 58 81 22 85 15 09 57 75 10 15 ff 26 ff 7f 35 c4 47 c4 ff 1d 00 66
   0.342038	[D3]  01 10 b1 a2 85 16 09 56 b1 a2 85 17 09 6e 75 01 15 00 25 01 35 00 45 00 65
   0.345210	[D3]  00 b1 a3 09 65 b1 a3 75 06 b1 01 85 18 75 10 09 44 26 84 03 66 21 d1 55 07
   0.351831	[D3]  b1 a3 85 1a 06 01 ff 65 00 55 00 75 08 15 00 25 02 09 43 b1 a2 09 43 81 23
   0.354863	[D3]  c0 c0
   0.362463	Using subdriver: CyberPower HID 0.8
   0.366770	[D3] Attempting Report Descriptor fix for UPS: Vendor: 0764, Product: 0501
   0.372723	[D4] Report Descriptor: hvt input LogMin: 260 LogMax: 270
   0.374254	[D4] Report Descriptor: output LogMin: 260 LogMax: 270
   0.374858	[D3] Fixing Report Descriptor. Set Output Voltage LogMin = 0, LogMax = 511
   0.375099	[D4] Report Descriptor: input LogMin: 0 LogMax: 255
   0.382931	[D3] Fixing Report Descriptor. Set Input Voltage LogMin = 0, LogMax = 511
   0.383466	[D2] Report Descriptor Fixed
   0.383860	[D1] 53 HID objects found
   0.384142	[D4] Entering libusb_get_report
   0.394083	[D3] Report[get]: (2 bytes) => 01 01
   0.394773	[D5] PhyMax = 0, PhyMin = 0, LogMax = 255, LogMin = 0
   0.395027	[D5] Unit = 00000000, UnitExp = 0
[... 1008 lines cut ...]
   2.928647	[D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
   2.933752	[D5] Unit = 00000000, UnitExp = 0
   2.934090	[D5] Exponent = 0
   2.937563	[D2] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x0b, Offset: 0, Size: 1, Value: 1
   2.937914	[D5] hu_find_infoval: found online (value: 1)
   2.938145	[D5] process_boolean_info: online
   2.938759	[D3] Report[buf]: (2 bytes) => 0b 11
   2.939009	[D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
   2.948769	[D5] Unit = 00000000, UnitExp = 0
   2.951168	[D5] Exponent = 0
   2.954619	[D2] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x0b, Offset: 1, Size: 1, Value: 0
   2.954956	[D5] hu_find_infoval: found !chrg (value: 0)
   2.955185	[D5] process_boolean_info: !chrg
   2.961011	[D3] Report[buf]: (2 bytes) => 0b 11
   2.961609	[D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
   2.961884	[D5] Unit = 00000000, UnitExp = 0
   2.962103	[D5] Exponent = 0
   2.962762	[D2] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x0b, Offset: 2, Size: 1, Value: 0
   2.962998	[D5] hu_find_infoval: found !dischrg (value: 0)
   2.975164	[D5] process_boolean_info: !dischrg
   2.975849	[D3] Report[buf]: (2 bytes) => 0b 11
   2.978808	[D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
   2.979095	[D5] Unit = 00000000, UnitExp = 0
   2.979715	[D5] Exponent = 0
   2.980005	[D2] Path: UPS.PowerSummary.Present

alexruffell, 2022-09-06:
@jimklimov Thank you for the instructions. I cloned my VM as I would not know how to revert back, and followed your instructions. I got to the point of getting the dump but it complains about there not being and file called usbhid-hid

```
./drivers/usbhid-ups -a cyberpower_rack_ups -d1
-bash: ./drivers/usbhid-ups: No such file or directory
alex@ubuntu-server-nut:~/nut$ ls ./drivers/usb*
./drivers/usb-common.c  ./drivers/usb-common.h  ./drivers/usbhid-ups.c  ./drivers/usbhid-ups.h
alex@ubuntu-server-nut:~/nut$
```

jimklimov, 2022-09-06:
@alexruffell : that looks like it did not actually succeed configuring or compiling NUT when you ran `./ci_build.sh`

jimklimov, 2022-09-06:
Also, just to not mix threads - here we try to figure out if dates are served in CPS USB HID payload, not the voltages (there's e.g. #1512 for that). There are several "vendor-extension" usages (not mapped from HEX to names) in the report above; I did not find what they may mean. Maybe they carry additional data, maybe not.

Perhaps some USB sniffer to catch traffic from their Windows programs (if any) can help make sense of this data, or asking the company by mail for a spec (and whether it may be published)...

jimklimov, 2022-09-09:
Regarding the original issue, with `battery.mfr.date`, one big question is if the UPS serves it at all.

As the mailing list thread linked from original post suggests, "exploring" the HID UPS device with the `usbhid-ups` driver might help. See also https://networkupstools.org/docs/developer-guide.chunked/ar01s04.html#_writing_a_subdriver suggestions and a bit later about "Updating a subdriver" to see what reports your device model actually gives which might not be covered by existing mappings.

Same discussion actually also touches on bogus `output.voltage` (e.g. `input.voltage: 120.0` vs `output.voltage: 264.0` in NUT v2.7.4 based report above) - this is hopefully fixed in current NUT, with #439 having been resolved.

flashydave, 2022-09-09:
I dont think CPS UPS'es do.

alexruffell, 2022-09-11:
I was wondering that too as the batteries do not have any smarts so it would be impossible to know when they were manufactured. My UPS allows me to keep track of when the batteries were installed in the web interface but I don't see the date in the data I posted above. So, I wonder whether the mfg date is actually the install date, or the data point is simply missing.

jimklimov, 2022-09-20:
Well, I did the best I could for now for remote investigation, code-digging and improvement on at least some rough edges. If someone comes up with discoveries where the CPS devices report e.g. battery's last-replacement date via USB HID protocol (note there are unnamed report branches we did not look into), feel free and welcome to open a new ticket/PR and reference this one for context.

jimklimov, 2022-09-20:
@alexruffell : since your UPS has web-management access, did you check if `snmp-ups` handles it, as an alternative possibly suitable driver?

closed 2022-09-20 by jimklimov

Timi7007, 2025-03-21:
Hello everybody,

I'm running NUT 2.8.2 via the Unraid plugin and can confirm this issue with my Cyberpower VP1000ELCD UPS:
```battery.mfr.date: CPS```

For those still wanting to keep track, the override in ```/etc/nut/ups.conf``` works fine:
```override.battery.mfr.date = "2021-01-24"```

PR #1645 (2022-09-20): Fixes to "battery.mfr.date" (cps-hid, powercom-hid)
* CPS: If data is served at a better USB HID report, use that (for most devices it is not served however)
* Powercom: Use correct NUT item name for the `battery.mfr.date` (if known)

Closes: #437
Closes: #1644