networkupstools/nut #439: CPS: patch HID Report Descriptor to fix "output.voltage"

clepple, 2017-06-04:
Observed in issue https://github.com/networkupstools/nut/issues/337#issuecomment-257284089
and on [nut-upsuser](http://lists.alioth.debian.org/pipermail/nut-upsuser/2017-June/010685.html).

Report ID 18 seems to be inheriting Logical Minimum/Logical Maximum from Report ID 16:
```
Collection(Physical)
    Report ID(16)
        Usage(LowVoltageTransfer)
        Logical Minimum(97)
        Logical Maximum(103)
        Feature(Data,Variable,Absolute)

        Usage(LowVoltageTransfer)
        Input(Constant,Variable,Absolute)

        Usage(HighVoltageTransfer)
        Logical Minimum(136)
        Logical Maximum(142)
        Feature(Data,Variable,Absolute)

        Usage(HighVoltageTransfer)
        Input(Constant,Variable,Absolute)

    Report ID(18)
        Usage(Voltage)
        Feature(Constant,Variable,Absolute)

        Report Size(8)
        Logical Minimum(0)
        Logical Maximum(255)
        Unit(SI Linear: ...)
        Unit Exponent(0)
```

I'm pretty sure this is not a bug in the NUT HID parser, but feel free to correct me. (There are other places in the CPS HID Report Descriptor that seem to mix up the locations of the parser tokens relative to Input/Output/Feature.)

clepple, 2017-06-04:
Original HID Report Descriptor patcher issue is https://github.com/networkupstools/nut/issues/169

csurf, 2020-10-29:
what's the status of this issue? I'm experiencing the same thing for a CyberPower CP1000PFCLCD UPS. Both the output voltage & battery voltage are incorrect.

CorvetteCole, 2021-04-27:
I would also love to hear if anyone has looked in to this further

serfriz, 2021-04-28:
Any update on this? My CyberPower CP1500PFCLCD shows an output.voltage of 142.0

arlenarlenarlen, 2021-07-28:
👋🏼  having the same issue here with my CP1500PFCLCD
`Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: 136`

slyticoon, 2021-08-16:
Having this issue with CP1500AVRLCD. Any updates or workarounds?

ArcticZero, 2021-09-16:
Chiming in to say I'm experiencing the same with my CP1500EPFLCD. Output voltage shows a frightening 263.0-264.0v, but is actually 239v. This needs to be fixed.

wmigas, 2021-11-09:
I have the same issue, with my CP1300EPFCLCD
I was comparing some value, it looks like output.voltage = input.voltage+32 in my case
```input.voltage: 233.0
input.voltage.nominal: 230
output.voltage: 265.0
```

```input.voltage: 234.0
input.voltage.nominal: 230
output.voltage: 266.0
```

when I run ` /lib/nut/usbhid-ups -DDD -a cyberpower1500`
I get one value which  match 32. It is  Path: `UPS.ff0100ba.ff0100bc`
see `Path: UPS.ff0100ba.ff0100bc`, Type: Feature, ReportID: 0x1c, Offset: 16, Size: 8, Value: 32```
can some one check if value  reported by `Path: UPS.ff0100ba.ff0100bc` match  diff between display output voltage and value reported by NUT?

spootle, 2021-11-09:
@wmigas I have a CP1500EPFCLCD - it's not always input + 32, sometimes it drops to a hard 260v
[image]

wmigas, 2021-11-09:
@spootle thank you for confirming it. I will try to play with it and check usb data which are send from USB but as clepple has mentioned probably the limit is linked to the broken HID Report Descrptor

badrpc, 2021-11-10:
I have CP1500EPFCLCD and I don't know if this will help in any way but I have noticed that what currently see as the output voltage follows the formula (you can see it on the graph):

* input <= 238 => output = input + 32
* 238 < input < 245 => output = 260
* 245 <= input => output = input + 16

260 has also happens to match the high transfer point but that can also be just a coincidence.

[image]

badrpc, 2021-11-10:
Looking at the second graph on https://github.com/networkupstools/nut/issues/439#issuecomment-964563190 it seems that
* input < 229 results in output = 260
* 229 <= input <= 238 results in output = input + 32

badrpc, 2021-11-10:
I have no knowledge of USB HID devices. So please excuse my stupid question. Running `usbhid-ups -DDD -a core` I got this:
```
   0.099763     Report[get]: (3 bytes) => 0f f2 00
   0.099791     Path: UPS.Input.Voltage, Type: Feature, ReportID: 0x0f, Offset: 0, Size: 16, Value: 242

   0.105764     Report[get]: (3 bytes) => 12 f2 00
   0.105793     Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: 260
```

Shouldn't value for UPS.Output.Voltage be 242? 0x00f2 is 242. I'll try to cut the input power later today and will check what I see in UPS.Output.Voltage

wmigas, 2021-11-10:
@bardpc I'm not an expert and I started reading documentation couple days ago, but it look like, that HID Report Descriptor which describes data structure is broken. If you run your command with more "D" letters you will see more debug data like this
```   0.365925     Entering libusb_get_report
   0.368509     Report[get]: (3 bytes) => 12 e0 00
   0.368540     PhyMax = 0, PhyMin = 0, LogMax = 270, LogMin = 260
   0.368543     Unit = 00f0d121, UnitExp = 7
   0.368549     Exponent = 0
   0.368556     hid_lookup_path: 00840004 -> UPS
   0.368560     hid_lookup_path: 0084001c -> Output
   0.368566     hid_lookup_path: 00840030 -> Voltage
   0.368574     Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: 260
```
I get e0 which is 224, but I returns 260.
assuming that it adds to each reading 32, then you get 224+32= 256
but there is magic line `PhyMax = 0, PhyMin = 0, LogMax = 270, LogMin = 260` which set min and max value, that's why the lowest value is 260.

in case e5 which is 229
I get 261 which match 229+32 and is more than logmin value

flashydave, 2021-11-11:
Has anybody compared the results with Cyberpower's own Linux based software? 
Maybe that might help determine where the problem lies.
I have a CP900EFPCLCD arriving shortly so can give you another test result and maybe help with the debug effort.

flashydave, 2021-11-12:
I see the same problem with my CP900EPFCLCD running firmware `CRIB101#7F1 ` with self compiled nut driver.version: `2.7.4-2486-gaa0b3d1`(which is quite old).

```
   2.160392     [D3] Report[buf]: (3 bytes) => 12 f3 00
   2.160671     [D5] PhyMax = 0, PhyMin = 0, LogMax = 270, LogMin = 260
   2.160898     [D5] Unit = 00f0d121, UnitExp = 7
   2.161479     [D5] Exponent = 0
   2.161794     [D2] Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: 260
```

f3 is 243 which is what the front panel was showing but upsc is reporting 260 as above.
I also have a problem with the Battery voltage which is showing as 24v when it contains only 1 12v lead acid cell.  The serial # of the UPS is showing 000000000 as well.

flashydave, 2021-11-13:
Looking a little further (at least on my CP900EPFCLCD) the output voltage issue could be fixed by overriding the logic in libhid.c where the usbhid defined LogMax or LogMin limits are applied.  This would expose the value rather than masking it.

I guess a driver config option could be introduced to make it generally configurable. Is that an acceptable approach? If so I could write the code.

In my case the value is correct at but I would suggest external programs (like graphing applications) could apply any arithmetic fixes needed for other models.

Regarding the battery voltages reading 24v not 12v I havent seen any values change yet (a little suspicious given the 0.1v resolution*) but it does look like the UPS is simply mis-reporting the value via the usbhid protocol based on it's incorrect notion of the battery fitted. 

`realpower.nominal` is also a factor of 2 in error compared with the front panel display. 

Not a lot can be done to fix these other than applying the corrections externally.

*Does anyone else see `battery.voltage` values changing?

spootle, 2021-11-13:
I have seen mention elsewhere there are no battery voltage sensors installed in these units so it never changes (same reason pwrstat doesn't report any value)

flashydave, 2021-11-13:
I saw that but then wondered why they bother to render it on the unit LCD.  Internally they must measure the voltage to determine the state of charge (or discharge for updating the remaining run  time and declaring LB)

wmigas, 2021-12-21:
there was bug in hid parser which return incorrect value in my case 0xea (234) to 0x10A (266) I've tracked the issues, but it looks like it has been fixed in https://github.com/networkupstools/nut/pull/1040.
I've checked branched 2.4.7 instead of master :-)
current master branch return correct value 0x104 (260) which is logical min.

to fix output value, there are only 2 options:
1. force CPS to fix HID report descriptor
2. implement https://github.com/networkupstools/nut/issues/169

flashydave, 2021-12-23:
I have a fix as per your option 1 above which works with my CPS900EPFCLCD. I will be submitting this as a pull request very shortly.

wmigas, 2021-12-24:
I've made fix which works for me and I would like to create PR, but I was thinking about more general solution. I've asked cleepple about his thought, see #169. It is not full solution, but at list on the subdriver level we will be able to fix Report descriptor. see changes in my fork https://github.com/wmigas/nut/commit/8af66bdd23fb9aef5c6c99fddef632640dd8d9a2

flashydave, 2021-12-24:
Mine works as a general solution framework by creating another subdriver
function that can be called from the HID Parser  By doing this any
fix for a manufacturer can be coded and constrained in its own file
rather than polluting the common code with exceptions.  In the general
case a stub function is called which does nothing but return a false flag.

I tested it today and it also works with the libusb-1.0 branch too.
I just need to get it back from the target computer to one with an
internet connection so I can create the PR.  I might get this done later
today - if not it will be immediately after Christmas.

regards
Dave

flashydave, 2021-12-24:
See Pull Requests #1245 and #1246 which I have just created.
***@***.***:flashydave/nut.git

regards
Dave

5ft24dave, 2022-01-12:
I am seeing the same with my Cyberpower CPS1500AVR  output is 118v on UPS display, shows 142 in NUT

flashydave, 2022-01-12:
The fault I am fixing prevents the limits of LogMin and LogMax being
applied to limit the output voltage. I am not sure what these are set to
on a non EU (ie 110v) UPS as mine is a UK version.

Maybe you could run upsc and advise what the input.transfer.high and 
input.transfer.low voltages show. I suspect (hope?) one of the low
voltage is 142v.

Also I presume the Input Voltage is shown correctly at 118?

5ft24dave, 2022-01-12:
upsc isn't showing the transfer variables.

pi@hampi:/lib/nut $ upsc servers
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 3842
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 27.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CP1500AVRLCDa
device.serial: CXEJX2003678
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.voltage: 119.0
input.voltage.nominal: 120
output.voltage: 136.0
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 14
ups.mfr: CPS
ups.model: CP1500AVRLCDa
ups.productid: 0501
ups.realpower.nominal: 900
ups.serial: CXEJX2003678
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764

flashydave, 2022-01-12:
OK a little strange. 
Are you able to run the driver on its own with -DDD option and capture
the HID? e.g. 
./usbhid-ups -DDD -a cyberpower 2>&1 | tee /root/cyberpower.log

It will show in the output near the beginning with lines starting like this:
   0.161995     [D2] HID descriptor length 441
   0.165284     [D2] Report Descriptor size = 441
   0.172065     [D3] Report Descriptor: (441 bytes) => 05 84 09 04 a1 01 09 24 a1 00 85 01 09 fe
   0.173111     [D3]  75 08 95 01 15 00 26 ff 00 b1 23 85 02 09 ff b1 23 85 1b 06 01 ff 09 d0 b
....

Dave

5ft24dave, 2022-01-12:
now I see it.  

0.000000	debug level is '3'
   0.001949	upsdrv_initups...
   0.003586	Checking device (0BDA/9210) (002/002)
   0.003796	- VendorID: 0bda
   0.003829	- ProductID: 9210
   0.003851	- Manufacturer: unknown
   0.003872	- Product: unknown
   0.003894	- Serial Number: unknown
   0.003915	- Bus: 002
   0.003937	- Device release number: 2001
   0.003957	Trying to match device
   0.003989	Device does not match - skipping
   0.004120	Checking device (1D6B/0003) (002/001)
   0.004218	- VendorID: 1d6b
   0.004242	- ProductID: 0003
   0.004263	- Manufacturer: unknown
   0.004283	- Product: unknown
   0.004356	- Serial Number: unknown
   0.004383	- Bus: 002
   0.004442	- Device release number: 0510
   0.004463	Trying to match device
   0.004488	Device does not match - skipping
   0.004537	Checking device (0764/0501) (001/003)
   0.011208	- VendorID: 0764
   0.011252	- ProductID: 0501
   0.011274	- Manufacturer: CPS
   0.011300	- Product: CP1500AVRLCDa
   0.011322	- Serial Number: CXEJX2003678
   0.011344	- Bus: 001
   0.011365	- Device release number: 0200
   0.011400	Trying to match device
   0.011679	Device matches
   0.011790	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.013980	HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 fe 01
   0.014016	HID descriptor length (method 1) 510
   0.014048	HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 fe 01
   0.014075	HID descriptor length (method 2) 510
   0.014097	HID descriptor length 510
   0.018469	Report Descriptor size = 510
   0.018526	Report Descriptor: (510 bytes) => 05 84 09 04 a1 01 09 24 a1 00 85 01 09 fe
   0.018575	 75 08 95 01 15 00 26 ff 00 b1 23 85 02 09 ff b1 23 85 1b 06 01 ff 09 d0 b1
   0.018616	 22 85 03 05 85 09 89 b1 23 85 04 09 8f b1 23 85 05 09 8b b1 23 85 06 09 2c
   0.018671	 b1 23 85 07 75 08 95 06 25 64 09 83 09 8d 09 8e 09 8c 09 29 09 67 b1 23 85
   0.018712	 08 75 08 95 01 65 00 09 66 81 23 09 66 b1 a3 09 68 75 10 27 ff ff 00 00 66
   0.018765	 01 10 81 23 09 68 b1 a3 09 2a 26 58 02 81 23 09 2a b1 a2 85 09 75 10 26 e0
   0.018806	 01 05 84 09 40 67 21 d1 f0 00 55 06 b1 23 85 0a 09 30 b1 a3 09 02 a1 02 65
   0.018883	 00 55 00 85 0b 75 01 95 06 25 01 05 85 09 d0 09 44 09 45 09 42 09 46 09 43
   0.018924	 81 23 09 d0 09 44 09 45 09 42 09 46 09 43 b1 a3 75 02 95 01 81 01 b1 01 c0
   0.018978	 85 0c 05 84 09 5a 75 08 15 01 25 03 b1 a2 09 5a 81 22 85 0d 09 fd 15 00 26
   0.019019	 ff 00 b1 23 c0 05 84 09 1a a1 00 85 0e 05 84 09 40 75 08 67 21 d1 f0 00 55
   0.019147	 07 b1 23 85 0f 75 10 09 30 b1 a3 c0 09 1c a1 00 85 10 09 53 15 61 25 67 b1
   0.019188	 a2 09 53 81 23 09 54 16 88 00 26 8e 00 b1 a2 09 54 81 23 85 12 09 30 b1 a3
   0.019242	 75 08 15 00 26 ff 00 65 00 55 00 85 13 09 35 b1 a3 85 14 09 58 25 06 b1 a2
   0.019282	 09 58 81 22 85 15 09 57 75 10 15 ff 26 ff 7f 35 c4 47 c4 ff 1d 00 66 01 10
   0.019322	 b1 a2 85 16 09 56 b1 a2 85 17 09 6e 75 01 15 00 25 01 35 00 45 00 65 00 b1
   0.019377	 a3 09 65 b1 a3 75 06 b1 01 85 18 95 02 75 10 09 44 26 dc 05 66 21 d1 55 07
   0.019416	 09 43 b1 a3 85 19 95 01 09 34 26 84 03 b1 a3 09 34 81 23 85 1d 09 33 26 dc
   0.019457	 05 b1 a3 09 33 81 23 85 1a 06 01 ff 65 00 55 00 75 08 15 00 25 02 09 43 b1
   0.019502	 a2 09 43 81 23 c0 85 1c 09 ba a1 02 75 10 27 ff ff 00 00 09 bb b1 22 75 08
   0.019540	 26 ff 00 09 bc b1 22 75 10 27 ff ff 00 00 09 bd b2 a2 01 c0 c0
   0.019925	Using subdriver: CyberPower HID 0.4
   0.019960	61 HID objects found
   0.021222	Report[get]: (2 bytes) => 01 01
   0.021273	Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   0.021966	Report[get]: (2 bytes) => 02 02
   0.022013	Path: UPS.PowerSummary.iSerialNumber, Type: Feature, ReportID: 0x02, Offset: 0, Size: 8, Value: 2
   0.022718	Report[get]: (2 bytes) => 1b 05
   0.022766	Path: UPS.PowerSummary.ff0100d0, Type: Feature, ReportID: 0x1b, Offset: 0, Size: 8, Value: 5
[... 91 lines cut ...]
   0.049713	Path: UPS.Output.Boost, Type: Feature, ReportID: 0x17, Offset: 0, Size: 1, Value: 0
   0.049734	Report[buf]: (2 bytes) => 17 00
   0.049749	Path: UPS.Output.Overload, Type: Feature, ReportID: 0x17, Offset: 1, Size: 1, Value: 0
   0.050441	Report[get]: (5 bytes) => 18 84 03 dc 05
   0.050466	Path: UPS.Output.ConfigActivePower, Type: Feature, ReportID: 0x18, Offset: 0, Size: 16, Value: 900
   0.050487	Report[buf]: (5 bytes) => 18 84 03 dc 05
   0.050504	Path: UPS.Output.ConfigApparentPower, Type: Feature, ReportID: 0x18, Offset: 16, Size: 16, Value: 1500
   0.051189	Report[get]: (3 bytes) => 19 bd 00
   0.051214	Path: UPS.Output.ActivePower, Type: Feature, ReportID: 0x19, Offset: 0, Size: 16, Value: 189
   0.051234	Report[buf]: (3 bytes) => 19 bd 00
   0.051250	Path: UPS.Output.ActivePower, Type: Input, ReportID: 0x19, Offset: 0, Size: 16, Value: 189
   0.051963	Report[get]: (3 bytes) => 1d c2 00
   0.051988	Path: UPS.Output.ApparentPower, Type: Feature, ReportID: 0x1d, Offset: 0, Size: 16, Value: 194
   0.052033	Report[buf]: (3 bytes) => 1d c2 00
   0.052078	Path: UPS.Output.ApparentPower, Type: Input, ReportID: 0x1d, Offset: 0, Size: 16, Value: 194
   0.052689	Report[get]: (2 bytes) => 1a 01
   0.052715	Path: UPS.Output.ff010043, Type: Feature, ReportID: 0x1a, Offset: 0, Size: 8, Value: 1
   0.052759	Report[buf]: (2 bytes) => 1a 01
   0.052802	Path: UPS.Output.ff010043, Type: Input, ReportID: 0x1a, Offset: 0, Size: 8, Value: 1
   0.053441	Report[get]: (6 bytes) => 1c 00 00 00 00 00
   0.053466	Path: UPS.ff0100ba.ff0100bb, Type: Feature, ReportID: 0x1c, Offset: 0, Size: 16, Value: 0
   0.053489	Report[buf]: (6 bytes) => 1c 00 00 00 00 00
   0.053506	Path: UPS.ff0100ba.ff0100bc, Type: Feature, ReportID: 0x1c, Offset: 16, Size: 8, Value: 0
   0.053529	Report[buf]: (6 bytes) => 1c 00 00 00 00 00
   0.053581	Path: UPS.ff0100ba.ff0100bd, Type: Feature, ReportID: 0x1c, Offset: 24, Size: 16, Value: 0

flashydave, 2022-01-12:
Decoding your HID Report Descriptor shows:
HighVoltageTransfer settings:
     Logical Minimum (136),  Logical Maximum (142)
and as with my CP900EPFCLCD-UK model these are erroneously
applied as the limits for the output voltage.
This explains why your previous email with upsc output of 142v

You can see what real value your UPS should have returned using more
debug to show the processing a little more clearlt.

   1.397440     [D3] Report[get]: (3 bytes) => 12 ee 00                   
   1.397968     [D5] PhyMax = 0, PhyMin = 0, LogMax = 511, LogMin = 0     
   1.398480     [D5] Unit = 00f0d121, UnitExp = 7                         
   1.398886     [D5] Exponent = 0                                         
   1.408680     [D5] hid_lookup_path: 00840004 -> UPS                     
   1.409087     [D5] hid_lookup_path: 0084001c -> Output                  
   1.409759     [D5] hid_lookup_path: 00840030 -> Voltage                 
   1.410097     [D1] Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x
12, Offset: 0, Size: 16, Value: 238

This is my UPS with my fix using 0 and 511 as limits but the first line 12 ee 00 is report 18 with value 238 (in hex).

Yours is returning:
>    0.043970	Report[get]: (3 bytes) => 12 76 00
>    0.044016	Path: UPS.Output.Voltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 16, Value: 142
which is 118v (as per your input) and looks correct but due to tbe report descriptor is being reported as 142.  I would have expected it to be 136 so that is slightly strange unless the code that applies the limits is confused. 

I dont know if you are capable of compiling my code but I think it should fix it.

Dave

5ft24dave, 2022-01-13:
Cool. I'll give it a shot

cpg-dev-team, 2022-01-13:
@flashydave @5ft24dave 
> which is 118v (as per your input) and looks correct but due to tbe report descriptor is being reported as 142. I would have expected it to be 136 so that is slightly strange unless the code that applies the limits is confused.

there was bug in hid parser which has been fixed in #1040 and it is merged to the master. however it is not yet available in debian repository. 

@5ft24dave if you compile flashydave PR then both of issues should be fixed.

5ft24dave, 2022-01-14:
Compiled, but can't figure out how to use the paths the debian package installed to... and with my existing confs moved to the new etc directory, it starts then dies.  usbhid-ups says it cannot connect as the kernel will not release the driver... and this is running as root like the deb package does without issue

clepple, 2022-01-14:
> Compiled, but can't figure out how to use the paths the debian package installed to... 

@5ft24dave check here: https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu

> and with my existing confs moved to the new etc directory, it starts then dies. usbhid-ups says it cannot connect as the kernel will not release the driver... and this is running as root like the deb package does without issue

NUT drivers drop privileges to a non-root user if you start them as root, so the `./configure` line in the wiki link above includes a username that should match the `nut` user in the .deb udev rules. Alternatively, there is a `-u root` flag to prevent dropping privileges.

5ft24dave, 2022-01-14:
Cool  Thank you!

flashydave, 2022-01-14:
This works for Debian.

 ./configure --with-ssl=no --enable-static --with-statepath=/var/run/nut
 --with-altpidpath=/var/run/nut --with-drvpath=/lib/nut
 --with-pidpath=/var/run/nut --datadir=/usr/share/nut --with-user=nut
 --with-group=nut --with-doc=yes --prefix= --sysconfdir=/etc/nut

make
sudo make install

5ft24dave, 2022-01-14:
Cool. Will give that a shot when I get home. Appreciate it

⁣Get BlueMail for Android ​

5ft24dave, 2022-01-15:
everything compiles, but when I get to sudo make install, I get errors...

***@***.***:~/nut $ sudo make install
Making install in include
make[1]: Entering directory '/home/pi/nut/include'
NUT_VERSION: "Please do not run git as root, your regular user account is
enough :)

If you need to run git with root rights for some other application than
what comes preinstalled on this image you can remove this sanity check:

git

You might have to restart your login session after doing that."
test -f "nut_version.h" || cp "nut_version.h.tmp" "nut_version.h"
cmp -s "nut_version.h.tmp" "nut_version.h" || cp "nut_version.h.tmp"
"nut_version.h"
rm -f "nut_version.h.tmp"
make  install-am
make[2]: Entering directory '/home/pi/nut/include'
make[3]: Entering directory '/home/pi/nut/include'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/home/pi/nut/include'
make[2]: Leaving directory '/home/pi/nut/include'
make[1]: Leaving directory '/home/pi/nut/include'
Making install in common
make[1]: Entering directory '/home/pi/nut/common'
make  install-am
make[2]: Entering directory '/home/pi/nut/common'
depbase=`echo common.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/bash ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.
-I../include    -I../include -g -O2 -Wno-unknown-warning-option
-Wno-system-headers -Wall -Wextra -Wsign-compare -pedantic -Wno-error -MT
common.lo -MD -MP -MF $depbase.Tpo -c -o common.lo common.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../include -I../include -g -O2
-Wno-unknown-warning-option -Wno-system-headers -Wall -Wextra
-Wsign-compare -pedantic -Wno-error -MT common.lo -MD -MP -MF
.deps/common.Tpo -c common.c  -fPIC -DPIC -o .libs/common.o
In file included from common.c:35:
../include/nut_version.h:7:27: warning: missing terminating " character
 #define NUT_VERSION_MACRO "Please do not run git as root, your regular
user account is enough :)
                           ^
../include/nut_version.h:9:1: error: unknown type name ‘If’
 If you need to run git with root rights for some other application than
 ^~
../include/nut_version.h:9:8: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or
‘__attribute__’ before ‘need’
 If you need to run git with root rights for some other application than
        ^~~~
../include/nut_version.h:9:8: error: unknown type name ‘need’
../include/nut_version.h:14:63: warning: missing terminating " character
 You might have to restart your login session after doing that."
                                                               ^
../include/nut_version.h:14:63: error: missing terminating " character
../include/nut_version.h:7:27: error: missing terminating " character
 #define NUT_VERSION_MACRO "Please do not run git as root, your regular
user account is enough :)

 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../include/nut_version.h:7:27: note: in definition of macro
‘NUT_VERSION_MACRO’
 #define NUT_VERSION_MACRO "Please do not run git as root, your regular
user account is enough :)

 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: warning: unrecognized command line option ‘-Wno-unknown-warning-option’
make[2]: *** [Makefile:550: common.lo] Error 1
make[2]: Leaving directory '/home/pi/nut/common'
make[1]: *** [Makefile:654: install] Error 2
make[1]: Leaving directory '/home/pi/nut/common'
make: *** [Makefile:625: install-recursive] Error 1

5ft24dave, 2022-01-15:
Fixed it!
the error says missing terminating " character in /home/pi/nut/include/nut_version.h
The quote was there.  however, autogen.sh left it an incomplete line  Added a CR at the end of the line and it compiled and runs without issue!

5ft24dave, 2022-01-15:
well that worked under Debian Bullseye on the pi.  trying it on Debian buster, and I'm getting the same error, but when I edit the nut_version.h file, the contents are not correct, they are:

 $ cat nut_version.h
/* Autogenerated file. Do not change. */
/* This file was generated by "make". */
/* The version number is determined by Git source commit hash
 * and number of commits since the most recent Git tag (if not
 * building the newest tagged commit itself - then just the tag).
 */
#define NUT_VERSION_MACRO "Please do not run git as root, your regular user account is enough :)

If you need to run git with root rights for some other application than
what comes preinstalled on this image you can remove this sanity check:

git

You might have to restart your login session after doing that."

flashydave, 2022-01-15:
This looks like an unrelated error.
My nut_version.h is intact with closing 
quote and terminating linefeed on both Debian 8.11 and on ubuntu 21.10 systems.

flashydave, 2022-01-15:
The nut_version is created by some shell script in include/Makefile and
looks to be very sensitive to git output.  I recall rpi OS sudo rules
tended to encourage git users not to run as root so maybe that causing
that issue.  

There is no reason (other than good practice) not to run as
any user. I managed the git usage inside a company with 1500 programmers for many years (and am also one of git's contributing authors) and have not come across this as being an issue!

5ft24dave, 2022-01-15:
so here is the complete procedure I am using, all requirements already loaded etc.  I'm using the new gh github cli because I can't figure out how to do a pull request with the original git command.

`curl -fsSL https://cli.github.com/packages/githubcli-archive-keyring.gpg | sudo dd of=/usr/share/keyrings/githubcli-archive-keyring.gpg

sudo apt update

sudo apt install gh

gh auth login

gh repo clone networkupstools/nut

cd nut

gh pr checkout 1245

./autogen.sh

./configure --with-ssl=no --with-cgi --enable-static --with-statepath=/run/nut \
--with-altpidpath=/run/nut --with-drvpath=/usr/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut  \
--with-pidpath=/run/nut --datadir=/usr/share/nut --with-user=nut \
--with-group=nut --with-doc=yes --prefix= --sysconfdir=/etc/nut

make

sudo make install

`

Works perfectlly on the first pi4 running the latest PiOS based on Bullseye (After adding the newline in nut_version.h) Debian 11.2, Kernel version 5.10.90-v7l+

identical commands on the other pi4 running Previous PiOS based on Buster (deb 10.11 , Kernel version 5.10.90-v7l+)

Since both have the same kernel versions, after runnung sudo make install on the pi that worked, I tar'd up the nut folder end untar'd it on the one that wasn't working.  sudo make install fails at the same point again...
is it possible that the new gh CLI is a little different in  the format of some of the git files it leaves and the older utilities on the older piOS  choke on them? If it wasn't for the fact that there are 50ft between the 2 UPS's, I'd run them both to the pi runnning the latest PiOS.  the one that is failing is running one of my 3D printers, and the Octopi image I run is still based on the old PiOS.  they haven't released a new one yet built on Bullseye.

flashydave, 2022-01-15:
Response inline....

> so here is the complete procedure I am using, all requirements already loaded etc.  I'm using the new gh github cli because I can't figure out how to do a pull request with the original git command.

git clone
git checkout

> `curl -fsSL https://cli.github.com/packages/githubcli-archive-keyring.gpg | sudo dd of=/usr/share/keyrings/githubcli-archive-keyring.gpg
>
> sudo apt update
>
> sudo apt install gh
>
> gh auth login
>
> gh repo [...]

The key thing to diagnose is to extract the shell script from
include/Makefile and execute it (or part of it) to find out where it is
failing. 
Start with just

git describe --tags 2>/dev/null | sed -e 's/^v\([0-9]\)/\1/' -e 's,^.*/,,'

I suspect it may be failing nearthere although I havent worked put what
its doing in detail. 
> identical commands on the other pi4 running Previous PiOS based on Buster (deb 10.11 , Kernel version 5.10.90-v7l+)
>
> Since both have the same kernel versions, after runnung sudo make install on the [...]
I heard there were issues with graphics libraries.

5ft24dave, 2022-01-15:
executing git describe --tags 2>/dev/null | sed -e 's/^v\([0-9]\)/\1/' -e 's,^.*/,,' gives me
2.7.4-4245-g36181e16
and since it references package version is determined in AC_INIT in configure.ac, here is that definition"
**_AC_INIT([nut],[2.7.4.1],[https://github.com/networkupstools/nut/issues])_**

5ft24dave, 2022-01-15:
> executing git describe --tags 2>/dev/null | sed -e 's/^v([0-9])/\1/' -e 's,^.*/,,' gives me 2.7.4-4245-g36181e16 and since it references package version is determined in AC_INIT in configure.ac, here is that definition" **_AC_INIT([nut],[2.7.4.1],[https://github.com/networkupstools/nut/issues])_**

working version uses make V4.3  failing is make V4.2.1...

jimklimov, 2022-01-16:
Did you plain `make` first (as a user)? Just `sudo make install` should be elevated, it seems.

NUT is CI-tested on several dozen OS and toolkit combos, though never built there as root.

flashydave, 2022-01-16:
I will lwt 5ft24dave answer your question but I'm not
quite sure we know what we are chasing beyond a corrupted creation of
nut_version.h.

The include/Makefile doesnt do anything clever beyond running an
embedded set of shell commands along with git so unless there is an
issue with the execution environment (eg PATH or unexpected warnings) I cant see
what difference the user running make or the version of make would have
at that point of the build.

Looking at NUT_VERSION_MACRO in configure.ac might narrow down where the
error is occuring.

5ft24dave, 2022-01-16:
Yes.   Autogen.sh configure and make are all built as the pi user.  Make install  is the only one using sudo.  Rpios buster gives the error every time
Rpios bullseye installs without issue

⁣Get BlueMail for Android ​

flashydave, 2022-01-16:
Can I suggest we move the build problem discussion to a newly created issue as it is not related to the CyberPower UPS or the voltage reporting.  In this way we can move on and close this ticket whilst dealing with the build problem separately.  It also means others experiencing the issue will see it and might contribute

5ft24dave, 2022-01-16:
we can ignore the issue.  its evidently part of one of the other software packages associated with Octoprint that is causing the sudo make install to fail.  I just flashed the same rpi4 with a fresh install of only raspberrypiOS Bullseye, and it builds without any complaints.  Flashed it fresh with the Octopi bullseye nightly image and tried the build again and it has the issue.
Problem solved.

flashydave, 2022-01-16:
Incidentally with the offending section extracted as a separate Makefile and put in a git repo to exercise the path through the shell commands you are utilising I have tested it using Make 4.2.1 on two rpi's running buster and it works as expected on both.

flashydave, 2022-01-16:
That is good albeit a bit strange.  I cant think what octopi have done
to cause the issue. Nevertheless I hope you can rearrange your hardware
to monitor your UPSes ok or can at least build it on another pi.

I assume after all that my pull request fix actually solves your original problem?

5ft24dave, 2022-01-17:
yes, fixes it nicely.  What I did was install the deb packages for nut, then deleted the files from those and copied the files from the compiled nut to those directories since the sudo make install wouldn't do it for me.  works like a champ.
Now just waiting for MODBUS over USB for the APC Back-UPS I have in another room LOL

flashydave, 2022-01-17:
Good news!

Incidentally I needed to replace a failed Tripp Lite UPS but couldnt get
delivery into the UK except following months of Brexit and Covid related
delays. I settled on CyberPower rather than APC due to their better
linux support (at least they attempt to supply some client software),
the price for their pure sine wave models and the fact yo cant change
batteries on APC back-ups models.

Having said that I did get close but saw the problems people are having
getting measurements from APC with NUT.

Modbus over USB as you observe appears to be the long term solution for
this but has its own issues currently.  I just needed something that
worked without having to dive into another huge development project.

Sometimes life is too short....Having spent >50 years programming and
engineering mission critical real-time systems professionally its time
to hang up my soldering iron and keyboard and have  a rest!

badrpc, 2022-01-17:
Just another data point: I have built NUT from https://github.com/flashydave/nut/commits/issue439 (FreeBSD, RPi 3B) and reported values now match what I expect - output == input when UPS is online and output == nominal when on battery:
[image]

Thank you so much @flashydave !

closed 2022-01-19 by jimklimov

nesanakin, 2022-06-24:
I have a CP1500PFCLCD and I'm still having this issue. The fix in #1245 seems to only apply to the CP*EPFCLCD series (looks like the EU model?) with productID 0x0501. The US CP*PFCLCD series with productID 0x0601 still reports the output voltage as 136V all the time. Would that fix work for this model as well? Seem like it would just take modifying a single conditional if so.

5ft24dave, 2022-06-24:
Fix applies to the US versions as well.  I was having that issue. It also appears to be merged into the current code as well

⁣Get BlueMail for Android ​

rapirent, 2022-07-08:
I have same problem with my CP1000PFCLCD (vendor#: 0x0601) in 2.8.0, and its voltage output is wrong (it displays 135v and seldomly varies but not match the lcd panel)

rapirent, 2022-07-09:
> The product ID for cyberpower is 0x0764. Did you mean product ID? if so the fix will not be applied as it only addresses models with productID 0x0501. It may be the same issue. Please can you run the driver under debug and report back the logs during the HID parsing process?
> […](#)

ths for the reply! Should I start with another issues?
I ran the `lsusb` and `upsc`.  Could you provide the exact command you want me to run if you want a further infomations?
And I wanna correct details about my problem, it seems like it always displays "135v" for output.voltage.

the debug message: 
```sh
[root@localhost ~]# lsusb -v -s 001

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  bcdDevice            5.10
  iManufacturer           3 Linux 5.10.23-v8.1.1.el8 dwc_otg_hcd
  iProduct                2 DWC OTG Controller
  iSerial                 1 3f980000.usb
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
 ```
```sh
[root@localhost ~]# upsc -l
cyberups1000v
[root@localhost ~]# upsc cyberups1000v
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 1550
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 13.8
battery.voltage.nominal: 12
device.mfr: CPS
device.model: CP1000PFCLCDa
device.serial: CX3LT2000019
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 001
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.product: CP1000PFCLCDa
driver.parameter.productid: 0601
driver.parameter.serial: CX3LT2000019
driver.parameter.synchronous: auto
driver.parameter.vendor: CPS
driver.parameter.vendorid: 0764
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.23 (API: 0x1000107)
input.voltage: 114.0
input.voltage.nominal: 120
output.voltage: 135.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 23
ups.mfr: CPS
ups.model: CP1000PFCLCDa
ups.productid: 0601
ups.realpower.nominal: 600
ups.serial: CX3LT2000019
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

[8 more replies]

commit 87dcce19 (2022-01-19): Fix #439 Incorrect Output Voltage on CPS*EFPCLCD