networkupstools/nut #442: upsd refused to start if a UPS is missing

ikus060, 2017-06-10:
Hello,

I'm running nut 2.7.2 on Debian Jessie. I'm currently experiencing a problem with the upsd service. It's refused to startup. Looking at the logs it's seams to be related to the fact one of the UPS is not connected or is not responding. 

```
Jun 10 16:29:24 kase upsdrvctl[19099]: Network UPS Tools - Generic HID driver 0.38 (2.7.2)
Jun 10 16:29:24 kase upsdrvctl[19099]: USB communication driver 0.32
Jun 10 16:29:24 kase upsdrvctl[19099]: No matching HID UPS found
Jun 10 16:29:24 kase upsdrvctl[19099]: Driver failed to start (exit status=1)
Jun 10 16:29:30 kase upsdrvctl[19099]: Network UPS Tools - Generic HID driver 0.38 (2.7.2)
Jun 10 16:29:30 kase upsdrvctl[19099]: USB communication driver 0.32
Jun 10 16:29:30 kase upsdrvctl[19099]: No matching HID UPS found
Jun 10 16:29:30 kase upsdrvctl[19099]: Driver failed to start (exit status=1)
Jun 10 16:29:36 kase upsdrvctl[19099]: Network UPS Tools - Generic HID driver 0.38 (2.7.2)
Jun 10 16:29:36 kase upsdrvctl[19099]: USB communication driver 0.32
Jun 10 16:29:36 kase upsdrvctl[19099]: No matching HID UPS found
Jun 10 16:29:36 kase upsdrvctl[19099]: Driver failed to start (exit status=1)
Jun 10 16:29:39 kase upsdrvctl[19099]: data receiving error (0 instead of 16 bytes)
Jun 10 16:29:42 kase upsdrvctl[19099]: data receiving error (0 instead of 16 bytes)
Jun 10 16:29:45 kase upsdrvctl[19099]: data receiving error (0 instead of 16 bytes)
Jun 10 16:29:45 kase upsdrvctl[19099]: Network UPS Tools - UPS driver controller 2.7.2
```

The situation is known to us, we are aware the UPS is not responding (it's currently in maintenance). But it's should not prevent "upsd" to startup. Otherwise a single UPS disconnected out of 10 should not  block upsd startup.

Did I miss anything ?

jimklimov, 2017-06-11:
Newer versions (git master) addressed this or similar issue; some solutions
may still be in PR branches though. One is that indeed nut-server (upsd)
should not be denied to start if some drivers did not connect; this depends
on how people write their startup scripts or which ones they adopt from
samples (whether those scripts bail if upsdrvctl fails or start
regardless). Another is, for systemd wraps, to manage each nut-driver
instance separately and not have 10 drivers as a single service unit. Yet
another, not sure if merged to upstream or still in discussion, is an
option to not abort the driver upon initial connection failures, but to
keep retrying as the driver would if it lost the link during work
("connection stale" state).

Note that one size never fits all. Some deployments only have one UPS for
the PC, so if it is AWOL - the upsd also is not quite needed. Likewise, not
failing on initial error is good for semistable setups, but would be folly
for initial ones where you can't be certain you got the config and cabling
right, or when a computer roams like a laptop...

Jim

ikus060, 2017-06-12:
Ok, thanks for the info. I will try git master version.

closed 2017-06-12 by ikus060