networkupstools/nut #448: Install pre-generated man pages from tarball

clepple, 2017-06-25:
If asciidoc is not installed when configuring NUT from a `make dist` tarball, the included pre-generated man pages do not get installed.

Reference: https://lists.alioth.debian.org/pipermail/nut-upsuser/2017-June/010786.html

jimklimov, 2017-06-25:
Looking at the mailing list message, the `configure` arguments include `--with-doc` which assumes a `yes` value for all formats (IIRC, can't look into code tonight). So if tools are not available to build man, html, html-chunked and pdf - the explicitly requested action can not be fulfilled, so the configuration fails.

It may be possible to relax the check for "man" type of documents and indeed distribute pre-generated versions from the tarball (e.g. add another value for fallback, like `--with-doc=man=fallback == --with-doc=man`), but this would not be possible for html and pdf docs which we do not pre-package. Then it may also be possible to change default `--with-doc` to only process manpages and not all doc types (IF this is the case now), or automate the guesswork for building or skipping the other types. I think I did something like that, but not sure at the moment if it has already landed into `upstream/master`, nor if what I did fully implements the graceful "build or skip" for any doc types.

Finally note that specifically due to `make dist` constraints, the Makefile rules for manpages vs. other doc types are somewhat different (man's are required unless explicitly skipped e.g. to speed up CI builds).

clepple, 2017-06-25:
I will admit that I got distracted while looking into this before, so I do not have a good bound for when things broke relative to previous releases. I suspect that there may be some related unresolved pull requests around when I was on vacation earlier this year. If you run across them before I do, feel free to link them here.

Also, I am fine with documenting around the problem in the short term, if something like `--with-doc=man` would work for Wolfy's RPM spec file. However, we can also take the opportunity to untangle some of the options for future releases.

jimklimov, 2017-06-26:
Just in case, current state should be documented : https://github.com/networkupstools/nut/blob/master/docs/configure.txt#L89

In the unlikely but possible case that (subsequent) changes in code diverged functionally from the description here, e.g changed or added some behaviors, this divergence can be treated as a bug and likely I am to blame for that discrepancy, then ;)

jimklimov, 2017-06-26:
> If the "--with-doc" argument is passed without a list, or lists just "=yes" 
> or "=all", it enables all supported formats with a "=yes". 
> 
> An (explicit!) "--with-doc=auto" argument tries to enable all supported 
> formats with an "=auto". 

In this case, I guess the current behavior is valid as documented (requiring ability to build the docs if just the short argument `--with-doc` is passed, and failing if the ability is not there). It may be or not be the case that this interpretation differs from earlier releases, and should be fixed (e.g. to assume `--with-doc=auto`). The reasonable case IMHO by autotools traditions is that `--with-something` means explicit enable, `--without-something == --with-something=no` means explicit disable, and not mentioning the flag is undetermined - so can mean `auto` for this case.

Also I (or anyone else) should still look into `man=auto` behavior, and if it really does not fall back to `make install`ing the SCMed copies of manpages - that should be amended. Gotta make an environment without asciidoc, or doctor local checkout recipes (require larger asciidoc version?), to test that code path :)

jimklimov, 2017-06-26:
Solution in the works at PR #451

closed 2017-07-04 by clepple