networkupstools/nut #450: Please to help to add Salicru UPS model in device list

CindyZhan, 2017-06-26:
Please help to add Salicru UPS in device list. Basic information are as follows:
Device manuafacture: Salicru 
Device type: UPS
Model Name：SLC TWIN PRO2<=3KVA
Connection: USB
VID/PID: 0x2E66/0X0201
Protocol: HID ("usbhid-ups" driver)
USB class: HID power device
Shutdown sequence: in our lab, this UPS passes shutdown sequence test.

Any other information required , or UPS samples required, please let me know.
Thanks in advance.

clepple, 2017-06-26:
per http://networkupstools.org/stable-hcl.html#footnotes , please include `upsc`, `upscmd -l` and `upsrw` output.

CindyZhan, 2017-06-29:
Hello Clepple: 
Thanks for your reply. I have a question when I try to get upscmd -l  and upsrw output. This issue is: 
As 0x2E66 do not list in NUT currently, so I use my base model to test. NUT version is 2.6.4.
-->My UPS base model ---  VID/PID=**0x06DA**/0xFFFF--- I type upsc, UPS information is displayed --- Then I type upscmd -l and upsrw,  **nothing displayed ;** 
-->Same UPS , I **change VID to 0x0463** while all other H/W and F/W keeps the same ---I type upsc, UPS information is displayed (more information compared to VID=0x06DA) ---- I type upscmd -l and upsrw, **some command and read write parameter supported by UPS are displayed.** 

Can you please help to clarify , which VID/PID should I use when I send you the output? Thanks a lot.

clepple, 2017-07-02:
@aquette can you take a look at this? I think we might have overlapping VID:PID combinations in `mge-hid.c` and `liebert-hid.c`.

@CindyZhan could you please run the driver with the `-x explore -DD` options, as shown here? http://networkupstools.org/docs/developer-guide.chunked/ar01s04.html#_writing_a_subdriver (We might not need to write an entirely new subdriver, but it would be good to see all the data.)

CindyZhan, 2017-07-04:
[Debugdata_Salicru.txt](https://github.com/networkupstools/nut/files/1121336/Debugdata_Salicru.txt)
Hello Clepple:
Please find the output after I run the usbhid driver with -DD -x explore option. (UPS is in battery mode in my test). Any other information required, let me know.
To add one more question: Does NUT check HID path's RO/RW feature? In this UPS's descriptor, UPS.Flow[4].ConfigPercentLoad is read only, but I find "ups.load.high" is displayed in upsrw output. And in "usbhid -DD -x explore" output, I find NUT show nothing about "RO/RW" feature. Please suggest. Thanks.

CindyZhan, 2019-10-09:
Hello clepple:
I'm sorry but may I know the status of this request? I still do not find this product added into NUT supported list. What should I do to make it done? Thanks.

CindyZhan, 2019-10-09:
@clepple , I add new comment regarding this request. Please help to check.

dmmop, 2021-08-14:
Hello,

Are there any news for this model?

Kind regards

jimklimov, 2021-12-01:
Heads-up: with PR #1199 merged into NUT master branch, there should now be a new subdriver for Salicru devices "out of the box". Can you please check if it also works for your model?

jimklimov, 2021-12-01:
As for the huge delays with HCL/DDL maintenance, sorry about that - getting code built and tested across a lot of platforms stole the priority, so sadly not much attention was paid to figuring out the correct way for documentation updates. Thinking of it, an actionable how-to, maybe with a github PR template, could help a lot - so users can contribute not just text reports, but ready-to-merge PRs...

jimklimov, 2021-12-01:
Also, this issue seems related to #1142 so cross-linking to not lose in the haystack again

jimklimov, 2022-01-10:
Finally figured out how to move ahead on HCL/DDL updates, PRs posted that should close this issue.

closed 2022-01-19 by jimklimov

jimklimov, 2022-01-19:
Merged that driver.list and built-in handling for the new VID/PID; feel free to open a new issue (and then reference this one and the PR please) if something still does not fit.