networkupstools/nut #453: [HCL] Cyber Power RT650EI supported by powerpanel

ichtyp, 2017-07-16:
Hey folks,

i just wanted to connect my new UPS from Cyber Power, Model "RT650EI" to my FreeNAS 11.x. It worked after a series of errors with the following driver:

Cyber Power Systems ups 2 AE550 USB (usbhid-ups)

upsc output:
root@wopr:~ # upsc ups@wopr
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 2925
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 92.7
battery.voltage.nominal: 12
device.mfr: CPS
device.model: RT Series
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ugen1.3
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 254
input.transfer.low: 200
input.voltage: 232.0
input.voltage.nominal: 0
output.voltage: 232.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 20
ups.mfr: CPS
ups.model: RT Series
ups.productid: 0501
ups.realpower.nominal: 400
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: 15300
ups.timer.start: 15300
ups.vendorid: 0764

Manufacturer Link: http://www.cyberpowersystems.de/produkte/backup-usv-serien/rt-serie.html

Best
Daniel

clepple, 2017-07-16:
Looks like this is using the `usbhid-ups` driver, right? (IIRC, `powerpanel` was for serial devices)

What sort of errors did you get?

The `battery.voltage` and `battery.voltage.nominal` values don't match. Is this an 8x12V battery system?

Also, have you had a chance to check the output of `upscmd -l` and `upsrw` to see what commands and writable variables are supported?

ichtyp, 2017-07-16:
**upscmd -l output is:**
root@wopr:~ # upscmd -l ups
Instant commands supported on UPS [ups]:

beeper.disable - Description unavailable
beeper.enable - Description unavailable
beeper.mute - Description unavailable
beeper.off - Description unavailable
beeper.on - Description unavailable
load.off - Description unavailable
load.off.delay - Description unavailable
load.on - Description unavailable
load.on.delay - Description unavailable
shutdown.return - Description unavailable
shutdown.stayoff - Description unavailable
shutdown.stop - Description unavailable
test.battery.start.deep - Description unavailable
test.battery.start.quick - Description unavailable
test.battery.stop - Description unavailable

**upsrw output:** 
root@wopr:~ # upsrw ups
[battery.charge.low]
Description unavailable
Type: STRING
Maximum length: 10
Value: 10

[battery.runtime.low]
Description unavailable
Type: STRING
Maximum length: 10
Value: 300

[input.transfer.high]
Description unavailable
Type: STRING
Maximum length: 10
Value: 254

[input.transfer.low]
Description unavailable
Type: STRING
Maximum length: 10
Value: 200

[ups.delay.shutdown]
Description unavailable
Type: STRING
Maximum length: 10
Value: 20

[ups.delay.start]
Description unavailable
Type: STRING
Maximum length: 10
Value: 30

The powerpanel software works, if i connect the ups directly to my windows 10 machine. i can browse the command website via http://localhost:3052/agent/battery_backup_info then. 

The Batter is a "DJW12-9.0(12V9.0AH)" Type Mantenance free Sealed Lead-acid Battery from a never hear of "LEOCH Battery co Ltd." Manufacturer.

I tried to connect to the ups using one of the many "powerpanel" drivers from nut, but none of them connected properly. the only one working is the one i mentioned above. this seemes to be the generic driver?

clepple, 2017-07-17:
> 
> The Batter is a "DJW12-9.0(12V9.0AH)" Type Mantenance free Sealed Lead-acid Battery from a never hear of "LEOCH Battery co Ltd." Manufacturer.
> 
So just one 12V battery, or several in series?
> I tried to connect to the ups using one of the many "powerpanel" drivers from nut, but none of them connected properly. the only one working is the one i mentioned above. this seemes to be the generic driver?
> 
Sorry for the confusion - I meant that the NUT powerpanel driver is serial-only. I have no idea about the manufacturer software, which seems to work with USB devices as well.

When you say "many powerpanel drivers"... I only see one: http://networkupstools.org/docs/man/powerpanel.html <http://networkupstools.org/docs/man/powerpanel.html>

How does FreeNAS present the list of drivers? Is there any documentation, like links to the man pages? I'm trying to understand what we can change to make this clearer.

ichtyp, 2017-07-17:
It is a single 12V battery, replacable by the user. 

The FreeNAS selection List looks like this:
[image]
I highlighted the one I chose.

I think it would be helpful to add another entry like "Cyber Power Systems RT Series...", if this in in your hands. If you couldn't do it, i can ask the FreeNAS guys for help as well, if this is GUI related.

commit fdd757df (2022-04-25): driver.list.in: Add HCL info for CyberPower Systems RT650EI (USB)

Closes: #453
closed 2022-04-25 by jimklimov