networkupstools/nut #456: CP900EPFCLCD

hdatma, 2017-08-01:
Hello, 

After extensive search, we finally identified the UPS that meets our hardware requirements, but have now a software problem: how to read and possibly write its data from OpenBSD via NUT. 

This is the model: 
https://www.cyberpower.com/ww/en/product/sku/CP900EPFCLCD
The "E" stands for "European", I guess, that is DIN plugs instead of IEC. 

The NUT page includes two models that are close, but not identical to the above:
http://networkupstools.org/ddl/Cyber_Power_Systems/CP1000PFCLCD.html
http://networkupstools.org/ddl/Cyber_Power_Systems/CP850PFCLCD.html

Is the available driver is compatible with CP900EPFCLCD?
http://networkupstools.org/docs/man/usbhid-ups.html

hdatma, 2017-08-09:
We have the hardware to test. This is what we see from OpenBSD. 

```
>dmesg
[...]
uhidev0 at uhub0 port 3 configuration 1 interface 0 "CPS CP900EPFCLCD" rev 2.00/0.01 addr 2
uhidev0: iclass 3/0, 26 report ids
upd0 at uhidev0
```

```
>console
[...]
==> /var/log/messages <==
Aug  9 19:23:53 server /bsd: uhidev0 at uhub0 port 3 configuration 1 interface 0 "CPS CP900EPFCLCD" rev 2.00/0.01 addr 2
Aug  9 19:23:53 server /bsd: uhidev0: iclass 3/0, 26 report ids
Aug  9 19:23:53 server /bsd: upd0 at uhidev0

==> /var/log/daemon <==
Aug  9 19:24:03 server sensorsd[33299]: upd0 has appeared
```

```
>sysctl hw.sensors
hw.sensors.km0.temp0=62.75 degC
hw.sensors.upd0.indicator0=Off (Charging), OK
hw.sensors.upd0.indicator1=Off (Discharging), OK
hw.sensors.upd0.indicator2=On (ACPresent), OK
hw.sensors.upd0.indicator3=Off (Overload), OK
hw.sensors.upd0.percent0=100.00% (RemainingCapacity), OK
hw.sensors.upd0.percent1=100.00% (FullChargeCapacity), OK
hw.sensors.upd0.timedelta0=4530.000000 secs (RunTimeToEmpty), OK
```

All of the above is automatic, bearing the fruits of sensorsd without any configuration. It is possible to configure /etc/sensorsd.conf to shut-down the system when hw.sensors.upd0.percent0 < 5 min, for example, and move on. 

Let us play with NUT now. 

```
> nut-scanner --usb_scan --disp_nut_conf
SNMP library not found. SNMP search disabled.
Neon library not found. XML search disabled.
AVAHI client library not found. AVAHI search disabled.
Scanning USB bus.
[nutdev1]
        driver = "usbhid-ups"
        port = "auto"
        vendorid = "0764"
        productid = "0501"
        product = "CP900EPFCLCD"
        serial = "000000000000"
        vendor = "CPS"
        bus = "000"
```

Appending the above to /etc/nut/ups.conf leads to the following error. 

```
>doas /usr/local/sbin/upsdrvctl start   
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
No matching HID UPS found - check permissions on /dev/ugen* and /dev/usb*
Driver failed to start (exit status=1)
```

So, port=auto does not work: nut wants an explicit device with pre-set permissions. 

```
>usbdevs -vd
Controller /dev/usb0:
addr 1: super speed, self powered, config 1, xHCI root hub(0x0000), AMD(0x1022), rev 1.00
  uhub0
 port 1 disabled
 port 2 disabled
 port 3 addr 2: full speed, power 2 mA, config 1, CP900EPFCLCD(0x0501), CPS(0x0764), rev 0.01, iSerialNumber 000000000000
   uhidev0
 port 4 addr 3: high speed, self powered, config 1, P'9221 Mobile Drive(0x1058), LaCie(0x059f), rev 0.06, iSerialNumber 00000000e46ad08900f0
   umass0
Controller /dev/usb1:
[...]
```

```
> ls -l /dev/uhid*
crw-rw----  1 root  wheel   62,   0 Mar  8  2016 /dev/uhid0
crw-rw----  1 root  wheel   62,   1 Mar  8  2016 /dev/uhid1
crw-rw----  1 root  wheel   62,   2 Mar  8  2016 /dev/uhid2
crw-rw----  1 root  wheel   62,   3 Mar  8  2016 /dev/uhid3
```

```
> ls -l /dev/usb*
crw-rw----  1 root  wheel   61,   0 Mar  8  2016 /dev/usb0
crw-rw----  1 root  wheel   61,   1 Mar  8  2016 /dev/usb1
crw-rw----  1 root  wheel   61,   2 Mar  8  2016 /dev/usb2
crw-rw----  1 root  wheel   61,   3 Mar  8  2016 /dev/usb3
crw-rw----  1 root  wheel   61,   4 Mar  8  2016 /dev/usb4
crw-rw----  1 root  wheel   61,   5 Mar  8  2016 /dev/usb5
crw-rw----  1 root  wheel   61,   6 Mar  8  2016 /dev/usb6
crw-rw----  1 root  wheel   61,   7 Mar  8  2016 /dev/usb7
```

```
> ls -l /dev/ugen*                     
crw-rw----  1 root  wheel   63,   0 Mar  8  2016 /dev/ugen0.00
crw-rw----  1 root  wheel   63,   1 Mar  8  2016 /dev/ugen0.01
crw-rw----  1 root  wheel   63,   2 Mar  8  2016 /dev/ugen0.02
crw-rw----  1 root  wheel   63,   3 Mar  8  2016 /dev/ugen0.03
crw-rw----  1 root  wheel   63,   4 Mar  8  2016 /dev/ugen0.04
crw-rw----  1 root  wheel   63,   5 Mar  8  2016 /dev/ugen0.05
[...]
```

to be continued

clepple, 2017-08-09:
> No matching HID UPS found - check permissions on /dev/ugen* and /dev/usb*
> Driver failed to start (exit status=1)
> 
> So, port=auto does not work: nut wants an explicit device with pre-set permissions.

These things are orthogonal.

usbhid-ups does not care what you set the "port" value to, as long as the config parser can read it. (libusb does not accept /dev paths, so we match based on VID/PID and/or USB strings returned from libusb.)

However, NUT drivers drop permissions before talking to USB devices, so you might need to adapt this set of FreeBSD instructions:

https://people.freebsd.org/~thierry/nut_FreeBSD_HowTo.txt

hdatma, 2017-08-10:
I must not change ownership to /dev/usb0, because it would make umass0 unreachable.

clepple, 2017-08-10:
Can you do something with group write permissions?

Worst case, you can have the NUT driver run as root - add "user = root" to ups.conf.