networkupstools/nut #459: CP850PFCLCD Reports Incorrect Voltages

ZacharyDuBois, 2017-08-05:
I receive the following incorrect values from my UPS.

`output.voltage`
`battery.voltage.nominal` (there is only 1x12v8Ah SLA in this model)

UPSD 2.7.2
```
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 5100
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 16.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CP850PFCLCD
device.serial: 000000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.version: 2.7.2
driver.version.data: CyberPower HID 0.3
driver.version.internal: 0.38
input.transfer.high: 136
input.transfer.low: 88
input.voltage: 119.0
input.voltage.nominal: 120
output.voltage: 136.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 5
ups.mfr: CPS
ups.model: CP850PFCLCD
ups.productid: 0501
ups.realpower.nominal: 510
ups.serial: 000000000000
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

clepple, 2017-08-05:
Might have been fixed in 2.7.3?

https://github.com/networkupstools/nut/commit/c8950dee9c91ce45d05d8f220ea26891fb92329e

clepple, 2017-08-05:
On the other hand, the HID descriptors for CyberPower are usually not that broken, and the nominal voltage can be used to scale the actual reading. Doesn't sound like the case here. 

If you have a maintenance window, can you stop the driver, then manually start it in explore mode? http://networkupstools.org/docs/man/usbhid-ups.html

- Charles

ZacharyDuBois, 2017-08-07:
Sorry, I'm very new to NUTs - I am running this on Debian Jessie. How would I go about manually starting explorer mode. I've tried adding exploreer to `/etc/nut/ups.conf` and then manually starting the daemon which failed.

clepple, 2017-08-07:
You can keep `/etc/nut/ups.conf` as before, and specify the mode on the command line. Stop NUT with something like `sudo service nut-client stop; sudo service nut-server stop`. Then run the driver manually: `sudo /lib/nut/usbhid-ups -DD -x explore -a name-of-ups 2>&1 | tee /tmp/CP850PFCLCD.log`. After 30-45 seconds you can hit Control-C to kill the driver, and then restart the NUT services.

CH3CN, 2017-08-19:
[CP1000PFCLCD.txt](https://github.com/networkupstools/nut/files/1235988/CP1000PFCLCD.txt)
I'm having the same issue with the output voltage on my CyberPower unit.  Attached is my log running explore mode.

clepple, 2017-08-19:
@CH3CN an output voltage that is close to the upper buck limit (HighVoltageTransfer) is a different issue: https://github.com/networkupstools/nut/issues/439

clepple, 2017-08-19:
@ZacharyDuBois I'm still interested in seeing the explore output from your unit for the nominal battery voltage.

However, that shouldn't affect the shutdown signal. I'd still test shutdown to make sure everything else is configured, but the low battery notification (sent as a boolean flag from the UPS) is much less likely to fail, than for the UPS to return incorrect voltages.

Also, if you do adjust the timers, beware that NUT allows timers to be set in seconds, but the CPS firmware seems to round down to minutes. https://github.com/networkupstools/nut/issues/432

ZacharyDuBois, 2017-08-26:
Sorry about the delay, there is a paste of it:
https://paste.zacharydubois.moe/view/raw/92853c99

clepple, 2017-08-26:
Thanks for posting that.

Note the "size: 8" (bits) in the following:
```
   0.098467	Path: UPS.PowerSummary.ConfigVoltage, Type: Feature, ReportID: 0x09, Offset: 0, Size: 8, Value: 24
   0.100621	Path: UPS.PowerSummary.Voltage, Type: Feature, ReportID: 0x0a, Offset: 0, Size: 8, Value: 24
```

Integer voltages for a 12V battery aren't that useful - usually a 16-bit field would be used with a scale factor.

I'm not sure why I didn't see [this note about the CP1500PFCLCD](http://networkupstools.org/ddl/Cyber_Power_Systems/CP1500PFCLCD.html) before. On the other hand, there are cases where other units had valid battery voltages, but needed the 2/3 scale factor (maybe the CP1500AVRLCDa?)

clepple, 2017-08-26:
@ZacharyDuBois I looked around a bit more at the other CP???PFCLCD data in the DDL, and it doesn't seem like there is much we can do about `battery.voltage`. On the other hand, the `output.voltage` issue in #439 might be easier to fix than I thought. Since the issues overlap, I'm going to close this one - feel free to reopen if you have new information, and you can subscribe to notifications on the other issue.

closed 2017-08-26 by clepple