networkupstools/nut #475: [HCL] PowerWalker VI 650 SH supported by usbhid-ups

DaveTheTwat, 2017-09-09:
PowerWalker VI 650 SH

battery.charge:100
battery.runtime:1567                                                                                       
battery.type:PbAc  
battery.voltage:14                        
device.mfr:PPC                                                      
device.model:OfflineUPS
device.serial:000000000
device.type:ups
driver.name:usbhid-ups
driver.parameter.pollfreq:30
driver.parameter.pollinterval:15
driver.parameter.port:auto
driver.parameter.synchronous:no
driver.version:2.7.3
driver.version.data: Phoenixtec/Liebert HID 0.3                                                                   
driver.version.internal:0.39
ups.load:20
ups.mfr:PPC
ups.model: Offline UPS
ups.productid:ffff
ups.serial:000000000
ups.status: OL CHRG
ups.vendorid: 06da

Shutdown sequence tested on FreeNAS-9.10.2-U1 (86c7ef5) successfully.  Emails from FreeNAS received notifying of On Batt and Shutdown.

https://www.powerwalker.com/?page=product&item=10120048&lang=en

DaveTheTwat, 2017-09-10:
Although, I have just received a COMMBAD email around 12 hours after getting this set up.  I think this may be a USB port/driver issue.  Disconnected USB cable from UPS port and reconnected, then restarted the UPS service in the GUI and it will show me results from upsc ups@localhost again.  Annoying intermittent problem.

clepple, 2017-09-10:
Interesting, I don't know if we have seen PowerWalker devices that are compatible with the `usbhid-ups` driver before.

Totally understood if you masked it out, but did the device report a serial number that matched the label? Or is it hard-coded to all zeroes?

Also, do the voltage, charge, load and runtime numbers track actual conditions? It is slightly odd that the battery.voltage isn't reported with at least one decimal point.

I'd be curious to see some of the other logs from when it got the COMMBAD signal. Many UPSes have transient errors that are basically timeouts - if the UPS is performing a self-test, the microcontroller might not be able to keep up with USB comms. We have tried to make the driver reconnect automatically in those cases, but the majority of that testing has been done on Linux. What version of FreeBSD does this version of FreeNAS use? After COMMBAD and before adjusting the cables, do you see the UPS in the output of `usbconfig` (run as root)?

DaveTheTwat, 2017-09-10:
Hi, I haven't masked the serial.  That is as how it shows in the upsc command.

The voltage, charge, load and runtime do change.  I tested the shutdown by pulling the plug for the UPS,  and after it had shutdown my NAS, I rebooted and ran the upsc command.  The voltage had gone from 14 to 12.6, the charge had gone down to about 69%, and the runtime had also dropped.  I did run the upsc quickly before it shutdown and noticed the load was up from about 20% to 41% and the ups.status had changed to something like "onbatt"?.  If not, onbatt, had definitely changed status.

I don't know if it makes any difference, but I am using the "APC ups 2 Back-UPS Pro USB USB (usbhid-ups)" driver.  I don't know if the driver is tweaked depending on make and model.

As a side note, I did first try the blazer_ser and blazer_usb drivers, as these were recommended for other PowerWalker devices, but they were not communicating with the UPS at all.

I have also tried some of the other "various" drivers in the FreeNAS dropdown, like a nutdrv_qx one, but same as the blazer ones, not working.

I haven't tried usbconfig when I get the COMMBAD message.  Next time it happens (hopefully there won't be a next time!), and if I'm home, I'll give it a go.  When I get this message though, I'm unable to use upsd.  This gives me a "data stale" error.

According to the FreeNAS-9.10.2 manual, this uses FreeBSD 10.3.

Another thing I noticed was that any of the USB ports I tried all showed up in FreeNAS as dev/ugen0.5 which isn't actually an option in the dropdown for the ports.  I had to type it in when trying the other drivers, but now use "auto".

I have added maxage=25 and pollinterval=20 to try and fix the "data stale" issues I was having with this driver.

clepple, 2017-09-10:
Thanks for confirming the serial number. I'll put a note next to that.

We still have some room for improvement in the number of digits after the decimal, but if it said 12.6, that's better than having it hard-coded to some number like 14.

I've looked at the FreeNAS configuration system for NUT a few times, and I'm not quite sure how it works. I'd turn it around and say that what matters is the resulting configuration block for that UPS in ups.conf. The `usbhid-ups` driver auto-detects a lot of things from the USB VID:PID, and as you have seen, pretty much ignores the "port" value.

A bit of explanation on the port option: for serial-based UPSes, it was necessary to specify the physical port that the UPS was plugged into. When USB drivers started to appear in Unix land, none of them had physical topology information - or if they did, it wasn't standardized. However, the USB Vendor ID and Product ID (VID:PID) are usually enough to separate an UPS from, say, a keyboard, so we got APIs that basically let you walk the list of VID:PID combos that are plugged into any USB port. However, that doesn't mean we're off the hook from having to worry about individual /dev nodes - the process talking to the USB device still needs read and most likely write permissions to a node like /dev/ugen0.5 (even if the library doesn't let us specify that node directly).

Upon further investigation, it is less surprising that your UPS works with `usbhid-ups` rather than `nutdrv_qx`. All of the existing PowerWalker entries (which have the Qx-style protocol) have a VID:PID of 0665:5161, and yours is 06da:ffff.

For the `usbhid-ups` driver, note that there is also a `pollfreq` option in addition to the standard `pollinterval`. In retrospect, we could have chosen a better name for `pollfreq` since it is for the less-frequent full polling cycle, but given that the default values are 30 and 2 for `pollfreq` and `pollinterval`, respectively, I would suggest bumping `pollfreq` up to some multiple of `pollinterval`.

Similarly, for the "data stale" notification, you could choose to have the data marked stale if it has missed more than two polling cycles by making `maxage` greater than 2x `pollinterval`.

DaveTheTwat, 2017-09-10:
Thanks for your suggestions.  I have added the pollfreq=40 option using a multiplier of pollinterval=20.  I have also changed the maxage to 60 using a multiplier of greater than 2x pollinterval.

When I made these changes, I did get another COMMBAD email, so I ran usbconfig and the UPS was still showing as connected by the usb port.  I could still run upsc and bring up the data from the UPS, so I'm going to say that it's still working, haha.

closed 2017-09-12 by clepple

kaminkehrer, 2018-02-20:
I use a Powerwalker VI 650 SH on a raspberry, The raspberry is installed with raspbian (jessie). But I have installed the stretch packages for nut, nut-client and nut-server. Therefore this packages are the current stable once.
I configured in ups.conf: pollfreq=30, pollinterval=15, driver= usbhid-ups and in upsd.conf: MAXAGE 60.
With this configuration I have COMMBAD (NOCOMM) every second to every third day. A USB cable disconnect for 10-20 seconds and connect again solve this problem.
Is there any possibility to support a proper connect.
I'm very interested to support fixing this error.

kaminkehrer, 2018-02-20:
`syslog.1:Feb 19 20:47:13 raspberrypi upsmon[641]: Poll UPS [Powerwalker@localhost] failed - Data stale
syslog.1:Feb 19 20:47:18 raspberrypi upsmon[641]: Poll UPS [Powerwalker@localhost] failed - Data stale
syslog.1:Feb 19 20:47:23 raspberrypi upsmon[641]: Poll UPS [Powerwalker@localhost] failed - Data stale
syslog.1:Feb 19 20:47:28 raspberrypi upsmon[641]: Poll UPS [Powerwalker@localhost] failed - Data stale
syslog.1:Feb 19 20:47:33 raspberrypi upsmon[641]: Poll UPS [Powerwalker@localhost] failed - Data stale
syslog.1:Feb 19 20:47:38 raspberrypi upsmon[641]: Poll UPS [Powerwalker@localhost] failed - Data stale
syslog.1:Feb 19 20:47:43 raspberrypi upsmon[641]: Poll UPS [Powerwalker@localhost] failed - Data stale
syslog.1:Feb 19 20:47:43 raspberrypi upsmon[641]: UPS Powerwalker@localhost is unavailable
syslog.1:Feb 19 20:47:48 raspberrypi upsmon[641]: Poll UPS [Powerwalker@localhost] failed - Data stale
syslog.1:Feb 19 20:47:53 raspberrypi upsmon[641]: Poll UPS [Powerwalker@localhost] failed - Data stale
syslog.1:Feb 19 20:47:58 raspberrypi upsmon[641]: Poll UPS [Powerwalker@localhost] failed - Data stale
syslog.1:Feb 19 20:48:03 raspberrypi upsmon[641]: Poll UPS [Powerwalker@localhost] failed - Data stale
---> reboot
syslog.1:Feb 19 20:48:19 raspberrypi systemd[1]: Started Various fixups to make systemd work better on Debian.
syslog.1:Feb 19 20:48:33 raspberrypi upsdrvctl[563]: Using subdriver: Phoenixtec/Liebert HID 0.3
syslog.1:Feb 19 20:48:33 raspberrypi upsdrvctl[563]: Network UPS Tools - Generic HID driver 0.41 (2.7.4)
syslog.1:Feb 19 20:48:33 raspberrypi upsdrvctl[563]: USB communication driver 0.33
syslog.1:Feb 19 20:48:33 raspberrypi upsdrvctl[563]: Network UPS Tools - UPS driver controller 2.7.4
syslog.1:Feb 19 20:48:33 raspberrypi usbhid-ups[580]: Startup successful
syslog.1:Feb 19 20:48:36 raspberrypi upsd[618]: fopen /var/run/nut/upsd.pid: No such file or directory
syslog.1:Feb 19 20:48:36 raspberrypi upsd[618]: listening on 192.168.178.50 port 3493
syslog.1:Feb 19 20:48:36 raspberrypi upsd[618]: listening on 127.0.0.1 port 3493
syslog.1:Feb 19 20:48:36 raspberrypi upsd[618]: Connected to UPS [Powerwalker]: usbhid-ups-Powerwalker
syslog.1:Feb 19 20:48:36 raspberrypi upsd[618]: listening on 192.168.178.50 port 3493
syslog.1:Feb 19 20:48:36 raspberrypi upsd[618]: listening on 127.0.0.1 port 3493
syslog.1:Feb 19 20:48:36 raspberrypi upsd[618]: Connected to UPS [Powerwalker]: usbhid-ups-Powerwalker
syslog.1:Feb 19 20:48:36 raspberrypi upsd[637]: Startup successful
syslog.1:Feb 19 20:48:37 raspberrypi upsmon[638]: fopen /var/run/nut/upsmon.pid: No such file or directory
syslog.1:Feb 19 20:48:37 raspberrypi upsmon[638]: UPS: Powerwalker@localhost (master) (power value 1)
syslog.1:Feb 19 20:48:37 raspberrypi upsmon[641]: Startup successful
syslog.1:Feb 19 20:48:37 raspberrypi systemd[1]: PID file /var/run/nut/upsmon.pid not readable (yet?) after start.
syslog.1:Feb 19 20:48:37 raspberrypi upsmon[642]: Init SSL without certificate database
syslog.1:Feb 19 20:48:38 raspberrypi upsd[637]: User upsmon_local@127.0.0.1 logged into UPS [Powerwalker] pi@raspberrypi:/var/log $`

DaveTheTwat, 2018-02-20:
Hi kaminkehrer, I have now set mine to pollinterval=120 pollfreq=240 and no maxage.  It still gives me the data stale message every few weeks, but is much better than lower settings