networkupstools/nut #476: usbhid-ups man page: explain difference between pollfreq and pollinterval

clepple, 2017-09-10:
I thought we had this covered somewhere else, but it probably makes sense to explain the `pollfreq` stuff in the man page, and highlight the difference between that and `pollinterval`.

aquette, 2017-11-17:
@clepple I'd appreciate a review of  cf32fd7 to ensure that it indeed clarifies the point :) thx

jimklimov, 2017-11-18:
I posted my comments to the commit.

clepple, 2017-11-21:
@aquette thanks for taking a look at this. I agree with @jimklimov's comments, but I still think this could be made clearer.

I almost wonder if we should put the discussion of USB "interrupts" into a different paragraph, since it's really just another form of polling. The main point I think we want to get across is that the usbhid-ups driver can overwhelm the UPS, regardless of the style of notifications (EP0 or interrupt).

aquette, 2017-11-27:
@clepple and @jimklimov : I've made a commit (a1f7445) taking into account @jimklimov comments, and clarifying the (in fact) 3 modes for drivers.
Then about usbhid-ups, what about putting the details of this "USB interrupts specific" into the IMPLEMENTATION section, and limit "pollfreq" to the basic, while cross-referencing these IMPLEMENTATION details? I'll try to make a PR proposal if that suits you

clepple, 2017-11-27:
@aquette sounds good - feel free to reuse the wording from any of the list emails I have written regarding USB interrupts.

aquette, 2017-11-27:
@clepple would be great if you could point or extract such words. As you know, I'm still a bit away from the lists... sadly. Thanks

clepple, 2017-12-30:
Working on this. I'm going to sync up the documentation between `usbhid-ups`, `nutdrv_qx` and `snmp-ups` since all three have the same concept of `pollfreq`.

closed 2018-01-02 by clepple