networkupstools/nut #481: Error: too many non-option arguments. Try -h for help.

Synergyst, 2017-09-19:
Hello.

I have tried this on two separate systems with the same result.
Am I not understanding something from the documentation on the website perhaps?

I am attempting to write a new sub-driver for my Mini-Box DCDC-USB device using NUT by following the directions in the following link:
[http://networkupstools.org/docs/developer-guide.chunked/ar01s04.html#hid-subdrivers](http://networkupstools.org/docs/developer-guide.chunked/ar01s04.html#hid-subdrivers)

However when running `/lib/nut/usbhid-ups -DD -u root -x explore -x vendorid=04d8 auto` I get the following error:
```
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     Error: too many non-option arguments. Try -h for help.
```

The device data from `lsusb `I receive is as follows:
`Bus 001 Device 020: ID 04d8:d003 Microchip Technology, Inc.`

When I plug the device into the USB port on my device `dmesg `gives the following output: 
```
[32083.222998] usb 1-1.2: new full-speed USB device number 20 using dwc_otg
[32083.364501] usb 1-1.2: New USB device found, idVendor=04d8, idProduct=d003
[32083.369068] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[32083.373508] usb 1-1.2: Product: DCDC-USB
[32083.377707] usb 1-1.2: Manufacturer: MINI-BOX.COM
[32083.381819] usb 1-1.2: SerialNumber: 1.00rel.
[32083.405727] input: MINI-BOX.COM DCDC-USB as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:04D8:D003.000C/input/input9
[32083.483857] hid-generic 0003:04D8:D003.000C: input,hiddev0,hidraw0: USB HID v1.01 Device [MINI-BOX.COM DCDC-USB] on usb-3f980000.usb-1.2/input0
```

If I missed out any relevant information please let me know so I can provide it.

Kind regards.

clepple, 2017-09-19:
There is a typo in the documentation: the `auto` at the end should be `-a auto`.

https://github.com/networkupstools/nut/issues/436

Synergyst, 2017-09-20:
Hello.

Thanks for the quick reply!

Now I am receiving this output:
```
# /root/nut-build/nut/drivers/usbhid-ups -DD -u root -x explore -x vendorid=04d8 -a auto
Network UPS Tools - Generic HID driver 0.42 (2.7.4-371-g30e04684)
USB communication driver 0.33
   0.000000     Error: Section auto not found in ups.conf
```

Am I still missing something..?

clepple, 2017-09-20:
That'll teach me to blindly believe the comments in issue #436 :-)

I think that section in the developer docs was written prior to the `-a name-of-UPS` flag was added, so the `auto` would have been the port name for a non-USB driver.

If you can use a revision that contains 8e94b560, you can specify everything on the command line with `-s name-of-UPS`. Otherwise, you will need to create a `[name-of-UPS]` section in `ups.conf` with a minimum of `port = auto`.

Synergyst, 2017-09-20:
Hello.

I created a blank section in my `ups.conf` file containing:
```
[dcdcusb]
       port = auto
```

This seems to have resolved the issue mentioned in this ticket, so I will mark it as solved for that.

However this did create another problem, however I think it is unrelated to NUT and more so to how the firmware communicates to the OS.

Kind regards.

closed 2017-09-20 by Synergyst