networkupstools/nut #484: Output load for Eaton UPS

realies, 2017-10-05:
When using an Eaton 5E 650i UPS with the proprietary Eaton UPS Companion, there is an additional value named "Output load" which seems like useful information to have via `nut`. Is this value accessible via `nut`?

[image]

```
# upsc ups
battery.charge: 100
battery.runtime: 1590
battery.type: PbAc
device.mfr: EATON
device.model: 5E 650i
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: MGE HID 1.39
driver.version.internal: 0.41
input.voltage: 229.0
outlet.1.status: on
outlet.desc: Main Outlet
outlet.id: 1
outlet.switchable: no
output.frequency: 50.0
output.frequency.nominal: 50
output.voltage: 226.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 02.06.0017
ups.load: 21
ups.mfr: EATON
ups.model: 5E 650i
ups.power.nominal: 650
ups.productid: ffff
ups.start.battery: yes
ups.status: OL
ups.timer.shutdown: -1
ups.vendorid: 0463
```

clepple, 2017-10-05:
I would think that it should be `ups.power.nominal` * `ups.load` / 100%, but the math doesn't quite work out. Maybe `ups.power.nominal` is a VA rating, and there is a lower max wattage that we are not reading?

@aquette do you have any newer protocol documentation? I don't know if the 5E follows the same conventions as the original MGE USB HID spec from 2001.

aquette, 2017-10-06:
There may be some hid paths missing.
@realies: would you be kind enough to make a debug run?
Stop the driver, and start usbhid-ups -D -a ups, and send back the dump
after 10 secs... thx

realies, 2017-10-06:
@aquette, sure.

```
# timeout 10s usbhid-ups -D -a ups -u root
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '1'
   0.000798     upsdrv_initups...
   2.306339     Using subdriver: MGE HID 1.39
   2.306363     57 HID objects found
   2.355127     Path: UPS.Flow.[4].ConfigApparentPower, Type: Feature, ReportID: 0x0d, Offset: 0, Size: 16, Value: 650
   2.355148     Path: UPS.Flow.[4].ConfigFrequency, Type: Feature, ReportID: 0x0d, Offset: 16, Size: 8, Value: 50
   2.399158     Path: UPS.Flow.[4].ConfigVoltage, Type: Feature, ReportID: 0x12, Offset: 0, Size: 8, Value: 230
   2.450137     Path: UPS.Flow.[4].FlowID, Type: Feature, ReportID: 0x0b, Offset: 0, Size: 8, Value: 4
   2.450167     Path: UPS.OutletSystem.Outlet.[1].FlowID, Type: Feature, ReportID: 0x0b, Offset: 8, Size: 8, Value: 4
   2.450186     Path: UPS.OutletSystem.Outlet.[1].OutletID, Type: Feature, ReportID: 0x0b, Offset: 16, Size: 8, Value: 1
   2.502105     Path: UPS.OutletSystem.Outlet.[1].PresentStatus.Switchable, Type: Feature, ReportID: 0x0c, Offset: 0, Size: 8, Value: 0
   2.502129     Path: UPS.OutletSystem.OutletSystemID, Type: Feature, ReportID: 0x0b, Offset: 24, Size: 8, Value: 1
   2.559141     Path: UPS.PowerConverter.Input.[1].InputID, Type: Feature, ReportID: 0x0f, Offset: 0, Size: 8, Value: 1
   2.603142     Path: UPS.PowerConverter.Input.[1].PresentStatus.Boost, Type: Feature, ReportID: 0x02, Offset: 0, Size: 8, Value: 0
   2.603173     Path: UPS.PowerConverter.Input.[1].PresentStatus.Boost, Type: Input, ReportID: 0x02, Offset: 0, Size: 8, Value: 0
   2.658147     Path: UPS.PowerConverter.Input.[1].Voltage, Type: Feature, ReportID: 0x0e, Offset: 0, Size: 16, Value: 232
   2.658171     Path: UPS.PowerConverter.Input.[3].InputID, Type: Feature, ReportID: 0x0f, Offset: 8, Size: 8, Value: 3
   2.658183     Path: UPS.PowerConverter.Input.[3].StartOnBattery, Type: Feature, ReportID: 0x0f, Offset: 16, Size: 8, Value: 1
   2.658203     Path: UPS.PowerConverter.Output.Frequency, Type: Feature, ReportID: 0x0e, Offset: 16, Size: 16, Value: 49.9
   2.658223     Path: UPS.PowerConverter.Output.OutputID, Type: Feature, ReportID: 0x0f, Offset: 24, Size: 8, Value: 1
   2.658240     Path: UPS.PowerConverter.Output.Voltage, Type: Feature, ReportID: 0x0e, Offset: 32, Size: 16, Value: 230
   2.658256     Path: UPS.PowerConverter.PowerConverterID, Type: Feature, ReportID: 0x0f, Offset: 32, Size: 8, Value: 1
   2.702117     Path: UPS.PowerSummary.AudibleAlarmControl, Type: Feature, ReportID: 0x14, Offset: 0, Size: 8, Value: 2
   2.702145     Path: UPS.PowerSummary.CapacityMode, Type: Feature, ReportID: 0x0c, Offset: 8, Size: 8, Value: 2
   2.748126     Path: UPS.PowerSummary.DelayBeforeShutdown, Type: Feature, ReportID: 0x09, Offset: 0, Size: 16, Value: -1
   2.748148     Path: UPS.PowerSummary.DesignCapacity, Type: Feature, ReportID: 0x0c, Offset: 16, Size: 8, Value: 100
   2.748159     Path: UPS.PowerSummary.FlowID, Type: Feature, ReportID: 0x0f, Offset: 40, Size: 8, Value: 4
   2.748170     Path: UPS.PowerSummary.FullChargeCapacity, Type: Feature, ReportID: 0x0c, Offset: 24, Size: 8, Value: 100
   2.804148     Path: UPS.PowerSummary.iDeviceChemistry, Type: Feature, ReportID: 0x10, Offset: 0, Size: 8, Value: 5
   2.804175     Path: UPS.PowerSummary.iManufacturer, Type: Feature, ReportID: 0x10, Offset: 8, Size: 8, Value: 1
   2.804186     Path: UPS.PowerSummary.iModel, Type: Feature, ReportID: 0x10, Offset: 16, Size: 8, Value: 8
   2.804195     Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x10, Offset: 24, Size: 8, Value: 2
   2.804205     Path: UPS.PowerSummary.iSerialNumber, Type: Feature, ReportID: 0x10, Offset: 32, Size: 8, Value: 4
   2.804216     Path: UPS.PowerSummary.iVersion, Type: Feature, ReportID: 0x10, Offset: 40, Size: 8, Value: 6
   2.851121     Path: UPS.PowerSummary.PercentLoad, Type: Feature, ReportID: 0x07, Offset: 0, Size: 8, Value: 23
   2.851146     Path: UPS.PowerSummary.PowerSummaryID, Type: Feature, ReportID: 0x0f, Offset: 48, Size: 8, Value: 1
   2.900121     Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x01, Offset: 0, Size: 1, Value: 1
   2.900147     Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Input, ReportID: 0x01, Offset: 0, Size: 1, Value: 1
   2.900160     Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x01, Offset: 1, Size: 1, Value: 0
   2.900171     Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Input, ReportID: 0x01, Offset: 1, Size: 1, Value: 0
   2.900183     Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x01, Offset: 2, Size: 1, Value: 1
   2.900194     Path: UPS.PowerSummary.PresentStatus.Charging, Type: Input, ReportID: 0x01, Offset: 2, Size: 1, Value: 1
   2.900205     Path: UPS.PowerSummary.PresentStatus.CommunicationLost, Type: Feature, ReportID: 0x01, Offset: 3, Size: 1, Value: 0
   2.900216     Path: UPS.PowerSummary.PresentStatus.CommunicationLost, Type: Input, ReportID: 0x01, Offset: 3, Size: 1, Value: 0
   2.900227     Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 4, Size: 1, Value: 0
   2.900245     Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Input, ReportID: 0x01, Offset: 4, Size: 1, Value: 0
   2.900262     Path: UPS.PowerSummary.PresentStatus.Good, Type: Feature, ReportID: 0x01, Offset: 5, Size: 1, Value: 1
[... cut ...]
   3.353597     upsdrv_updateinfo...
   3.603746     Got 0 HID objects...
   3.603774     Quick update...
   3.603886     upsdrv_updateinfo...
   3.853990     Got 0 HID objects...
   3.854017     Quick update...
   5.605711     upsdrv_updateinfo...
   5.855877     Got 0 HID objects...
   5.855903     Quick update...
   7.607408     upsdrv_updateinfo...
   7.857536     Got 0 HID objects...
   7.857560     Quick update...
   9.609108     upsdrv_updateinfo...
   9.859230     Got 0 HID objects...
   9.859250     Quick update...
```

aquette, 2017-10-07:
thanks.

Output load is mapped to the data "UPS.PowerConverter.Output.ActivePower"
which is missing here, and published in NUT as "ups.realpower".
Charles was almost right with his guess, modulo the power factor and some
(hem) black magic.
The approximation used in EUC, in case this data is missing, is the
following:

round ("UPS.PowerSummary.PercentLoad" * 0.01 *
"UPS.Flow[4].ConfigApparentPower" * powerFactor)

with powerFactor mapped to "UPS.PowerConverter.Output.PowerFactor" or
defaulting to 0.80 otherwise.
As you can see with such models, it's really an approximative
approximation, which gives in your case the following (matching EUC UI
values):

27 * 0.01 * 650 * 0.80 = 140 W

I've cc'ed myself at work to check on monday to check, since I've no device
/ time at home currently, but don't have allocated time nor management
support at work either... Mass user lobbying would help

realies, 2017-10-11:
Let me know if you require any additional data sampling.

aquette, 2017-11-17:
hi @realies 
sorry for the silence, I had this patch ready since a month, tested on a 5E, but still not yet non-regression tested on other units (providing the missing data on 5E).
I'll try to do this test today to then push into the mainline of NUT. In the meantime, if you can test also on your side, that's always welcome.

aquette, 2017-11-17:
non-regression testing done on a 5PX and good to go. Hence, merging the branch. Will be available for 2.7.5

closed 2017-11-17 by aquette

realies, 2018-03-09:
Getting back to this, how reliable would the power factor of `0.80` be? https://github.com/dmacias72/NUT-unRAID/issues/2#issuecomment-371788242