networkupstools/nut #485: C++ exception specifications are deprecated

clepple, 2017-10-07:
We are starting to see a lot of the following warnings in Ubuntu artful:

`warning: dynamic exception specifications are deprecated in C++11`

Apparently, these don't help the compiler: https://softwareengineering.stackexchange.com/questions/114338/why-are-exception-specifications-bad

Thoughts?

clepple, 2017-10-07:
(definitely not something for the next release)

EmilienKia, 2018-03-09:
Does it mean default standard target for C++in NUT is C++11 ?

clepple, 2018-03-10:
@EmilienKia not necessarily. I'm just pointing out two things:
1. People are going to be seeing a lot of warnings with certain C++ compilers
2. The deprecation might mean that we can simplify the generated code by not specifying exception types (if I understand that StackOverflow post correctly) for potentially several C++ standards, and also get rid of said warnings at the same time.

(I'm not advocating zero warnings, just that a lot of repeated warnings can make it hard to find the ones that matter. If needed, we can go in and turn off individual warnings with CXXFLAGS.)

http://buildbot.networkupstools.org/public/nut/waterfall?show=Ubuntu-artful-x64

K0HAX, 2021-06-05:
Building on Gentoo testing has been broken by this issue. C++19 fails to compile with this error now.

I copied the ebuild (gentoo's build description language) and customized it so I could still build nut. 
Here is my [modification](https://github.com/K0HAX/gentoo-overlay/blob/46297c5d0c6edc90aee4fdc191dedf6af3514bf3/sys-power/nut/nut-2.7.4-r7.ebuild#L117). I'm just forcing C++14 during the build.

jimklimov, 2021-09-21:
Recent year's work on the NUT master branch to reduce the warnings, and to build on CI with many distros*compiler*standards in a matrix, did address C++ code in several PRs; would the esteemed commenters please check if their builds of the current codebase have fewer/none of these?

jimklimov, 2024-07-30:
I think by now all the complaining instances are either gone or rather hidden by `#if (defined __cplusplus) && (__cplusplus < 201100)` to cater to both old and new compilers. The NUT CI farm tests with a range of language specs and they all pass without warnings.

closed 2024-07-30 by jimklimov