networkupstools/nut #490: blazer_usb error processing is completely bogus.

blacklion, 2017-10-19:
blazer_usb uses such code to process low-level errors:
```
	ret = (*subdriver_command)(cmd, buf, buflen);
	if (ret >= 0) {
		return ret;
	}

	switch (ret)
	{
	case -EBUSY:		/* Device or resource busy */
		fatal_with_errno(EXIT_FAILURE, "Got disconnected by another driver");

	case -EPERM:		/* Operation not permitted */
		fatal_with_errno(EXIT_FAILURE, "Permissions problem");

	case -EPIPE:		/* Broken pipe */
		if (usb_clear_halt(udev, 0x81) == 0) {
			upsdebugx(1, "Stall condition cleared");
			break;
		}
#ifdef ETIME
	case -ETIME:		/* Timer expired */
#endif
		if (usb_reset(udev) == 0) {
			upsdebugx(1, "Device reset handled");
		}
	case -ENODEV:		/* No such device */
	case -EACCES:		/* Permission denied */
	case -EIO:		/* I/O error */
	case -ENXIO:		/* No such device or address */
	case -ENOENT:		/* No such file or directory */
		/* Uh oh, got to reconnect! */
		usb->close(udev);
		udev = NULL;
		break;

	case -ETIMEDOUT:	/* Connection timed out */
	case -EOVERFLOW:	/* Value too large for defined data type */
#ifdef EPROTO
	case -EPROTO:		/* Protocol error */
#endif
	default:
		break;
	}
```
Let see what does `ret` really means?! It is result of usb_XXX call (I'm using cypress subdriver as example, but it is common for all subdrivers):
```
/* Write data in 8-byte chunks */
/* ret = usb->set_report(udev, 0, (unsigned char *)&tmp[i], 8); */
ret = usb_control_msg(udev, USB_ENDPOINT_OUT + USB_TYPE_CLASS + USB_RECIP_INTERFACE,
	0x09, 0x200, 0, &tmp[i], 8, 5000);
if (ret <= 0) {
	upsdebugx(3, "send: %s", ret ? usb_strerror() : "timeout");
	return ret;
}
```

But libusb has its own error codes, which are NOT COMPATIBLE with `errno` constants! Here are libusb error codes:

```
enum libusb_error {
        LIBUSB_SUCCESS = 0,
        LIBUSB_ERROR_IO = -1,
        LIBUSB_ERROR_INVALID_PARAM = -2,
        LIBUSB_ERROR_ACCESS = -3,
        LIBUSB_ERROR_NO_DEVICE = -4,
        LIBUSB_ERROR_NOT_FOUND = -5,
        LIBUSB_ERROR_BUSY = -6,
        LIBUSB_ERROR_TIMEOUT = -7,
        LIBUSB_ERROR_OVERFLOW = -8,
        LIBUSB_ERROR_PIPE = -9,
        LIBUSB_ERROR_INTERRUPTED = -10,
        LIBUSB_ERROR_NO_MEM = -11,
        LIBUSB_ERROR_NOT_SUPPORTED = -12,
        LIBUSB_ERROR_OTHER = -99,
};
```
So, `-1` is LIBUSB_ERROR_IO and must lead to reconnect, but it is treated as `-EPERM` and kill driver.
Looks like, these drivers have same problem too:
```
nutdrv_qx.c
riello_usb.c
usbhid-ups.c
```
And `libusb.c` wrapper has same erroneous code.
This problem is long-standing one for FreeBSD, where sometimes drive get IO error which could be fixed by reconnection, but driver kill itself because it thinks it is permission problem.

clepple, 2017-10-19:
Since you mentioned `libusb.c`, I am assuming you are looking at code from the master branch, which assumes that the API is closest to libusb-0.1. There is an issue that describes some of the work to port to libusb-1.0: https://github.com/networkupstools/nut/issues/300 and a branch which contains the latest code: https://github.com/zykh/nut/tree/libusb-1.0%2B0.1

Can you please try this 1) with the `libusb-1.0+0.1` branch, since that is the way forward, and 2) with `nutdrv_qx` because the `blazer_*` drivers are no longer maintained, and will be deprecated?

Also, it would be helpful to know more about your system. Which version of FreeBSD, and how does it report its libusb API version? (This is interesting, because I have been testing both master and the libusb-1.0 branch for months on FreeBSD 10.x and 11.x with an MGE UPS, and I have not seen any cases where the `usbhid-ups` driver terminates - just the occasional stale data when the UPS is busy doing a self-test.)

blacklion, 2017-10-19:
libusb.c presents in latest released 2.4.7 too, with same problem.

Problem with blazer_usb is reported all over the internet for FreeBSD 9, 10 and 11.  I have problems (driver dies twice a day) on FreeBSD 11.1-STABLE with old Ippon UPS. (cypress subdriver). I didn';t have them for years, but about month ago they have been started (I don't know why). Googling shows, that it is known problem, but not 100% reproducable. May be, it is something cable-related, or some minor problems with cheap UPSes
FreeBSD uses `--with-usb=auto` argument for `configure` script and it shows:
```
checking for libusb version via pkg-config... checking via libusb-config... none found
checking for libusb cflags...
checking for libusb ldflags... -lusb
checking for usb.h... yes
checking for usb_init... yes
```
Anyway treat libusb 0.1 error codes as "negated errno" error codes is wrong, I can not find any documentation which says that libusb 0.1 uses such error codes! libusb-0.1.12 documentation says only "less than zero on error" without any other details, and as FreeBSD uses native libusb 0.1 API emulation (not libusb 0.1 sources themselves), it uses libusb 1.0 error codes, which is perfectly legal — they are less than zero and it is only promise from libusb 0.1 API!

blacklion, 2017-10-19:
I'll try `nutdrv_qx` from latest release right now, but it has exactly same problem in release (I could say by looking at sources), and I'll try `libusb-1.0+0.1` later.

Anyway, IMHO, current sources has big problem with libusb error processing, as it uses invalid assumptions.

blacklion, 2017-10-19:
It is interesting: branch `libusb-1.0+0.1` has `blazer_usb` (which will be deprecated) converted, but `nutdrv_qx` don't! 
I'm not sure, I need to try it, as it will have same problem :-)
I tried to build branch on FreeBSD, but looks like I can not use autohell right, as I get strange `./configure` errors, but I could say,that `libusb`  is detected as 1.0:
````
checking for libusb 1.0 version via pkg-config... 1.0.13 found
checking for libusb 0.1 version via pkg-config... checking via libusb-config... none found
checking for libusb cflags...
checking for libusb ldflags... -lusb
checking for libusb.h... yes
checking for libusb_init... yes
checking for libusb_set_auto_detach_kernel_driver... yes
checking for libusb_detach_kernel_driver... yes
checking whether to build USB drivers... yes (libusb-1.0)
````

clepple, 2017-10-20:
> It is interesting: branch libusb-1.0+0.1 has blazer_usb (which will be deprecated) converted, but nutdrv_qx don't! [sic]

Not sure what you mean here. I don't have a USB-enabled Qx UPS to test with, but there appears to be both libusb-1.0 and 0.1 code at the beginning of `nutdrv_qx.c`.

> I tried to build branch on FreeBSD, but looks like I can not use autohell right, as I get strange ./configure errors,

Does that branch configure and build in the end? If not, please file an issue with the error messages.

Looking back through the archives, it does seem that the FreeBSD libusb error codes don't line up with libusb-0.1, but then again, the FreeBSD kernel previously masked a lot of separate errors behind `EIO`: http://lists.alioth.debian.org/pipermail/nut-upsuser/2015-February/009541.html

blacklion, 2017-10-20:
> Not sure what you mean here. I don't have a USB-enabled Qx UPS to test with, but there appears to be both libusb-1.0 and 0.1 code at the beginning of nutdrv_qx.c.
Here is code from `nutdrv_qx.c` from this branch (commit dfd514e7a01bb0a78cd7defca99c0494bf50d83f)
```
		switch (ret)
		{
		case -EBUSY:		/* Device or resource busy */
			fatal_with_errno(EXIT_FAILURE, "Got disconnected by another driver");

		case -EPERM:		/* Operation not permitted */
			fatal_with_errno(EXIT_FAILURE, "Permissions problem");

		case -EPIPE:		/* Broken pipe */
			if (usb_clear_halt(udev, 0x81) == 0) {
				upsdebugx(1, "Stall condition cleared");
				break;
			}
	#ifdef ETIME
		case -ETIME:		/* Timer expired */
	#endif	/* ETIME */
			if (usb_reset(udev) == 0) {
				upsdebugx(1, "Device reset handled");
			}
		case -ENODEV:		/* No such device */
		case -EACCES:		/* Permission denied */
		case -EIO:		/* I/O error */
		case -ENXIO:		/* No such device or address */
		case -ENOENT:		/* No such file or directory */
			/* Uh oh, got to reconnect! */
			usb->close(udev);
			udev = NULL;
			break;

		case -ETIMEDOUT:	/* Connection timed out */
		case -EOVERFLOW:	/* Value too large for defined data type */
#ifdef EPROTO
		case -EPROTO:		/* Protocol error */
#endif
		default:
			break;
		}
```
It is (wrong) code for libusb-0.1, not for 1.0 (which uses other error values). `blazer_usb.c` `#define` error codes according to used libusb version (but, again, uses wrong assumption about `-errno` in case of libusb-0.1).
> Looking back through the archives, it does seem that the FreeBSD libusb error codes don't line up with libusb-0.1
libusb-0.1 doesn't have any error codes specified, so it is impossible to be «line up» with it.
This infamous «Spurious permission problem on FreeBSD» bug have clear explanation: it is return value `-1` which is `LIBUSB_ERROR_IO` and not `-EPERM`.

clepple, 2017-10-20:
I see your point, but the NUT code was written when libusb-0.1 was the only API around, and all of the other platforms had consistent error codes. So while it may not have been a well-defined API, it was the de facto standard.

Tagging @zykh (in case this was not on your radar).

blacklion, 2017-10-20:
As soon as usage of libusb-1.0 API with standardized error codes (LIBUSB_ERROR_XXXX) will landed, it will be irrelevant :-) When I reported this problem, I didn't know, that there are plans to support libusb-1.0 API.

Now I've added local patch to FreeBSD ports to adapt error codes for current release (which is pure libusb-0.1).

I hope, that all drivers will be properly converted to support proper error codes in libusb-1.0 API mode and this branch will landed :-)

zykh, 2018-01-11:
hmm..

@blacklion, you're right on the USB error handling in `libusb-1.0+0.1`'s `nutdrv_qx` vs `blazer_usb`. It's completely my fault, I assumed Arno did the same things on both `blazer_usb` and `nutdrv_qx` and I only checked them very quickly, and since I didn't experienced any related issue I didn't recheck that part of the code. Now, it should be fixed (but, maybe, we should put all those things in a common place to ease maintenance).

But, on the other issue, I'm not so sure we're the (only) ones to blame. @clepple has already said it all, but, just to add on it, as I see it, if even libusb developers, after all, bothered to make libusb-compat return the same values libusb 0.1 returned, maybe, just maybe, also FreeBSD should have chosen that path.

Also, the 'cure' currently used in NUT FreeBSD port doesn't seem quite right to me: in FreeBSD, if I understand it correctly, libusb 0.1 functions which are expected to return a negative value in case of errors, almost always return `-1` on errors (the other possible returned values are `-ETIMEDOUT` and `-ENXIO`, for the functions depending on `usb_std_io()`), not a libusb 1.0 error code.

blacklion, 2018-01-11:
@zykh "fault" is too strong, IMHO :-) I hope, transition to libusb-1.0 API will be completed in near future and libusb-1.0 error codes are well-defined and truly cross-platform.

jimklimov, 2021-12-01:
May be related to #137