networkupstools/nut #492: Add appdata metainfo for supported USB devices

bigon, 2017-10-24:
Hi,

Would be nice to have an appdata file with the devices supported by the drivers

Not sure if that file should be split per driver or if we could ship everything in one file

clepple, 2017-11-26:
@bigon can you please provide additional information (reference and possibly examples) for this request?

bigon, 2017-11-27:
I've been pointed to the following resources:

https://wiki.debian.org/AppStream/Guidelines
http://people.skolelinux.org/pere/blog/Using_appstream_with_isenkram_to_install_hardware_related_packages_in_Debian.html

clepple, 2017-11-27:
Thanks.

As a note to whoever gets around to this, it is recommended that the metadata *not* be licensed under the GPL: https://www.freedesktop.org/software/appstream/docs/chap-Metadata.html#tag-metadata_license