networkupstools/nut #495: OpenUPS driver reports wrong output current while charging

julien23dev, 2017-11-03:
While charging, the OpenUPS driver reports wrong ouput current as I know my load on output is near 20W (1,7A @12V):
battery.capacity: 100
battery.charge: 72
battery.charge.low: 5
battery.charge.warning: 20
battery.current: 4.210
battery.mfr.date: ?
battery.runtime: 3932100
battery.temperature: 40.50
battery.type: ?
battery.voltage: 13.78
device.mfr: Mini-Box.Com
device.model: OPEN-UPS
device.serial: PBSO4-LiFePO
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: openUPS HID 0.4
driver.version.internal: 0.41
input.current: 4.225
input.voltage: 23.96
**output.current: 1.080**
output.voltage: 12.01
ups.mfr: Mini-Box.Com
ups.model: OPEN-UPS
ups.productid: d004
ups.serial: PBSO4-LiFePO
**ups.status: OL CHRG**
ups.vendorid: 04d8

If I unplug the power and the board switch output on battery, the output current is correct :
battery.capacity: 100
battery.charge: 73
battery.charge.low: 5
battery.charge.warning: 20
battery.current: 653.640
battery.mfr.date: ?
battery.runtime: 39060
battery.temperature: 38.10
battery.type: ?
battery.voltage: 13.17
device.mfr: Mini-Box.Com
device.model: OPEN-UPS
device.serial: PBSO4-LiFePO
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: openUPS HID 0.4
driver.version.internal: 0.41
input.current: 0.000
input.voltage: 1.21
**output.current: 1.724**
output.voltage: 11.98
ups.mfr: Mini-Box.Com
ups.model: OPEN-UPS
ups.productid: d004
ups.serial: PBSO4-LiFePO
**ups.status: OB DISCHRG**
ups.vendorid: 04d8

If I use OpenUPS monitoring software, the output power (expressed un Watts) is always correct. I suspect the nut driver should use a different scaling factor according to charging state...

clepple, 2017-11-03:
I realize it is likely the problem is on the NUT side, but for reference, which version of firmware is installed on the OpenUPS?

julien23dev, 2017-11-03:
I also feel like it's a NUT issue. I use FW 1.8. Regards.

clepple, 2017-11-26:
@julien23dev Can you ask the OpenUPS vendor for the correct scale factors for FW 1.8 (and also the preferred way to parse/identify the FW version from the USB host side)? The NUT code already has two separate values for charge/discharge, and I believe those numbers trace back to a patch from Nicu Pavel (of Mini-Box.com) in 2012.

https://github.com/networkupstools/nut/blob/master/drivers/openups-hid.c#L38-L39