networkupstools/nut #499: nut-scanner on libusb-1.0+0.1: usb_get_string_simple() needs to be fixed for 1.0 API

clepple, 2017-11-26:
http://lists.alioth.debian.org/pipermail/nut-upsuser/2017-November/010981.html:

```
On Fedora 27, running commit 2999c95f0:

# nut-scanner
Cannot load USB library (/usr/lib64/libusb-1.0.so.0.1.0) : /usr/lib64/libusb-1.0.so.0.1.0: undefined symbol: usb_get_string_simple. USB search disabled.
```

aquette, 2017-11-28:
@clepple I've tested again libusb1.0 branch and reproduced the issue:
$ ./tools/nut-scanner/nut-scanner -U
Cannot load USB library (/lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0) : /lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0: undefined symbol: usb_get_string_simple. USB search disabled.

Then used fix_usb_scan_issue_499 and it worked:
[nutdev1]
	driver = "usbhid-ups"
	port = "auto"
	vendorid = "0463"
	productid = "FFFF"
	product = "5E"
	vendor = "EATON"
	bus = "002"

Good for me to merge into the libusb1.0 branch, thx

aquette, 2018-02-28:
assigned to @clepple since he made the fixed
ticket will be closed once we merge the libusb1 PR

closed 2022-01-11 by jimklimov