networkupstools/nut #501: nut-scanner: libusb-1.0 is not found on Raspbian jessie (8.x)

clepple, 2017-11-26:
Using https://github.com/networkupstools/nut/tree/libusb-1.0%2B0.1 @ 80bc452 and basic `./configure` options, the `get_libname()` function does not find libusb-1.0, which is located in `/lib/arm-linux-gnueabihf`.

Due to #500, this fails silently, regardless of the number of `-D` options passed in.

clepple, 2017-11-26:
This might be relevant, from `config.log`:
```
#define AUTOTOOLS_HOST_ALIAS "armv6l-unknown-linux-gnueabihf"
#define AUTOTOOLS_BUILD_ALIAS "armv6l-unknown-linux-gnueabihf"
#define AUTOTOOLS_TARGET_ALIAS "armv6l-unknown-linux-gnueabihf"
#define AUTOTOOLS_HOST_SHORT_ALIAS "armv6l-linux-gnueabihf"
#define AUTOTOOLS_BUILD_SHORT_ALIAS "armv6l-linux-gnueabihf"
#define AUTOTOOLS_TARGET_SHORT_ALIAS "armv6l-linux-gnueabihf"
```

jimklimov, 2017-11-27:
Interesting, so the autotools vars are `armv6l-*` but the system path is just `arm-*`?
Actually, can confirm such mismatch on `armv7l` as well.

Not sure if/when I'd get to investigating and fixing this, so feel free to overtake ;)