networkupstools/nut #505: Building PDF files in parallel with a2x often fails

clepple, 2017-11-28:
Ran into this while testing pull #418. A `make -j10` with PDF output enabled will often fail due to parallel a2x processes using the same names for intermediate files.

Proposed solution: uniquely named temporary directories.

StefanBruens, 2018-11-14:
In kicad this was "solved" by using a dependency between the various output formats. Thus different manuals are built in parallel, but e.g. user-manual.{txt,pdf} were built sequentially.

bigon, 2019-07-27:
Adding `.NOTPARALLEL:` at the end of the Makefile in the `docs/` directory workaround the problem

It will prevent that makefile to be run in parallel and prevent the build to fail

jimklimov, 2021-11-14:
Should be fixed with #1154 (and possibly other contemporary efforts)

closed 2021-11-14 by jimklimov