networkupstools/nut #506: "data stale" syslog frequency message by upsmon

mailinglists35, 2017-11-28:
If the usb cable is unplugged, upsmon sends the following message to syslog:

````Nov 28 20:43:35 term upsmon: Poll UPS [nutdev1@localhost] failed - Data stale````

I have read the manual pages but I couldn't find a way to suppress these messages or reduce the frequency.
I have hopefully set NOCOMMWARNTIME to 43200 and restarted upsmon but the message is still logged every five seconds.

Note that I *do* want a.s.a.p. notification that ups is stale but I do *NOT* want any repeat of that notification.

this is the upsmon.conf

````
MONITOR nutdev1@localhost 1 upsmon pass master
MINSUPPLIES 1
SHUTDOWNCMD "/usr/bin/logger shutdown"
POLLFREQ 5
POLLFREQALERT 5
HOSTSYNC 15
DEADTIME 15
POWERDOWNFLAG /etc/killpower
NOTIFYMSG ONLINE        "UPS %s on line power"
NOTIFYMSG ONBATT        "UPS %s on battery"
NOTIFYMSG LOWBATT       "UPS %s battery is low"
NOTIFYMSG FSD           "UPS %s: forced shutdown in progress"
NOTIFYMSG COMMOK        "Communications with UPS %s established"
NOTIFYMSG COMMBAD       "Communications with UPS %s lost"
NOTIFYMSG SHUTDOWN      "Auto logout and shutdown proceeding"
NOTIFYMSG REPLBATT      "UPS %s battery needs to be replaced"
NOTIFYMSG NOCOMM        "UPS %s is unavailable"
NOTIFYMSG NOPARENT      "upsmon parent process died - shutdown impossible"
NOTIFYFLAG ONLINE       SYSLOG+WALL+EXEC
NOTIFYFLAG ONBATT       SYSLOG+WALL+EXEC
NOTIFYFLAG LOWBATT      SYSLOG+WALL+EXEC
NOTIFYFLAG FSD  SYSLOG+WALL+EXEC
NOTIFYFLAG COMMOK       SYSLOG+WALL+EXEC
NOTIFYFLAG COMMBAD      SYSLOG+WALL+EXEC
NOTIFYFLAG SHUTDOWN     SYSLOG+WALL+EXEC
NOTIFYFLAG REPLBATT     SYSLOG+WALL+EXEC
NOTIFYFLAG NOCOMM       SYSLOG+WALL+EXEC
NOTIFYFLAG NOPARENT     SYSLOG+WALL+EXEC
RBWARNTIME 43200
NOCOMMWARNTIME 43200
FINALDELAY 5
NOTIFYCMD /usr/sbin/upssched
````

mailinglists35, 2017-11-29:
the same for this message for upsmon running on a monitoring-only instance of upsmon (every POLLFREQ  seconds)

````upsmon: UPS [nutdev1@remote-host]: connect failed: Connection failure: Connection refused````

is there any way to keep polling but warn **ONLY ONCE**, aka suppress subsequent identical messages? I could workaround this at the syslog level but I have a feeling that is not the right way to fix it.

jimklimov, 2017-12-04:
I think the design stance was that if you are doing administrative activity (e.g. UPS is in repairs and you know it won't be available until further notice) then you just comment away the `MONITOR` line in `upsmon.conf` and restart the client.

These log messages are there to bring your attention to the problem, perhaps annoy the human and make it intervene and analyze the situation - and make some meaningful reaction, such as fixing the cable, logging a bug against the driver that locked up (maybe the HW driver in OS, not NUT driver), comment away the `MONITOR` line...

In this sense, I'd say it is not a bug but feature ;)

mailinglists35, 2017-12-05:
I need to centrally monitor hundreds of point of sale machines connected to ups-es so support staff will get notified when communication with ups is gone or when battery has failed. manually intervening for each event is not a viable option in this case.

jimklimov, 2023-01-07:
Cheers, sorry it took a while of collecting dust - PR with the feature posted.

PR #1784 (2023-01-08): upsmon: Introduce POLLFAIL_LOG_THROTTLE_MAX
Closes: #506

````
Jan 07 22:49:50 pve nut-monitor[4103053]:   97.199079        Reloading configuration
Jan 07 22:49:50 pve nut-monitor[4103053]:   97.199178        Applying debug_min=2 from upsmon.conf
...

Jan 07 22:50:26 pve nut-monitor[4106871]:    0.001183        [D1] Trying to connect to UPS [eco650]
Jan 07 22:50:26 pve nut-monitor[4106871]:    0.001410        [D1] Logged into UPS eco650
Jan 07 22:50:26 pve nut-monitor[4106871]:    0.001440        [D2] pollups: eco650
Jan 07 22:50:26 pve nut-monitor[4106871]:    0.001472        Poll UPS [eco650 [... cut]
closed 2023-01-08 by jimklimov