networkupstools/nut #510: Is CyberPower OR1500LCDRTXL2U supported ?

ikus060, 2017-12-20:
Hello,

I own a CuberPower OR1500LCDRTXL2U. I hope this model is supported by Nut. I was expecting to use the `usbhid-ups` driver, but the usb on this device doesn't seams to be working. in `dmesg` the device get disconnected without reason:
```
[12217182.121445] usb 4-2: new low-speed USB device number 4 using xhci_hcd
[12217182.347217] usb 4-2: New USB device found, idVendor=0764, idProduct=0601
[12217182.347249] usb 4-2: New USB device strings: Mfr=3, Product=1, SerialNumber=0
[12217182.347277] usb 4-2: Product: OR1500LCDRTXL2U
[12217182.347294] usb 4-2: Manufacturer: CPS
[12217182.347688] usb 4-2: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
[12217182.347727] usb 4-2: ep 0x2 - rounding interval to 128 microframes, ep desc says 160 microframes
[12217182.537894] hid-generic 0003:0764:0601.0006: hiddev0,hidraw0: USB HID v1.10 Device [CPS OR1500LCDRTXL2U] on usb-0000:07:00.0-2/input0
[12217188.897175] usb 4-2: USB disconnect, device number 4
```

Next, I'm trying with the `powerpanel` driver, without success:
```
$ sudo /lib/nut/powerpanel -a cyberpower2 -D -D -D -D
Network UPS Tools - CyberPower text/binary protocol UPS driver 0.26 (2.7.2)
Warning: This is an experimental driver.
Some features may not function correctly.

   0.000000	debug level is '4'
   0.012680	Trying binary protocol...
   0.214777	send: (2 bytes) => 46 0d
   0.665520	read: timeout
   0.665591	  \_: (0 bytes) =>
   0.766291	send: (2 bytes) => 46 0d
   1.217156	read: timeout
   1.217228	  \_: (0 bytes) =>
   1.317908	send: (2 bytes) => 46 0d
   1.768766	read: timeout
   1.768837	  \_: (0 bytes) =>
   1.789727	Trying text protocol...
   1.891107	send: (2 bytes) => 0d 0d
   2.241970	read: timeout
   2.242041	  \_: (0 bytes) =>
   2.393189	send: (3 bytes) => 50 34 0d
   2.568379	read: (51 bytes) => 23 4f 52 31 35 30 30 4c 43 44 52 54 58 4c 32 55 2c 30
   2.568543	 30 31 2e 30 2c 42 46 4a 38 31 30 34 2d 34 39 31 20 2c 43 79 62 65 72 50 6f
   2.568609	 77 65 72 20 20 20 20 20
   2.568670	CyberPower UPS with text protocol on /dev/ttyUSB2 detected
   2.719816	send: (3 bytes) => 50 33 0d
   2.820301	read: (24 bytes) => 23 31 32 2c 34 78 31 2c 37 2c 30 2c 34 2c 39 2c 31 32
   2.820428	 2c 34 78 32 2c 39
   2.971523	send: (3 bytes) => 50 32 0d
   3.071887	read: (22 bytes) => 23 31 35 30 30 2c 31 31 32 35 2c 31 32 30 2c 30 34 37
   3.071962	 2c 30 36 33
   3.222931	send: (3 bytes) => 50 31 0d
   3.323415	read: (18 bytes) => 23 31 32 30 2c 31 34 30 2c 39 30 2c 31 30 2c 33 30 30
   3.474339	send: (3 bytes) => 50 36 0d
   3.604037	read: (40 bytes) => 23 31 33 36 2c 31 33 37 2c 31 33 38 2c 31 33 39 2c 31
   3.604131	 34 30 2c 31 34 31 2c 31 34 32 2c 31 34 33 2c 31 34 34 2c 31 34 35
   3.755223	send: (3 bytes) => 50 37 0d
   3.855655	read: (30 bytes) => 23 38 36 2c 38 37 2c 38 38 2c 38 39 2c 39 30 2c 39 31
   3.855724	 2c 39 32 2c 39 33 2c 39 34 2c 39 35
   4.006801	send: (3 bytes) => 50 38 0d
   4.107270	read: (27 bytes) => 23 31 30 2c 32 30 2c 33 30 2c 34 30 2c 35 30 2c 36 30
   4.107374	 2c 37 30 2c 38 30 2c 39 30
   4.258463	send: (3 bytes) => 50 35 0d
   4.358938	read: (4 bytes) => 23 31 32 30
   4.510058	send: (3 bytes) => 50 39 0d
   4.610584	read: (4 bytes) => 23 c3 d7 f0
   4.711389	send: (2 bytes) => 43 0d
   4.811863	read: (4 bytes) => 23 2d 31 30
   4.912685	send: (2 bytes) => 44 0d
   5.113156	read: (35 bytes) => 23 49 31 31 36 2e 30 4f 31 31 36 2e 30 4c 30 33 36 42
   5.113229	 31 30 30 46 30 36 30 2e 30 52 30 30 33 32 53 80 84
   5.363801	read: timeout
   5.363892	  \_: (23 bytes) => c0 88 80 0d 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   5.363912	 00 00 00 00 00
   5.363927	Communications with UPS lost: Status read failed!
   5.364133	dstate_init: sock /var/run/nut/powerpanel-cyberpower2 open on fd 5
   5.464966	send: (2 bytes) => 44 0d
   5.665390	read: (35 bytes) => 23 49 31 31 35 2e 30 4f 31 31 35 2e 30 4c 30 34 32 42
   5.665464	 31 30 30 46 30 36 30 2e 30 52 30 30 33 33 53 80 84
   5.916017	read: timeout
   5.916108	  \_: (23 bytes) => c0 88 80 0d 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   5.916127	 00 00 00 00 00
   5.916142	Communications with UPS lost: Status read failed!
```

clepple, 2017-12-21:
We have an entry for the [OR700LCDRM1U](http://networkupstools.org/ddl/Cyber_Power_Systems/OR700LCDRM1U.html) using usbhid-ups in the DDL, and that has the same USB IDs as the OR1500LCDRTXL2U. So I suspect that usbhid-ups is the correct driver (though it might be possible to extend powerpanel to cover this model as well).

Maybe this disconnection issue? http://lists.alioth.debian.org/pipermail/nut-upsuser/2012-October/007983.html

ikus060, 2017-12-21:
Hello Charles,

In my case, the behaviour seams a bit different. When plugged using USB cable, the device is disconnecting after ~8sec, and their not loop. The device is not trying to connect again. It simply disconnect and that all. To make it reconnect, I need to power cycle the UPS.

With few trial and error, I manage to powercycle the UPS and start the `usbhid-ups` driver at the right moment. I may try to automate this somehow. But clearly it's not a viable solution and may need human intervention if the timing is incorrect.

From my point of view, the USB interface for this device is not usable with nut. Will also try to open a case with CyberPower to know more about this "feature".

The serial interface on the other hand seams more stable and always available. But the `powerpanel` driver provided by nut doesn't seams to have proper support for my device. Would it be possible to work with someone on this ? I have a fairly good background in development, but my C a bit rusty.

ikus060, 2018-03-15:
Would love to have feedback for my PR #538.

jimklimov, 2021-11-07:
Should be supported after the merge of #538 (so with builds of current NUT master branch, or eventually with release 2.7.5 and newer).

closed 2021-11-07 by jimklimov