networkupstools/nut #515: Eaton 5E 850i: problem loading usbhid-ups driver

tomaszg7, 2018-01-11:
Hi, I'm using NUT 2.7.4 with Eaton 5E 850i UPS on Gentoo (kernel 4.12.12, libusb-1.0.21). Rules in `/etc/hotplug` and `/lib/udev/rules.d/` both seem to be installed, but I have problem loading the driver - it usually doesn't start the first time I try it, but after a few times it usually loads correctly. In this way, each time I boot my computer I have to make sure manually that the driver is loaded. I'd be grateful for any help with the issue.

Here's the example of log when the driver didn't load correctly:
```
/lib64/nut/usbhid-ups -a eaton -DDD
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '3'
   0.000874     upsdrv_initups...
[....]
   0.041393     Checking device (0463/FFFF) (009/002)
   0.045264     - VendorID: 0463
   0.045295     - ProductID: ffff
   0.045302     - Manufacturer: unknown
   0.045309     - Product: unknown
   0.045315     - Serial Number: unknown
   0.045322     - Bus: 009
   0.045328     - Device release number: 0001
   0.045334     Trying to match device
   0.045359     Device matches
   0.045388     failed to claim USB device: Device or resource busy
   0.045599     detached kernel driver from USB device...
   0.045664     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.047216     Unable to get HID descriptor (Broken pipe)
   0.047252     HID descriptor length (method 1) -1
   0.047271     HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 25 02
   0.047281     HID descriptor length (method 2) 549
   0.047290     HID descriptor length 549
   0.049217     Unable to get Report descriptor: Broken pipe
[....]
   0.323563     No appropriate HID device found
   0.323582     No matching HID UPS found
```

Here's a part of output when the driver did connect:
```
  0.043309     Checking device (0463/FFFF) (009/004)
   0.314141     - VendorID: 0463
   0.314175     - ProductID: ffff
   0.314182     - Manufacturer: EATON
   0.314188     - Product: 5E
   0.314194     - Serial Number: unknown
   0.314200     - Bus: 009
   0.314206     - Device release number: 0001
   0.314212     Trying to match device
   0.314227     Device matches
   0.314292     failed to claim USB device: Device or resource busy
   0.314496     detached kernel driver from USB device...
   0.314526     nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.383141     HID descriptor, method 1: (9 bytes) => 09 21 10 01 21 01 22 25 02
   0.383167     HID descriptor length (method 1) 549
   0.383192     HID descriptor, method 2: (9 bytes) => 09 21 10 01 21 01 22 25 02
   0.383203     HID descriptor length (method 2) 549
   0.383213     HID descriptor length 549
   2.341016     Report Descriptor size = 549
   2.341068     Report Descriptor: (549 bytes) => 05 84 09 04 a1 01 09 1e a1 84 09 43 85 0d
   2.341089      75 10 95 01 15 00 27 ff ff 00 00 66 21 d1 55 07 b1 83 09 42 75 08 26 ff 00
   2.341113      66 01 f0 55 00 b1 83 09 40 85 12 67 21 d1 f0 00 55 07 b1 83 09 1f 85 0b 65
   2.341142      00 55 00 b1 03 c0 09 18 a1 00 09 20 a1 81 09 1f b1 03 09 21 b1 03 09 02 a1
   2.341162      02 09 6c 85 0c b1 03 c0 c0 09 19 85 0b b1 03 c0 09 16 a1 00 09 1a a1 81 09
   2.341180      1b 85 0f b1 03 09 02 a1 02 09 6e 85 02 b1 83 09 6e 81 83 c0 09 30 85 0e 75
   2.341199      10 27 ff ff 00 00 67 21 d1 f0 00 55 07 b1 83 c0 09 1a a1 83 09 1b 85 0f 75
   2.341218      08 26 ff 00 65 00 55 00 b1 03 06 ff ff 09 47 b1 83 c0 05 84 09 1c a1 00 09
   2.341236      32 85 0e 75 10 27 ff ff 00 00 66 01 f0 55 0f b1 83 09 1d 85 0f 75 08 26 ff
   2.341268      00 65 00 55 00 b1 03 09 30 85 0e 75 10 27 ff ff 00 00 67 21 d1 f0 00 55 07
   2.341286      b1 83 c0 09 17 85 0f 75 08 26 ff 00 65 00 55 00 b1 03 c0 09 24 a1 00 09 5a
   2.341303      85 14 b1 82 05 85 09 2c 85 0c b1 03 05 84 09 57 85 09 75 10 15 ff 26 ff 7f
   2.341322      66 01 10 b1 82 05 85 09 83 85 0c 75 08 15 00 26 ff 00 65 00 b1 03 05 84 09
   2.341340      1f 85 0f 65 00 b1 03 05 85 09 67 85 0c 65 00 b1 03 09 89 85 10 65 00 b1 03
   2.341359      05 84 09 fd b1 03 06 ff ff 09 f0 b1 03 05 84 09 fe b1 03 09 ff b1 03 06 ff
   2.341377      ff 09 f1 b1 03 05 84 09 35 85 07 65 00 b1 83 09 25 85 0f 65 00 b1 03 09 02
   2.341395      a1 02 05 85 09 d0 85 01 75 01 25 01 b1 83 09 d0 81 83 09 42 b1 83 09 42 81
   2.341413      83 09 44 b1 83 09 44 81 83 05 84 09 73 b1 83 09 73 81 83 05 85 09 45 b1 83
   2.341431      09 45 81 83 05 84 09 61 b1 83 09 61 81 83 09 62 b1 83 09 62 81 83 05 85 09
   2.341448      4b b1 83 09 4b 81 83 05 84 09 65 75 08 26 ff 00 b1 83 09 65 81 83 09 69 b1
   2.341466      83 09 69 81 83 c0 05 85 09 66 85 06 65 00 b1 83 09 66 81 83 09 68 75 10 27
   2.341484      ff ff 00 00 66 01 10 b1 83 09 68 81 83 05 84 09 30 85 07 75 08 26 ff 00 67
   2.341497      21 d1 f0 00 55 07 b1 83 c0 c0
   2.341608     Using subdriver: MGE HID 1.39
   2.341622     57 HID objects found
[...]
```

Output of `lsusb` before loading the driver:
```
lsusb -v:

Bus 009 Device 002: ID 0463:ffff MGE UPS Systems UPS
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0463 MGE UPS Systems
  idProduct          0xffff UPS
  bcdDevice            0.01
  iManufacturer           1 (error)
  iProduct                2 (error)
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               20mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     549
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              20
Device Status:     0x0001
  Self Powered
```

clepple, 2018-01-12:
We do have configuration parameters to retry starting the driver if it does not detect the UPS. See `maxretry` and `retrydelay` here: http://networkupstools.org/docs/man/ups.conf.html

I know this sounds like it is masking the underlying problem, but we are almost finished rewriting the NUT USB code (#300) to actually use the libusb-1.0 API (your system is probably using libusb-compat), so if nothing else, the new code may have slightly different symptoms.

You can also check for other daemons which might be probing the UPS in undesirable ways (maybe ModemManager, or something probing for USB HID devices).

tomaszg7, 2018-01-13:
Ah, since I have `libusb-compat-0.1.5` then it might be a source of the trouble. Since I don't have a more recent version of the old lib in my distro, I'll wait for the code rewrite to land. Thanks.

aquette, 2018-01-15:
hi @tomaszg7 
since you're running gentoo, you may give a try to a GH downloaded version:
https://github.com/zykh/nut/tree/libusb-1.0%2B0.1
Also, according to your test results (and I may do some too), I may had some retries for the initial connection and descriptors retrieval

tomaszg7, 2018-01-15:
For now using retries works for me rather well. I can patiently wait for the code to mature. However (I didn't understand fully) if you want me to build and test it, I will try to do it and report back.

alstef, 2018-02-13:
Having the same issue on Ubuntu 16.04 LTS with Eaton 5E 850i only the driver never loads regardless of number of tries. The odd thing is that another Eaton 5E 650i works fine. Wondering if product identification in `lsusb -v` has anything to do with it since the 850 returns
```
iManufacturer           1 (error)
iProduct                2 (error)
```
instead of 
```
iManufacturer           1 EATON
iProduct                2 5E
```
Maybe this throws off the driver somehow. Using `libusb-1.0-0:amd64` on kernel `Linux 4.4.0-112-generic`. Will try a newer kernel and see what happens since it kinda looks like an USB issue

alstef, 2018-02-13:
There's also this suggestion https://askubuntu.com/questions/966070/eaton-mge-5e-ups-driver-not-working that the cable might be at fault, guess will try a new cable also :(

alstef, 2018-02-13:
Yup, cable problem. Switched cables now both work.

clepple, 2018-02-15:
@alstef if you run into this again, the first place to check for low-level USB problems is `dmesg`. The timeout errors (-110, if memory serves) when the kernel is retrying will show up there first, and eventually will propagate to userspace.

The `lsusb` error could be at that level, or higher. The `iManufacturer` and `iProduct` numbers are in the first descriptor fetched from the device, but the strings need to be retrieved separately. It is not uncommon for the first request to succeed, and then to have another error occur that takes a while to recover from.

linuxlurak, 2018-04-05:
Something similar happend to me:

> Apr  5 14:02:38 localhost upsmon[1915]: Poll UPS [eaton3s@localhost] failed - Driver not connected
> Apr  5 14:02:53 localhost upsmon[1915]: message repeated 3 times: [ Poll UPS [eaton3s@localhost] failed - Driver not connected]
> 
> ..
> 
> Apr  5 14:03:05 localhost nut-server[2134]:  * Starting NUT - power devices information server and drivers
> Apr  5 14:03:08 localhost upsmon[1915]: Poll UPS [eaton3s@localhost] failed - Write error: Broken pipe
> Apr  5 14:03:08 localhost upsmon[1915]: Communications with UPS eaton3s@localhost lost

I changed the USB cable. The delivered one looked quiet cheap. But no improvement.

Then I ran  `upsdrvctl start`. This gave me:

> Network UPS Tools - UPS driver controller 2.7.2
> Network UPS Tools - Generic HID driver 0.38 (2.7.2)
> USB communication driver 0.32
> 
> Fatal error: 'maxretry' is not a valid variable name for this driver.
> 
> Look in the man page or call this driver with -h for a list of
> valid variable names and flags.

So I commented maxretry in /etc/nut/usp.conf and, what, you think, happend? See this:
`upsdrvctl start`:

> Network UPS Tools - UPS driver controller 2.7.2
> Network UPS Tools - Generic HID driver 0.38 (2.7.2)
> USB communication driver 0.32
> Using subdriver: MGE HID 1.33

and after `service nut-server restart`:

> Apr  5 14:03:08 localhost upsmon[1915]: Communications with UPS eaton3s@localhost lost
> Apr  5 14:03:10 localhost usbhid-ups[2144]: Startup successful
> Apr  5 14:03:10 localhost upsd[2145]: listening on 127.0.0.1 port 3493
> Apr  5 14:03:10 localhost upsd[2145]: Connected to UPS [eaton3s]: usbhid-ups-eaton3s
> Apr  5 14:03:10 localhost upsd[2146]: Startup successful
> Apr  5 14:03:10 localhost nut-server[2134]:    ...done.
> Apr  5 14:03:10 localhost systemd[1]: Started LSB: Network UPS Tools initscript.
> Apr  5 14:03:13 localhost upsd[2146]: User admin@127.0.0.1 logged into UPS [eaton3s]
> Apr  5 14:03:13 localhost upsmon[1915]: Communications with UPS eaton3s@localhost established

This state survived a reboot. 

I don't know if this info helps.

csamsel, 2018-04-05:
I have a similiar if not the same problem with my Eaton unit, a Eclipse Eco 800. I always thought it's because i'm using USB passthrough to a VMware ESXi VM making USB flacky - but it might be unrelated after all.

Im using Gentoo and its nut package too. Maybe that's the offender?

In my case the upsdrv always often starts correctly, but crashes/disappears after a few days.

csamsel, 2018-04-11:
I've been using a different USB cable instead of the flimsy one that came with the unit - no disconnects so far. Sorry for the noise. Although some kind of error handling and an automatic recovery of the driver would still be nice.

tomaszg7, 2018-04-11:
I didn't replace the cable. Once I set `maxretry` I don't have any problems in initialization nor any disconnections afterwards.

csamsel, 2018-04-11:
@aquette I am not 100% sure it has been fixed, but it seems so; 
Usually it took a few days for a problem to occur, it's been a week now with a different cable and no problems so far.

For reference these are log files of occurred problems. All could be fixed be restarting the nut driver.

```
Jan  3 19:32:18 oxygen-system-vm upsd[4678]: Data for UPS [eaton] is stale - check driver
Jan  3 19:32:33 oxygen-system-vm upsmon[5302]: Poll UPS [eaton@localhost] failed - Data stale
Jan  3 19:32:33 oxygen-system-vm upsmon[5302]: Communications with UPS eaton@localhost lost
Jan  3 19:32:38 oxygen-system-vm upsd[4678]: UPS [eaton] data is no longer stale
Jan  3 19:33:03 oxygen-system-vm upsmon[5302]: Communications with UPS eaton@localhost established
...
Jan  5 21:13:36 oxygen-system-vm upsmon[5303]: Startup successful
Jan  5 21:13:36 oxygen-system-vm upsd[4679]: User upsmon@::1 logged into UPS [eaton]
Jan  5 21:13:36 oxygen-system-vm upsmon[5304]: Poll UPS [eaton@localhost] failed - Driver not connected
Jan  5 21:13:36 oxygen-system-vm upsmon[5304]: Communications with UPS eaton@localhost lost
Jan  5 21:13:41 oxygen-system-vm upsd[4679]: User upsmon@192.168.2.12 logged into UPS [eaton]
Jan  5 21:13:41 oxygen-system-vm upsmon[5304]: Poll UPS [eaton@localhost] failed - Driver not connected
Jan  5 21:13:41 oxygen-system-vm upsmon[5304]: UPS eaton@localhost is unavailable
Jan  5 21:13:46 oxygen-system-vm upsmon[5304]: Poll UPS [eaton@localhost] failed - Driver not connected
Jan  5 21:13:51 oxygen-system-vm upsmon[5304]: Poll UPS [eaton@localhost] failed - Driver not connected
..

Mar 12 02:44:29 oxygen-system-vm usbhid-ups[4554]: libusb_get_report: Input/output error
Mar 12 02:44:29 oxygen-system-vm upsd[4591]: Data for UPS [eaton] is stale - check driver
Mar 12 02:44:31 oxygen-system-vm usbhid-ups[4554]: Can't claim USB device [0463:ffff]: No such file or directory
Mar 12 02:44:31 oxygen-system-vm upsd[4591]: Can't connect to UPS [eaton] (usbhid-ups-eaton): No such file or directory
Mar 12 02:44:43 oxygen-system-vm upsmon[5148]: Poll UPS [eaton@localhost] failed - Driver not connected
Mar 12 02:44:43 oxygen-system-vm upsmon[5148]: Communications with UPS eaton@localhost lost
Mar 12 02:45:13 oxygen-system-vm upsmon[5148]: Poll UPS [eaton@localhost] failed - Driver not connected
...
```

linuxlurak, 2018-04-11:
@aquette I'm on a new cable at the moment. No issues since I last posted.

henfri, 2019-08-18:
Hello,

I have the same (?) Issue: 
After reboot, the device does not show up. 
lsusb -v does not show the device. Unplugging and re-plugging the device does fix this.

I have switched to a different USB-Cable -without success. Any Idea how I could fix this? Can I re-trigger whatever I retrigger by unplugging by software instead of doing it physically?

Regards,
Hendrik

donPabloNow, 2022-03-09:
Similarly to the last example, I was unable to use the service owing to technical issues with my USB cable. Fortunately, I came across this topic and decided to try a new cable, which proved to be the solution I was seeking. Thank you to everyone who shared their experiences, you helped me get to the bottom of this somewhat perplexing issue - hardware issues, who would have guessed it lol

jimklimov, 2022-04-10:
Just in case, also linking to #630 about a different type of problem that can be seen between Eaton 5E and linux

closed 2024-09-24 by jimklimov