networkupstools/nut #516: OpenUPS v1.9 firmware reports wrong battery current

rmk92, 2018-01-27:
OpenUPS v1.9 firmware appears to report positive values for the battery current whether the battery is charging or discharging.  The USB poewr HID specifications say that the current is supposed to be positive when charging and negative while discharging.

In my case, I have 12V output, ,12V input, and a 12V 7Ah battery connected.

Investigating this issue (and other issues) with nut-2.7.2 with OpenUPS, it appears that the HID Report Descriptor is non-conformant with the HID specification in many respects:

1. It fails to set an initial logical minimum/maximum values, leading to the first few report IDs being clamped to zero.
2. It fails to set the units and unit exponent _before_ the feature/input/output item that it corresponds with (leading to some fun where (eg) UPS.PowerStatus.ConfigVoltage is unit-less, UPS.PowerStatus.ConfigCurrent has units of Volts, UPS.PowerStatus.Temperature gets units of Amps, and the PresentStatus flags get units of Volts (meaning some get a value of 0.01).
3. It fails to set the logical minimum/maximum for the UPS.PowerStatus.Current correctly - as mentioned above, this is reported as a signed value, but the logical minimum is zero, the logical maximum is 65535 - this should be min -32768 max 32767.

These issues can be fixed by arranging for nut to accept a replacement HID Report Descriptor with fixes in place, and this allows me to get sensible UPS.PowerStatus.Current values.

Other firmware issues:

1. It reports UPS.PowerStatus.PresentStatus.BattteryPresent as zero even when a battery is connected.
2. UPS.PowerStatus.Input.Current reports absurd values which suggest that this UPS is an energy creation unit - with an output current of 1.3A, battery current of 0A, the input current can be reported as 0.03A.

Note that UPS.PowerStatus.Output.Voltage and UPS.PowerStatus.Output.Current appear to have correct Units/UnitExp, with a logical min/max of 0/65535, so this is not the issue reported in another bug report on this device.

I am attempting to talk to Mini-Box.Com about this, but they have yet to reply to my emails earlier this week covering some of this.  However, if they remain unresponsive, having a method where the HID report descriptor can be replaced may be the only real solution.

I'm including this here so that people in the NUT project are aware of the issues with this hardware.

clepple, 2018-01-27:
> having a method where the HID report descriptor can be replaced may be the only real solution.

Agreed. Let us know if you have thoughts on a general way to do this. Some other notes on the topic are linked to this issue:

https://github.com/networkupstools/nut/issues/169

davidwkerr, 2022-07-20:
Inaction on this is a real pity. I have 4 openups units. Hardware is great. I have learned to ignore the input and output reported currents which are wrong. Mini Box does not seem to care.

jimklimov, 2022-10-11:
I suppose nobody with the units contributed an updated driver?

That said, recent improvements in master branch included facilities that
can help:
1) Detection of mismatched LogMin/LogMax (#1024 etc IIRC)
2) Report descriptor fixups for known-broken cases - you may want to extend
that for the driver.

jimklimov, 2024-12-16:
FWIW, some work was done about general tech for hot-patching the presumed-broken HID Report Descriptors, most recently touched in PR #2718 - someone with the device and a bit of programming courage can check the `fix_report_desc` approach and define the needed logic in the respective subdriver.