networkupstools/nut #517: Model: PC-1000VA not connecting

lyda, 2018-01-28:
I've tried the `blazer_usb`, `nutdrv_qx`, `nutdrv_atcl_usb` drivers but with no success. Searching this repo it seems that the best option is `nutdrv_atcl_usb` for a Powercool PC-1000VA model, but when I run it with this `/etc/nul/ups.conf` file:

```
[fry1000]
	driver = nutdrv_atcl_usb
	vendor = 0001
	desc = "Fry 1000VA UPS"
        port = auto
```

I have this happen:

```
# /lib/nut/nutdrv_atcl_usb -DD -a fry1000
Network UPS Tools - 'ATCL FOR UPS' USB driver 1.11 (2.7.4.1)
Warning: This is an experimental driver.
Some features may not function correctly.

   0.000000	debug level is '2'
   0.001137	Searching for USB device...
Segmentation fault (core dumped)
```

```
# /lib/nut/nutdrv_qx -DD -a fry1000
Network UPS Tools - Generic Q* USB/Serial driver 0.29 (2.7.4.1)
USB communication driver (libusb 1.0) 0.02
   0.000000	debug level is '2'
   0.001348	upsdrv_initups...
   0.007621	Checking device (1D6B/0002)
   0.007680	Failed to open device, skipping. (Access denied (insufficient permissions))
   0.007805	Checking device (1D6B/0001)
   0.007822	Failed to open device, skipping. (Access denied (insufficient permissions))
   0.007847	Checking device (067B/25A1)
   0.007860	Failed to open device, skipping. (Access denied (insufficient permissions))
   0.007867	Checking device (1D6B/0002)
   0.007878	Failed to open device, skipping. (Access denied (insufficient permissions))
   0.007885	Checking device (1D6B/0001)
   0.007896	Failed to open device, skipping. (Access denied (insufficient permissions))
   0.007902	Checking device (1D6B/0002)
   0.007914	Failed to open device, skipping. (Access denied (insufficient permissions))
   0.007924	Checking device (0001/0000)
   0.011196	- VendorID: 0001
   0.011238	- ProductID: 0000
   0.011244	- Manufacturer: unknown
   0.011255	- Product: MEC0003
   0.011260	- Serial Number: unknown
   0.011391	- Bus: 004
   0.011400	- Device release number: 0100
   0.011410	Trying to match device
   0.011475	Device matches
   0.011487	Reading first configuration descriptor
   0.011505	auto detached kernel driver from USB device
   0.011537	Claimed interface 0 successfully
   0.011593	qx_process_answer: short reply (input.voltage)
   0.011609	qx_process_answer: short reply (input.voltage)
   0.011629	qx_process_answer: short reply (input.voltage)
   0.011648	voltronic_qs_protocol: invalid protocol [M]
   0.011663	voltronic_qs_protocol: invalid protocol [M]
   0.011675	voltronic_qs_protocol: invalid protocol [M]
   0.011693	voltronic_qs_hex_protocol: invalid protocol [M]
   0.011706	voltronic_qs_hex_protocol: invalid protocol [M]
   0.011718	voltronic_qs_hex_protocol: invalid protocol [M]
   0.011735	qx_process_answer: short reply (input.voltage)
   0.011747	qx_process_answer: short reply (input.voltage)
   0.011758	qx_process_answer: short reply (input.voltage)
   0.011777	qx_process_answer: short reply (input.voltage)
   0.011789	qx_process_answer: short reply (input.voltage)
   0.011799	qx_process_answer: short reply (input.voltage)
   0.017190	received 47 (40)
[... 663 lines cut ...]
            Item(Main  ): End Collection, data=none
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)
```

clepple, 2018-01-28:
despite the bogus USB VID:PID (0001:0000), the USB HID Report Descriptor looks long enough to be real. Please check out this section in the developer docs:

http://new.networkupstools.org/docs/developer-guide.chunked/ar01s04.html#hid-subdrivers

or if that host is down for some reason: http://networkupstools.org/docs/developer-guide.chunked/ar01s04.html#hid-subdrivers

"How to make a new subdriver to support another USB/HID UPS"

zykh, 2018-01-28:
What's wrong with `nutdrv_qx`? It seems to be working, with protocol `Q1` and with the `krauler` subdriver (other `MEC0003` devices work with the similar `fabula` subdriver, but this device apparently doesn't report an 'iManufacturer' string and so it ends up with the `krauler` one -- you should try them both and see which one works better).

clepple, 2018-01-29:
@lyda ignore my previous message; I saw the SI units in the HID descriptor and didn't read the other debug messages.

@zykh is right  - the `nutdrv_qx` output indicates that the driver started its main loop (it will not go into the background if you pass one or more `-D` flags.

deaddox, 2018-04-20:
Hi Did you ever manage to get the PowerCool working with NUT? If so could you share a copy of your config?

jammmyb, 2021-04-13:
Any joy on this?

been researching trying to get this to talk to NUT under unRAID. 

For a cheap effective UPS, it’d be great to get it working!