networkupstools/nut #519: PowerWalker VI 3000 RT HID Does not work

evildemons, 2018-01-30:
Hi,

I have tried but it does not work to get out the mass of "null" in the log.

`20180130 143109 NA NA NA [NA] NA NA                                             
20180130 143609 NA NA NA [NA] NA NA                                             
20180130 144109 NA NA NA [NA] NA NA                                             
20180130 144609 NA NA NA [NA] NA NA                                             
20180130 145109 NA NA NA [NA] NA NA                                             
20180130 145609 NA NA NA [NA] NA NA                                             
20180130 150109 NA NA NA [NA] NA NA                                             
20180130 150609 NA NA NA [NA] NA NA                                             
20180130 151109 NA NA NA [NA] NA NA                                             
20180130 151609 NA NA NA [NA] NA NA `

`lure: Connection refused                                                        
Jan 30 15:19:26 freenas upsmon[81220]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Jan 30 15:19:31 freenas upsmon[81220]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Jan 30 15:19:36 freenas upsmon[81220]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Jan 30 15:19:41 freenas upsmon[81220]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Jan 30 15:19:47 freenas upsmon[81220]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Jan 30 15:19:52 freenas upsmon[81220]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Jan 30 15:19:57 freenas upsmon[81220]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Jan 30 15:20:02 freenas upsmon[81220]: UPS [ups]: connect failed: Connection fai
lure: Connection refused`

https://www.powerwalker.com/?lang=en&page=product&item=10120024

clepple, 2018-02-01:
The intended sequence of connections looks like this:

upslog/upsmon -> upsd -> driver

The "connection refused" message from upsmon probably means that upsd is not running (although it might also be caused by an incorrect IP address in the MONITOR line).

That said, some systems will not start upsd until the driver is running correctly. Without knowing more about your system (OS, version of NUT, etc) it is difficult to predict why the driver hasn't started.

evildemons, 2018-02-01:
FreeNAS 11-U1 (freebsd) Network UPS Tools upsc 2.7.4

clepple, 2018-02-03:
I can’t remember how much FreeNAS differs from a ports-based installation, so I don’t know the exact path to the drivers and logs. I would recommend checking the logs for any information as to why upsd is not running, then check for usbhid-ups. (I am making assumptions based on “HID” in the name)

evildemons, 2018-02-03:
@clepple 

I've also tried with usbhid-ups do not seem to work and I do not know what I'm doing wrong either.

If that does not work, I can buy another UPS.

`Feb  3 20:41:25 freenas upsmon[91532]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Feb  3 20:41:31 freenas upsmon[91532]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Feb  3 20:41:36 freenas upsmon[91532]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Feb  3 20:41:41 freenas upsmon[91532]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Feb  3 20:41:46 freenas upsmon[91532]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Feb  3 20:41:51 freenas upsmon[91532]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Feb  3 20:41:56 freenas upsmon[91532]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Feb  3 20:42:02 freenas upsmon[91532]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Feb  3 20:42:02 freenas upsmon[91532]: UPS ups is unavailable                   
Feb  3 20:42:07 freenas upsmon[91532]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Feb  3 20:42:12 freenas upsmon[91532]: UPS [ups]: connect failed: Connection fai
lure: Connection refused                                                        
Feb  3 20:42:17 freenas upsmon[91532]: UPS [ups]: connect failed: Connection fai
lure: Connection refused`

clepple, 2018-02-03:
What output do you get from “upsdrvctl -DD start”?

evildemons, 2018-02-03:
@clepple 

`[root@freenas ~]# upsdrvctl -DD start                                                                                               
Network UPS Tools - UPS driver controller 2.7.4                                                                                     
   0.000000                                                                                                                         
If you're not a NUT core developer, chances are that you're told to enable debugging                                                
to see why a driver isn't working for you. We're sorry for the confusion, but this is                                               
the 'upsdrvctl' wrapper, not the driver you're interested in.                                                                       
                                                                                                                                    
Below you'll find one or more lines starting with 'exec:' followed by an absolute                                                   
path to the driver binary and some command line option. This is what the driver                                                     
starts and you need to copy and paste that line and append the debug flags to that                                                  
line (less the 'exec:' prefix).                                                                                                     
                                                                                                                                    
   0.000074     Starting UPS: ups                                                                                                   
   0.000088     1 remaining attempts                                                                                                
   0.000092     exec:  /usr/local/libexec/nut/blazer_usb -a ups                                                                     
Network UPS Tools - Megatec/Q1 protocol USB driver 0.12 (2.7.4)                                                                     
No supported devices found. Please check your device availability with 'lsusb'                                                      
and make sure you have an up-to-date version of NUT. If this does not help,                                                         
try running the driver with at least 'subdriver', 'vendorid' and 'productid'                                                        
options specified. Please refer to the man page for details about these options                                                     
(man 8 blazer_usb).                                                                                                                 
                                                                                                                                    
   9.976790     Driver failed to start (exit status=1) `

clepple, 2018-02-03:
What about “usbconfig”?

evildemons, 2018-02-04:
@clepple 

`ugen0.1: <Intel EHCI root HUB> at usbus0, cfg=0 md=HOST spd=HIGH (480Mbps) pwr=SAVE (0mA)                                           
ugen1.1: <Intel EHCI root HUB> at usbus1, cfg=0 md=HOST spd=HIGH (480Mbps) pwr=SAVE (0mA)                                           
ugen0.2: <vendor 0x8087 product 0x0024> at usbus0, cfg=0 md=HOST spd=HIGH (480Mbps) pwr=SAVE (0mA)                                  
ugen1.2: <vendor 0x8087 product 0x0024> at usbus1, cfg=0 md=HOST spd=HIGH (480Mbps) pwr=SAVE (0mA)                                  
ugen0.3: <Winbond Electronics Corp Hermon USB hidmouse Device> at usbus0, cfg=0 md=HOST spd=FULL (12Mbps) pwr=ON (100mA)            
ugen1.3: <PPC HID UPS> at usbus1, cfg=0 md=HOST spd=FULL (12Mbps) pwr=ON (100mA)`

clepple, 2018-02-04:
I forgot that it does not print numeric IDs by default. Can you please try `usbconfig -d ugen1.3 dump_device_desc` ?

evildemons, 2018-02-04:
@clepple 

`root@freenas ~]# usbconfig -d ugen1.3 dump_device_desc                                                                             
ugen1.3: <PPC HID UPS> at usbus1, cfg=0 md=HOST spd=FULL (12Mbps) pwr=ON (100mA)                                                    
                                                                                                                                    
  bLength = 0x0012                                                                                                                  
  bDescriptorType = 0x0001                                                                                                          
  bcdUSB = 0x0200                                                                                                                   
  bDeviceClass = 0x0000  <Probed by interface class>                                                                                
  bDeviceSubClass = 0x0000                                                                                                          
  bDeviceProtocol = 0x0000                                                                                                          
  bMaxPacketSize0 = 0x0008                                                                                                          
  idVendor = 0x06da                                                                                                                 
  idProduct = 0xffff                                                                                                                
  bcdDevice = 0x0015                                                                                                                
  iManufacturer = 0x0001  <PPC>                                                                                                     
  iProduct = 0x0002  <HID UPS        >                                                                                              
  iSerialNumber = 0x0004  <718317100029    >                                                                                        
  bNumConfigurations = 0x0001`

clepple, 2018-02-04:
Ah, the VID:PID combo 06da:ffff was added in commit a08a2292afc559d6dfb9eede014a941d8f6b511d, which was after v2.7.4 was released (we are still sorting things out for the next formal release).

Theoretically (assuming this is the same type of device), you can change the FreeNAS configuration to usbhid-ups, and run this from the command line:

    /usr/local/libexec/nut/usbhid-ups -a ups -x vendorid=06da -x explore -DD

If that works, you could apply the patch from the aforementioned commit and rebuild.

evildemons, 2018-02-05:
@clepple 

Where should you enter it in FreeNAS?

clepple, 2018-02-05:
The driver debug command (`/usr/local/libexec/nut/usbhid-ups -a ups -x vendorid=06da -x explore -DD`) would be run as root in the same console as the other commands. If you do run it, note that it won't stop on its own (you can kill it with Ctrl-C after a minute - the polling cycle will repeat), and that the log is probably best saved off to a file (rather than posting inline).

I don't know the details of rebuilding NUT in FreeNAS, but it seems like it should be pretty similar to ports in FreeBSD. Maybe check with the FreeNAS forums? (We might also have some details in the [NUT mail archives](http://networkupstools.org/support.html) - I do not have time to sort through all of the search results at the moment.)

clepple, 2018-02-05:
(edited the title, and adjusted formatting on first post)

evildemons, 2018-02-05:
@clepple 

Do not know if you notice I've changed my post.

`  16.920290     Quick update...                                                 
  18.935207     upsdrv_updateinfo...                                            
  18.936153     Got 2 HID objects...                                            
  18.936188     Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID:
 0x06, Offset: 0, Size: 8, Value: 100                                           
  18.936200     Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x
06, Offset: 8, Size: 32, Value: 1315                                            
  18.936210     Quick update...                                                 
  20.942203     upsdrv_updateinfo...                                            
  21.207654     libusb_get_interrupt: Connection timed out                      
  21.207666     Got 0 HID objects...                                            
  21.207671     Quick update...                                                 
  22.974191     upsdrv_updateinfo...                                            
  23.245632     libusb_get_interrupt: Connection timed out                      
  23.245660     Got 0 HID objects...                                            
  23.245664     Quick update...                                                 
  24.994667     upsdrv_updateinfo...                                            
  25.192215     Got 2 HID objects...                                            
  25.192231     Path: UPS.PowerSummary.RemainingCapacity, Type: Input, ReportID:
 0x06, Offset: 0, Size: 8, Value: 100                                           
  25.192243     Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x
06, Offset: 8, Size: 32, Value: 1335                                            
  25.192253     Quick update...     `

clepple, 2018-02-05:
Nope, the email notifications only show new posts, so thank you for pointing that out.

Also, for multi-line output, I'd recommend using triple backpacks as mentioned under "code fencing" here: https://guides.github.com/features/mastering-markdown/ 

That output is a good sign. (The "connection timed out" messages are okay - they just mean that the UPS has not sent any updates during the polling interval, so NUT asks again to make sure it is still there.)

My previous comment about adding the patch still stands. (If the driver knows about a specific Vendor ID, you can use configuration options to add a new Product ID, but since new Vendor IDs usually come with a lot of vendor-specific changes, the only way to add a new vendor ID is to rebuild the driver. In your case, it looks like the vendor ID is very similar to existing MGE/Eaton models.)

evildemons, 2018-02-05:
@clepple 

Where would I change to add it or should I wait for 2.7.5?

clepple, 2018-02-06:
Up to you - we don't have a strict schedule for new releases, and no guarantees as to how long before it hits the ports tree, which is why I was recommending that you patch.

However, I don't have any FreeNAS systems here (only a few FreeBSD boxes without enough RAM for things like ZFS), so your best bet might be to ask in the FreeNAS forums. It might be as simple as adding the patch to /usr/ports/sysutils/nut, changing the package description, and rebuilding.

evildemons, 2018-02-07:
@clepple 

I made it work.

clepple, 2018-02-08:
Glad to hear it! Any tips for other FreeNAS users? (link to forum post, etc.)

evildemons, 2018-02-08:
//i.imgur.com/eIDKt1o.png

Everything I've done.