networkupstools/nut #520: [HCL] Cyber Power Systems CP1500PFCLCD supported by usbhid

webtroter, 2018-02-01:
Cyber Power Systems CP1500PFCLCD,

upsc CP1500PFCLCD@localhost
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 1530
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CP1500PFCLCD
device.serial: 000000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 139
input.transfer.low: 88
input.voltage: 116.0
input.voltage.nominal: 120
output.voltage: 140.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 25
ups.mfr: CPS
ups.model: CP1500PFCLCD
ups.productid: 0501
ups.realpower.nominal: 900
ups.serial: 000000000000
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
upsrw CP1500PFCLCD@localhost
[battery.charge.low]
Remaining battery level when UPS switches to LB (percent)
Type: STRING
Maximum length: 10
Value: 10

[battery.runtime.low]
Remaining battery runtime when UPS switches to LB (seconds)
Type: STRING
Maximum length: 10
Value: 300

[input.transfer.high]
High voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 139

[input.transfer.low]
Low voltage transfer point (V)
Type: STRING
Maximum length: 10
Value: 88

[ups.delay.shutdown]
Interval to wait after shutdown with delay command (seconds)
Type: STRING
Maximum length: 10
Value: 20

[ups.delay.start]
Interval to wait before (re)starting the load (seconds)
Type: STRING
Maximum length: 10
Value: 30

upscmd -l CP1500PFCLCD@localhost
Instant commands supported on UPS [CP1500PFCLCD]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.mute - Temporarily mute the UPS beeper
beeper.off - Obsolete (use beeper.disable or beeper.mute)
beeper.on - Obsolete (use beeper.enable)
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test

Cannot test  yet.

https://www.cyberpowersystems.com/product/ups/cp1500pfclcd/

clepple, 2018-02-07:
Thanks, and please keep us posted on shutdown test results. It should work fine, but please note the offdelay/ondelay issue with CPS units: http://new.networkupstools.org/docs/man/usbhid-ups.html#_extra_arguments

Also note the other Cyber Power Systems issues: https://github.com/networkupstools/nut/labels/CyberPower%20%28CPS%29

ShahimEssaid, 2019-09-22:
I'm not able to set battery.charge.low

~# upsd -V
Network UPS Tools upsd 2.7.4

On Debian Stretch

~# upsc CyberPower@10.27.47.49
Init SSL without certificate database
battery.charge: 58
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 4944
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CP1500PFCLCD
device.serial: 000000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 139
input.transfer.low: 88
input.voltage: 120.0
input.voltage.nominal: 120
output.voltage: 136.0
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 0
ups.mfr: CPS
ups.model: CP1500PFCLCD
ups.productid: 0501
ups.realpower.nominal: 900
ups.serial: 000000000000
ups.status: OL CHRG
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764

~# upsrw -s battery.charge.low=20 -u * -p * CyberPower@10.27.47.49
OK

~# upsc CyberPower@10.27.47.49 battery.charge.low
Init SSL without certificate database
10

misiektoja, 2019-09-29:
Hey, I have the same issue as ShahimEssaid with the same CyberPower UPS CP1500EPFCLCD. I have nut-ups (2.7.4-16) configured on my AsusWRT-Merlin router and even though upsrw shows the change was successful in fact it is not reflected in the UPS config. Google search brought me here as there is not that much info about it.

# upsrw -s battery.charge.low=5 -u upsuser -p mypwd CyberpowerUPS
OK

# upsc CyberpowerUPS battery.charge.low
10

Any ideas ? I'm using nut-driver-usbhid-ups instead of nut-driver-powerpanel as the latter one does not load correctly (and on few websites it was suggested to use first one).

stompro, 2021-06-29:
Hello, I've noticed that with one Cyberpower model... not this one, (PR1000LCDRTXL2U) that only certain values for battery.charge.low are allowed.  Anything outside of the allowed values are rounded or ignored.

For that model, the only allowed values are [60,55,50,45,40,35,30,25,20]

So in some cases, your UPS may not support a value of 10 for the battery.charge.low.

A bash loop like this can help you map out the allowed values.
for i in {90..0}; do echo "set to $i"; upsrw -s battery.charge.low=$i -u * -p * cps-big; sleep 1; upsc cps-big battery.charge.low;echo ""; done

closed 2022-04-25 by jimklimov

jimklimov, 2024-12-18:
PR #2718 is presumed to have fixed the broken CPS input/output voltage reports mentioned in this issue (although it did not focus on cases of "voltage too high" like here).

commit e896de37 (2022-04-25): driver.list.in: Add HCL info for CyberPower Systems CP1500PFCLCD (USB)

Closes: #520