networkupstools/nut #522: Kitbox GM312 Mini UPS (manufacturer: Smart Key Energy)

ulesin, 2018-02-14:
Hi, does anyone know what drivers work for this ups? Do I have to try all of them? Thank you very much.

clepple, 2018-02-14:
Does that UPS have a control interface? There is a USB Type A port on it, but that would most likely be to provide power to a device, rather than for shutdown signaling and control. Are there any Type B ports?

ulesin, 2018-02-14:
If only use a USB type A, is not possible to works like a UPS? with NUT? Type A or B just change their forms, information in comunications are the same, isn't?

clepple, 2018-02-15:
> Type A or B just change their forms, information in comunications are the same, isn't?

You might be thinking of USB OTG, which usually uses Mini-B connectors. Usually the device on the A side is a host controller, and the B side is a peripheral.

That said, I don't know specifically what the vendor has done with this UPS. Do they mention any control software, even if it is for Windows?