networkupstools/nut #526: Add D-Bus API to query/signal the state of the UPS

bigon, 2018-02-19:
It would be nice to have a D-Bus API so the users could get the status of the UPS

This would allow the proper integration of notifications in the desktop

I don't know if a standardized API already exists for this.

clepple, 2018-02-20:
Before DKP was superseded by UPower, we had a build option for USB-based UPSes to make a combined driver and DeviceKitPower server. I never tried it myself, but my understanding was that it was a lot simpler for new users since it did not require configuration.

There has not been much demand for a UPower-based replacement, although I suspect many users are fine with the basic single-UPS code in UPower. Are you proposing something that would be able to handle more than one UPS?

Also, the Python GUI provides desktop notifications.

I'm not opposed to the idea, but I do think we should consider the use cases, and how they differ from existing tools.

pbeza, 2018-02-20:
This issue came from [this question](https://unix.stackexchange.com/questions/425249/setup-nut-power-failure-notifications-when-running-computer-on-ups-battery/425257). I'm looking for a way to get GUI notifications (preferably popup message native for given desktop environment) when there is power failure and my computer is running on UPS battery (to know that I have little time to save the work and turn off the system). Is there any better way than this one that is described on page 71 of [this book](http://rogerprice.org/NUT/ConfigExamples.A5.pdf).

clepple, 2018-02-20:
@pbeza it looks like the package for the Python nut-monitor GUI doesn't recommend `python-notify`, but if you install that, and configure it to launch in your desktop session startup, you should get GUI notifications: https://github.com/networkupstools/nut/blob/v2.7.4/scripts/python/app/NUT-Monitor#L631

(NUT-Monitor only seems to notify for errors and the "on battery" status, so the shutdown testing commands won't necessarily help for testing GUI notifications. For that, you could configure `dummy-ups` in the same `ups.conf`, and point NUT-Monitor to that driver as well.)

@bigon should `python-notify` be a `Suggests:` dependency?

clepple, 2018-02-20:
> configure it to launch in your desktop session startup

@pbeza sorry if this wasn't clear: configure NUT-Monitor to launch when you log in (might be as easy as just saving the session when you log out)

pbeza, 2018-02-20:
I've tested it and it's true that `nut-monitor` displays GUI notification when running on batteries if `python-notify` is installed (BTW: there is also `python-notify2` – is it OK to use it instead `python-notify`?). *That's great, but* I feel like those notifications should be handled in more elegant way – preferably so that I don't need to run some GUI application (I mean `nut-monitor`) at system startup.

I imagine that it should be implemented so that my UPS was listed by `gnome-power-statistics` (as other battery based devices do – e.g. wireless mouse and keyboard). I believe that it should be integrated with the desktop environment just like laptops' batteries have notifications integrated within desktop environments.

I'm not much into UPower, so I need to ask you: are my expectations related to UPower support? What do you think about [my](https://en.wikipedia.org/wiki/UPower) expectations? Are there sane?

clepple, 2018-02-21:
Although I haven't looked into it further, I think your expectations would probably be met by some sort of bridge that connects to NUT `upsd` as a client, and publishes the values over D-Bus in the same way as other UPower devices, and that would probably get picked up by gnome-power-statistics.

We've discussed this idea a bit before, but as you can imagine, a large portion of the NUT user base is running the code on a server, and therefore they want remote notifications rather than local desktop notifications. Unless someone else with UPower experience steps up to write that code, it's going to take a while to get through the current backlog of issues (we are trying to wrap up a large rewrite of the USB-related code that has been dragging on for some time). Hence, the recommendation to use `nut-monitor` for now.

bigon, 2018-02-21:
FTR upower (and GNOME) already support minimal functionalities for the UPS. When nut is not installed can read the battery level of the USB ones

clepple, 2018-02-21:
@bigon agreed that UPower handles many of these cases, if the UPS is USB HID PDC compliant. That overlap probably accounts for why this is not requested more frequently from NUT.

However, there are a huge number of non-PDC UPSes (especially those served by blazer_usb and nutdrv_qx), and even for PDC devices, there are a lot of fixes for scaling problems. Unfortunately, these less expensive models tend not to have unique USB identifiers, and so they tend to involve more manual setup. I want to set expectations appropriately for users that not everything will be plug-and-play (particularly if the UPower code didn't detect the UPS to begin with), and might involve a bit of configuration (at least to confirm that they are not trying to monitor, say, a USB hub with an UPS driver).

Is it fair to say that the title of this issue has morphed into something like "Publish NUT UPS status to UPower", or are you thinking that we need a more generic D-Bus API?

bigon, 2018-02-23:
@clepple good question, I guess that exposing that centrally via upower might be a good idea.

But on the other side, upower is mostly used in desktop environments, I don't know if we have cases where we want to query nut using D-Bus on a server

jimklimov, 2022-02-13:
Just in case: do we have a logged issue (or what wize words and terms to put into one) about an opposite thing: a NUT driver to monitor a built-in battery (whatever GUI sees and exposes without NUT, notably may differ or not for Linux, BSD, MacOS, Solaris/illumos and Windows) as an UPS - e.g. to orchestrate shutdowns of local VMs and containers same as a server would with NUT and external UPS?

clepple, 2022-02-13:
#319 covers monitoring the battery on a Linux laptop (not depending on D-Bus). I don't think we have many of the other OS cases enumerated yet.

We probably have this covered for macOS, though occasionally the APIs subtly change: https://networkupstools.org/docs/man/macosx-ups.html