networkupstools/nut #529: Move killpower flag file to /run

bigon, 2018-02-23:
Hi,

Shouldn't the killpower flag file moved to /run (or /run/nut, or...) to avoid abusing /etc?

That would requires an upgrade path an maybe coordination with other to fix initscripts?

jimklimov, 2018-02-23:
I am not sure what the reason was for the initial choice - possibly systems from long ago that did not have the likes of tmpfs, but did unmount "all" filesystems they could (so would've lost a /var/* location).

Probably makes sense to have this (dir or full dir/file pathname) a configure-customizable value (and likely defaulting to legacy /etc), with the rest being left to distro/packaging maintainers.

jimklimov, 2023-01-03:
Came back to this idea (thought it anew and found in history ;) happens...) and looked at the codebase. Indeed, `killpower` and `POWERDOWNFLAG` strings (and related C `powerdownflag` variable name) are only mentioned in `clients/upsmon.c`, docs, sample configs, augeas tools lenses, and init-scripts. In fact the filename should not be mentioned in the latter since `upsmon -K` should hide the internal detail and check for the flag in whatever arcane ways are possible.

I looked at `upsmon.c` and it only deals with configured setting from `ups.conf` and does not hardcode any path, and supports lack of it (checks for NULL before use). It also removes the file on any startup except when quick-checking for existence (`upsmon -K`).

For this issue, I think it should become a configure-time option to set the default (to bake into `conf/upsmon.conf.sample`) - probably keeping current value for least-surprise, with distros free to override for whatever FS they keep R/O mounted in the end. Probably using the PID directory would be a good choice for most systems, unless it is in a `/var` partition that gets unmounted :)

closed 2023-01-04 by jimklimov

jimklimov, 2024-04-06:
Followed up in #2383

commit 83be8e69 (2023-01-04): configure.ac and many others: make default POWERDOWNFLAG configurable [#529]

Closes: #529