networkupstools/nut #530: Riello IDG400 wrong values & not powering off

sgiannulli, 2018-02-25:
Hello,
I just changed my old UPS with a new Riello IDG400:
1: it shows battery charge 255% - battery.runtime 3932100 - Runtime (H:M:S): 1092:15:00
is this an hardware problem or something on Nut side?
2: I use a raspberry as nut server to shutdown all lan clients after 180 seconds without main power; the old ups was also powered off after FINALDELAY 5 seconds while the Riello keeps staying on battery; is this something related to the first problem?
Thanks

jimklimov, 2018-02-25:
Is this model on HCL? If yes, might it be that the vendor changed specs/protocol keeping same model name?

Beside a possible issue with this piece of hardware, this might be a NUT problem at least in way of not having the driver support for this model/update of the device. Or a vendor's one, requiring new ways to communicate where old ways don't suffice. Rolling back to the NUT problem. Maybe some other NUT driver or sub-driver already supports the needed protocol?

As for UPS poweroff - this as well may be a HW issue (e.g. not everyone has such feature at all), or likely that your current driver can't tell the UPS what it wants in a way that both sides understand.

sgiannulli, 2018-02-25:
Yes IDG 400 is listed in HCL but really do not know what to say about the rest. Maybe calling the Cypress subdriver and Mustek protocol

clepple, 2018-02-26:
From http://lists.alioth.debian.org/pipermail/nut-upsdev/2018-February/007373.html :

> this type of UPS doesn't support realtime autonomy calculation, instead a fixed countdown must be used.

Note that we have a pending update to the HCL (5 star -> 4 star) because the IDG400 was not in the protocol documents. http://new.networkupstools.org/stable-hcl.html?manufacturer=Riello

sgiannulli, 2018-02-26:
well, apparently replacing my old UPS with this one was not a good choice. Even if I won't care about realtime autonomy calculation (and this is what I do as I set the nut server to shutdown everything after 180 seconds on battery) I can't believe that a recent brand ups won't respect the power off and re-start when mains back.
Thanks a lot for your comments and great work

clepple, 2018-03-01:
> I can't believe that a recent brand ups won't respect the power off and re-start when mains back.

You may want to check out the reply (and patch) on the nut-upsdev thread:

> 	-	small changes in riello_usb.c and riello_ser that solved problem with UPS that not support realtime autonomy calculation (and also internal temperature);
> 	-	the second problem I think is related to the first because the shutdown command that I sent to UPS by NUT monitor works without problems;

http://lists.alioth.debian.org/pipermail/nut-upsdev/2018-February/007374.html

sgiannulli, 2018-03-01:
yes.....an help to apply the patch and test it please?

clepple, 2018-03-02:
sure - OS details please?

sgiannulli, 2018-03-02:
Raspbian Stretch so basically Debian

clepple, 2018-03-05:
1) make sure you have "deb-src" lines to match the "deb" lines in /etc/apt/sources
2) run "sudo apt-get update" if you had to change anything
3) run "sudo apt-get build-dep nut"
3.5) optional: remove "asciidoc" (or install the build-deps manually, and omit asciidoc) to save a bit of build time
4) run "sudo apt-get -y install git" if you do not already have Git installed
5) clone the source to your working directory: "git clone https://github.com/networkupstools/nut.git"
5.5) "cd nut", apply the patch downloaded from the URL above: "zcat path/to/attachment.bin | patch -p0"
6) run this mega-command:

./configure --includedir=/usr/include --mandir=/usr/share/man \
       --infodir=/usr/share/info --sysconfdir=/etc/nut --localstatedir=/var \
       --libexecdir=/usr/lib/nut --srcdir=. --enable-maintainer-mode \
       --disable-silent-rules --libdir=/usr/lib/arm-linux-gnueabihf \
       --with-ssl --with-nss --with-cgi --with-dev --enable-static \
       --with-statepath=/var/run/nut --with-altpidpath=/var/run/nut \
       --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut \
       --with-htmlpath=/usr/share/nut/www --with-pidpath=/var/run/nut \
       --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/arm-linux-gnueabihf/pkgconfig \
       --with-user=nut --with-group=nut --with-udev-dir=/lib/udev \
       --with-systemdsystemunitdir=/lib/systemd/system

7) run "make"

sgiannulli, 2018-03-05:
Ok I will purge remove nut installed from standard raspbian packages and test the patch. Keep you posted.
Many thanks

clepple, 2018-03-05:
not necessary to purge the packages - the configuration I suggested lets you use all of the startup scripts and other NUT binaries as-is.

sgiannulli, 2018-03-06:
followed your guide twice and nothing changed:
driver.version.internal | 0.03
should not be 0.04?
and always battery charge 255% - battery.runtime 3932100 - Runtime (H:M:S): 1092:15:00 ups.temperature 255

sgiannulli, 2018-03-06:
I have applied the patch, wrong values disappeared but ups keeps staying on battery when main power off, so nothing changed on this side

CentroricercheRPS, 2018-03-13:
Hi,
we are the mantainer of riello drivers.
When main power off, NUT wait the battery low signals before send a shutdown command to the ups.
From what we understood you have a fixed time to shutdown the ups after main power off. Is this right?
In every case please could you post your NUT configuration (the main parameters of ups.conf, upsmon.conf etc...) so we try to solve your issue?

sgiannulli, 2018-03-13:
Hi,
zip file with confs attached.
[nutconfig.zip](https://github.com/networkupstools/nut/files/1808069/nutconfig.zip)
my config tells the nut server (a Raspberry with Raspbian Stretch installed) to shutdown server and clients after 3 minutes working on battery. This was working with my old UPS (an EQ Power LN651T)managed through the blazer_usb driver; after FINALDELAY the UPS put itself down too. This is not happening with the Riello IDG400 so I'm wondering if `/lib/systemd/system-shutdown/nutshutdown` works with the IDG and expecially if the IDG will return on when main power is back again.
Thanks

ulesin, 2018-03-21:
Hello everyone! Any news? I am very interested in this topic because I have an UPS IDG 800. Thank you very much!

autofrank, 2019-07-17:
> 1. make sure you have "deb-src" lines to match the "deb" lines in /etc/apt/sources
> 2. run "sudo apt-get update" if you had to change anything
> 3. run "sudo apt-get build-dep nut"
>    3.5) optional: remove "asciidoc" (or install the build-deps manually, and omit asciidoc) to save a bit of build time
> 4. run "sudo apt-get -y install git" if you do not already have Git installed
> 5. clone the source to your working directory: "git clone https://github.com/networkupstools/nut.git"
>    5.5) "cd nut", apply the patch downloaded from the URL above: "zcat path/to/attachment.bin | patch -p0"
> 6. run this mega-command:
> 
> ./configure --includedir=/usr/include --mandir=/usr/share/man 
> --infodir=/usr/share/info --sysconfdir=/etc/nut --localstatedir=/var 
> --libexecdir=/usr/lib/nut --srcdir=. --enable-maintainer-mode 
> --disable-silent-rules --libdir=/usr/lib/arm-linux-gnueabihf 
> --with-ssl --with-nss --with-cgi --with-dev --enable-static 
> --with-statepath=/var/run/nut --with-altpidpath=/var/run/nut 
> --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut 
> --with-htmlpath=/usr/share/nut/www --with-pidpath=/var/run/nut 
> --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/arm-linux-gnueabihf/pkgconfig 
> --with-user=nut --with-group=nut --with-udev-dir=/lib/udev 
> --with-systemdsystemunitdir=/lib/systemd/system
> 
> 1. run "make"

HI @clepple 
Could you clarify how to perform step 5.5. 
Did the patch get downloaded with the clone - if so where is it ?
If not - how can I get the patch and what is it called ?
I'm not terribly familiar with the raspian commands so any help really appreciated 

Thanks

autofrank, 2019-07-17:
> Hello everyone! Any news? I am very interested in this topic because I have an UPS IDG 800 and I have the same problems with 255 values. Is there another form to patch It? I don't know the exactly commands I have to put from point 5.5, It doesn't work for me. Thank you very much!

Hi @ulesin 
Did you manage to resolve ?

Thanks

clepple, 2019-07-18:
@autofrank I think I was referring to this patch from this message to nut-upsdev:

http://lists.alioth.debian.org/pipermail/nut-upsdev/2018-February/007374.html

It’s the link below this: “Desc: riello.diff.gz”

autofrank, 2019-07-18:
Thanks Charles..I located the patch and applied it.I'm trying to run the mega  "./ configure....." command you suggest but all the combinations I try don't work. It seems like the command is not supported.  I can only find a configure.au file on the pi and  I'm also not sure if the command should be run as a single long string or a set of separate commands Any help appreciated...Thanks Could you help meSent from my Samsung Galaxy smartphone.

http://lists.alioth.debian.org/pipermail/nut-upsdev/2018-February/007374.html

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/networkupstools/nut/issues/530?email_source=notifications\u0026email_token=AE3NARJEMFUPPTHVMMHNI2DP77FK7A5CNFSM4ESJPNSKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD2HC6SQ#issuecomment-512634698",

"url": "https://github.com/networkupstools/nut/issues/530?email_source=notifications\u0026email_token=AE3NARJEMFUPPTHVMMHNI2DP77FK7A5CNFSM4ESJPNSKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD2HC6SQ#issuecomment-512634698",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]

clepple, 2019-07-18:
Hmm, it looks like the backslashes got stripped out. It should look like the `./configure" line (step 7) on this page: https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu

The backslashes make it all one command. It would need to be run from within the "nut" directory created by `git clone`.

Also, I must have dropped the step that says to run "./autogen.sh", which creates the `configure` script from `configure.ac`.

autofrank, 2019-07-18:
> Hmm, it looks like the backslashes got stripped out. It should look like the `./configure" line (step 7) on this page: [...]

Thanks @clepple 
I'll give that a go this evening ...
Just so I'm clear - the command in your post appears over several lines. I assume I don't need to sequence that into one string. The reason I ask is because the Pi wouldn't accept it last night as it tried to execute the parts at the start. It sounds like that is because the backslashes were missing ....

clepple, 2019-07-18:
Yes, use backslashes to join all of the lines into one command, as shown in the wiki page. (I will try to edit the comment in this issue, but no promises, since there are backslashes in the raw text already.)

autofrank, 2019-07-18:
> Yes, use backslashes to join all of the lines into one command, as shown in the wiki page. (I will try to edit the comment in this issue, but no promises, since there are backslashes in the raw text already.)

Perfect @clepple 
It looks like I should run it as attached 

[image]

clepple, 2019-07-18:
In general, either use backslashes and multiple lines, or no backslashes and all on one line. (They might not be a problem in this case- hard to see on a small screen. )

autofrank, 2019-07-18:
> In general, either use backslashes and multiple lines, or no backslashes and all on one line. (They might not be a problem in this case- hard to see on a small screen. )

Thanks @clepple 
I got that part to work earlier but the "make"command  won't work 
I've decided to start from scratch and see if'll work from a fresh raspbian buster image

autofrank, 2019-07-18:
@clepple 

make still not working 

sequence of events as follows..

login as: pi
pi@192.168.1.239's password:
Linux raspberrypi 4.19.57+ #1244 Thu Jul 4 18:42:50 BST 2019 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set                                                                                         a new password.

pi@raspberrypi:~ $ sudo apt-get update
Get:1 http://archive.raspberrypi.org/debian buster InRelease [25.1 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [205 kB]
Fetched 13.2 MB in 39s (339 kB/s)
Reading package lists... Done
pi@raspberrypi:~ $ sudo apt-get build-dep nut
Reading package lists... Done
E: You must put some 'source' URIs in your sources.list
pi@raspberrypi:~ $ sudo nano  /etc/apt/source.list
pi@raspberrypi:~ $ sudo nano  /etc/apt/sources
pi@raspberrypi:~ $ ls
pi@raspberrypi:~ $ ls
pi@raspberrypi:~ $ ls -l
total 0
pi@raspberrypi:~ $ sudo apt-get -y install git
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  git-man libcurl3-gnutls liberror-perl
Suggested packages:
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-cvs git-mediawiki git-svn
The following NEW packages will be installed:
  git git-man libcurl3-gnutls liberror-perl
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 6,144 kB of archives.
After this operation, 32.8 MB of additional disk space will be used.
Get:1 http://ftp.heanet.ie/mirrors/raspbian/raspbian buster/main armhf libcurl3-gnutls armhf 7.64.0-4 [291 kB]
Get:2 http://ftp.heanet.ie/mirrors/raspbian/raspbian buster/main armhf liberror-perl all 0.17027-2 [30.9 kB]
Get:3 http://ftp.heanet.ie/mirrors/raspbian/raspbian buster/main armhf git-man all 1:2.20.1-2 [1,619 kB]
Get:4 http://ftp.heanet.ie/mirrors/raspbian/raspbian buster/main armhf git armhf 1:2.20.1-2 [4,202 kB]
Fetched 6,144 kB in 3s (2,207 kB/s)
Selecting previously unselected package libcurl3-gnutls:armhf.
(Reading database ... 37620 files and directories currently installed.)
Preparing to unpack .../libcurl3-gnutls_7.64.0-4_armhf.deb ...
Unpacking libcurl3-gnutls:armhf (7.64.0-4) ...
Selecting previously unselected package liberror-perl.
Preparing to unpack .../liberror-perl_0.17027-2_all.deb ...
Unpacking liberror-perl (0.17027-2) ...
Selecting previously unselected package git-man.
Preparing to unpack .../git-man_1%3a2.20.1-2_all.deb ...
Unpacking git-man (1:2.20.1-2) ...
Selecting previously unselected package git.
Preparing to unpack .../git_1%3a2.20.1-2_armhf.deb ...
Unpacking git (1:2.20.1-2) ...
Setting up libcurl3-gnutls:armhf (7.64.0-4) ...
Setting up liberror-perl (0.17027-2) ...
Setting up git-man (1:2.20.1-2) ...
[... 294 lines cut ...]
checking for FreeIPMI ldflags... -lfreeipmi -lipmimonitoring
checking for freeipmi/freeipmi.h... no
checking for ipmi_monitoring.h... no
checking for library containing ipmi_ctx_create... no
checking for library containing ipmi_ctx_find_inband... no
checking for library containing ipmi_monitoring_init... no
checking for library containing ipmi_monitoring_sensor_read_record_id... no
checking for library containing ipmi_sdr_cache_ctx_destroy... no
checking for library containing ipmi_sdr_ctx_destroy... no
checking whether to build IPMI driver... no
checking whether to build Mac OS X meta-driver... no
checking whether i2c_smbus_read_word_data is declared... no
checking whether i2c_smbus_write_word_data is declared... no
checking whether i2c_smbus_read_block_data is declared... no
checking whether to build i2c based drivers... no
checking for Mozilla NSS version via pkg-config... none found
checking for Mozilla NSS cflags...
checking for Mozilla NSS ldflags... -lnss3 -lnssutil3 -lsmime3 -lssl3 -lplds4 -lplc4 -lnspr4
checking for NSS_Init... no
checking for nss.h... yes
checking for ssl.h... no
configure: error: Mozilla NSS not found (required for SSL support)
pi@raspberrypi:~/nut $ make
make: *** No targets specified and no makefile found.  Stop.
pi@raspberrypi:~/nut $

autofrank, 2019-07-18:
I think the issue is that i have a makefile.am and a makefile.in but no makefile
.. don;t know how to solve that one

clepple, 2019-07-19:
Did you run `sudo apt-get build-dep nut`? That grabs the dependencies.

`./configure` needs to not error out, then it will create `Makefile`s. (You can remove some of the options like the SSL-related ones if you don't want to build those components.)

autofrank, 2019-07-20:
@clepple 
I finally got it working , thanks for the assistance

The 255 is now gone but my battery capacity is showing as 7 and the temp reading is now gone
see below
`

battery.capacity: 7
--
battery.voltage: 13.5
battery.voltage.nominal: 12
device.mfr: RPS S.p.a.
device.model: UNV1
device.serial:
device.type: ups
driver.name: riello_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4-716-g0c48e7b2
driver.version.internal: 0.04
input.bypass.frequency: 409.50
input.bypass.voltage: 4095
input.frequency: 50.10
input.voltage: 241
output.frequency: 50.10
output.frequency.nominal: 50.0
output.L1.current: 0
output.L1.power: 0
output.L1.realpower: 0
output.L2.current: 0
output.L2.power: 0
output.L2.realpower: 0
output.L3.current: 0
output.L3.power: 0
output.L3.realpower: 0
output.power.percent: 0
output.voltage: 237
output.voltage.nominal: 230
ups.firmware: SWM038-01-02
ups.load: 0
ups.mfr: RPS S.p.a.
ups.model: UNV1
ups.power.nominal: 800
ups.productid: 5500
ups.realpower.nominal: 480
ups.serial:
ups.status: OL
ups.vendorid: 04b4

`
For the /etc/nut/ups.conf
Should I be using riello_usb.c instead of riello_usb
I tried this and it didn't start

autofrank, 2019-07-20:
> I have applied the patch, wrong values disappeared but ups keeps staying on battery when main power off, so nothing changed on this side

@sgiannulli 
Just wanted to check if you got the correct values
I applied the patch and no longer have 255 as battery charge but now  I have 7
Is this what you are also seeing (IDG800)

battery.capacity: 7
--
battery.voltage: 13.5
battery.voltage.nominal: 12
device.mfr: RPS S.p.a.
device.model: UNV1
device.serial:
device.type: ups
driver.name: riello_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4-716-g0c48e7b2
driver.version.internal: 0.04
input.bypass.frequency: 409.50
input.bypass.voltage: 4095
input.frequency: 50.10
input.voltage: 241
output.frequency: 50.10
output.frequency.nominal: 50.0
output.L1.current: 0
output.L1.power: 0
output.L1.realpower: 0
output.L2.current: 0
output.L2.power: 0
output.L2.realpower: 0
output.L3.current: 0
output.L3.power: 0
output.L3.realpower: 0
output.power.percent: 0
output.voltage: 237
output.voltage.nominal: 230
ups.firmware: SWM038-01-02
ups.load: 0
ups.mfr: RPS S.p.a.
ups.model: UNV1
ups.power.nominal: 800
ups.productid: 5500
ups.realpower.nominal: 480
ups.serial:
ups.status: OL
ups.vendorid: 04b4

autofrank, 2019-07-21:
> Hi,
> we are the mantainer of riello drivers.
> When main power off, NUT wait the battery low signals before send a shutdown command to the ups.
> From what we understood you have a fixed time to shutdown [...]

Hi @CentroricercheRPS 
I installed the patched ups driver and the battery.charge  and battery.runtime are missing
how can I calculate how much capacity I have left if power is being drawn from the UPS during an outage ?

[image]

clepple, 2019-07-21:
@autofrank `riello_usb` is still the name to use in `ups.conf`. The `riello_usb.c` file is just the primary source file that is used to build the driver.

I must have been thinking of a different patch, though. All this one does is prevent `battery.charge` and `battery.runtime` from being displayed if the values correspond to `0xff`. I assume this means that the UPS does not calculate these values.

autofrank, 2019-07-21:
> @autofrank `riello_usb` is still the name to use in `ups.conf`. The `riello_usb.c` file is just the primary source file that is used to build the driver.
>
> I must have been thinking of a different [...]

thanks @clepple 
I'm in the process of running it down to see how long it lasts but it seems that even from the  riello sw itself (power shield3) I can't seem to find the rundown time... may need to return it for another device...

CentroricercheRPS, 2019-08-28:
Yes, as @clepple <https://github.com/clepple> sayd some devices do not
provide the charge or authonomy indication.

NilByte, 2020-12-12:
I have the same problem with Riello IDG400 not accepting paramters offdelay, ondelay and lowbatt in ups.conf.

I tried to follow the git clone procedure to see if it would change but it didn't.

```
admin@raspberrypi:~ $ sudo service nut-client restart
admin@raspberrypi:~ $ sudo systemctl status nut-server
● nut-server.service - Network UPS Tools - power devices information server
   Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-12-12 15:11:02 GMT; 24s ago
  Process: 19988 ExecStart=/sbin/upsd (code=exited, status=0/SUCCESS)
 Main PID: 19993 (upsd)
    Tasks: 1 (limit: 4915)
   Memory: 480.0K
   CGroup: /system.slice/nut-server.service
           └─19993 /lib/nut/upsd

Dec 12 15:11:02 raspberrypi upsd[19988]: listening on 192.168.1.105 port 3493
Dec 12 15:11:02 raspberrypi upsd[19988]: listening on 127.0.0.1 port 3493
Dec 12 15:11:02 raspberrypi upsd[19988]: listening on 127.0.0.1 port 3493
Dec 12 15:11:02 raspberrypi upsd[19988]: Connected to UPS [UPS-1]: riello_usb-UPS-1
Dec 12 15:11:02 raspberrypi upsd[19988]: Connected to UPS [UPS-1]: riello_usb-UPS-1
Dec 12 15:11:02 raspberrypi upsd[19993]: Startup successful
Dec 12 15:11:02 raspberrypi systemd[1]: Started Network UPS Tools - power devices information server.
Dec 12 15:11:02 raspberrypi upsd[19993]: User upsmon@127.0.0.1 logged into UPS [UPS-1]
Dec 12 15:11:17 raspberrypi upsd[19993]: User upsmon@127.0.0.1 logged out from UPS [UPS-1]
Dec 12 15:11:17 raspberrypi upsd[19993]: User upsmon@127.0.0.1 logged into UPS [UPS-1]
admin@raspberrypi:~ $ upsc UPS-1
Init SSL without certificate database
battery.capacity: 7
battery.charge: 255
battery.runtime: 3932100
battery.voltage: 13.5
battery.voltage.nominal: 12
device.mfr: RPS S.p.a.
device.model: UNV1
device.serial:
device.type: ups
driver.name: riello_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4-2050-gd90c5d30
driver.version.internal: 0.03
input.bypass.frequency: 409.50
input.bypass.voltage: 4095
input.frequency: 50.00
input.voltage: 228
output.frequency: 50.00
output.frequency.nominal: 50.0
output.L1.current: 0
output.L1.power: 0
output.L1.realpower: 0
output.L2.current: 0
output.L2.power: 0
output.L2.realpower: 0
output.L3.current: 0
output.L3.power: 0
output.L3.realpower: 0
output.power.percent: 0
output.voltage: 222
output.voltage.nominal: 230
ups.firmware: SWM038-01-02
ups.load: 0
ups.mfr: RPS S.p.a.
ups.model: UNV1
ups.power.nominal: 400
ups.productid: 5500
ups.realpower.nominal: 240
ups.serial:
ups.status: OL
ups.temperature: 255
ups.vendorid: 04b4
admin@raspberrypi:~ $ lsusb
Bus 002 Device 002: ID 2109:0715 VIA Labs, Inc.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04b4:5500 Cypress Semiconductor Corp. HID->COM RS232 Adapter
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

It shows driver version 2.7.4-2050-gd90c5d30, but internal driver version is still 0.03.

Any ideas on how to set a timed offdelay to let the client shutdown before the UPS shuts down, that is not based on battery status?

The non opensource driver from Riello has this functionality working, so the riello_usb driver seems to miss this functionality.

geoghegan, 2021-07-31:
I've an IPG800 which is listed as a 4* driver. I am running nut on a Debian 10 rPi also. Here is the output from uspc (with the mains power killed)

```
Init SSL without certificate database
battery.capacity: 7
battery.charge: 255
battery.runtime: 3932100
battery.voltage: 12.3
battery.voltage.nominal: 12
device.mfr: RPS S.p.a.
device.model: USV5
device.serial:
device.type: ups
driver.name: riello_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 0.03
input.bypass.frequency: 409.50
input.bypass.voltage: 4095
input.frequency: 0.00
input.voltage: 4
output.frequency: 50.00
output.frequency.nominal: 50.0
output.L1.current: 0
output.L1.power: 0
output.L1.realpower: 0
output.L2.current: 0
output.L2.power: 0
output.L2.realpower: 0
output.L3.current: 0
output.L3.power: 0
output.L3.realpower: 0
output.power.percent: 0
output.voltage: 231
output.voltage.nominal: 230
ups.firmware: SWM036-01-04
ups.load: 0
ups.mfr: RPS S.p.a.
ups.model: USV5
ups.power.nominal: 800
ups.productid: 5500
ups.realpower.nominal: 480
ups.serial:
ups.status: OB
ups.temperature: 255
ups.vendorid: 04b4
[root@morbo /home/nick]$ upsc ipg800
Init SSL without certificate database
battery.capacity: 7
battery.charge: 255
battery.runtime: 3932100
battery.voltage: 12.3
battery.voltage.nominal: 12
device.mfr: RPS S.p.a.
device.model: USV5
device.serial:
device.type: ups
driver.name: riello_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 0.03
input.bypass.frequency: 409.50
input.bypass.voltage: 4095
input.frequency: 0.00
input.voltage: 4
output.frequency: 50.00
output.frequency.nominal: 50.0
output.L1.current: 0
output.L1.power: 0
output.L1.realpower: 0
output.L2.current: 0
output.L2.power: 0
output.L2.realpower: 0
output.L3.current: 0
output.L3.power: 0
output.L3.realpower: 0
output.power.percent: 0
output.voltage: 230
output.voltage.nominal: 230
ups.firmware: SWM036-01-04
ups.load: 0
ups.mfr: RPS S.p.a.
ups.model: USV5
ups.power.nominal: 800
ups.productid: 5500
ups.realpower.nominal: 480
ups.serial:
ups.status: OB
ups.temperature: 255
ups.vendorid: 04b4
```

The charge and runtime are obviously wrong.

@NilByte : Where did you find the non-FOSS driver?

NilByte, 2021-08-01:
I'm using upsmon downloaded from Riello:

https://www.riello-ups.com/downloads/1-powershield-free

V.5.3 (Build: 000) for Linux Raspbian Raspberry Pi

https://www.riello-ups.com/uploads/file/402/1402/upsmon-5.3.0-linux-3.12-armhf.deb

There are versions for Debian listed as well.

NilByte, 2022-01-18:
There is a mailing list here suggesting update of the riello_ser and riello_usb from the original driver author which could possibly fix some of the problems that are experienced.

Would it be possible for one of the NUT devs to have a look at this?

https://www.mail-archive.com/nut-upsdev@alioth-lists.debian.net/msg00106.html

From one of the attachments riello.diff:

```
Index: drivers/riello_ser.c
===================================================================
--- drivers/riello_ser.c	(revisione 7681)
+++ drivers/riello_ser.c	(copia locale)
@@ -37,14 +37,17 @@
 #include "riello.h"
 
 #define DRIVER_NAME	"Riello serial driver"
-#define DRIVER_VERSION	"0.03"
+#define DRIVER_VERSION	"0.04"
 
+#define DEFAULT_OFFDELAY   5
+#define DEFAULT_BOOTDELAY  5
+
 /* driver description structure */
 upsdrv_info_t upsdrv_info = {
 	DRIVER_NAME,
 	DRIVER_VERSION,
 	"Elio Parisi <e.parisi@riello-ups.com>",
-	DRV_EXPERIMENTAL,
+	DRV_STABLE,
 	{ NULL }
 };
 
@@ -57,6 +60,9 @@
 uint8_t input_monophase;
 uint8_t output_monophase;
 
+static unsigned int offdelay = DEFAULT_OFFDELAY;
+static unsigned int bootdelay = DEFAULT_BOOTDELAY; 
+
 extern uint8_t commbyte;
 extern uint8_t wait_packet;
 extern uint8_t foundnak;
@@ -751,6 +757,13 @@
 	dstate_addcmd("shutdown.stop");
 	dstate_addcmd("test.battery.start");
 
+	dstate_setinfo("ups.delay.shutdown", "%d", offdelay);
+	dstate_setflags("ups.delay.shutdown", ST_FLAG_RW | ST_FLAG_STRING);
+	dstate_setaux("ups.delay.shutdown", 3);
+	dstate_setinfo("ups.delay.reboot", "%d", bootdelay);
+	dstate_setflags("ups.delay.reboot", ST_FLAG_RW | ST_FLAG_STRING);
+	dstate_setaux("ups.delay.reboot", 3);
+
 	if (typeRielloProtocol == DEV_RIELLOGPSER)
 		dstate_addcmd("test.panel.start");
 
@@ -803,9 +816,12 @@
 	dstate_setinfo("input.bypass.frequency", "%.2f", DevData.Fbypass/10.0);
 	dstate_setinfo("output.frequency", "%.2f", DevData.Fout/10.0);
 	dstate_setinfo("battery.voltage", "%.1f", DevData.Ubat/10.0);
-	dstate_setinfo("battery.charge", "%u", DevData.BatCap);
-	dstate_setinfo("battery.runtime", "%u", DevData.BatTime*60);
-	dstate_setinfo("ups.temperature", "%u", DevData.Tsystem);
+	if ((DevData.BatCap < 0xFFFF) &&  (DevData.BatTime < 0xFFFF)) {
+		dstate_setinfo("battery.charge", "%u", DevData.BatCap);
+		dstate_setinfo("battery.runtime", "%u", DevData.BatTime*60);
+	}
+	if (DevData.Tsystem < 0xFF)
+		dstate_setinfo("ups.temperature", "%u", DevData.Tsystem);
 
 	if (input_monophase) {
 		dstate_setinfo("input.voltage", "%u", DevData.Uinp1);
Index: drivers/riello_usb.c
===================================================================
--- drivers/riello_usb.c	(revisione 7681)
+++ drivers/riello_usb.c	(copia locale)
@@ -33,14 +33,17 @@
 #include "riello.h"
 
 #define DRIVER_NAME	"Riello USB driver"
-#define DRIVER_VERSION	"0.03"
+#define DRIVER_VERSION	"0.04"
 
+#define DEFAULT_OFFDELAY   5  /*!< seconds (max 0xFF) */
+#define DEFAULT_BOOTDELAY  5  /*!< seconds (max 0xFF) */
+
 /* driver description structure */
 upsdrv_info_t upsdrv_info = {
 	DRIVER_NAME,
 	DRIVER_VERSION,
 	"Elio Parisi <e.parisi@riello-ups.com>",
-	DRV_EXPERIMENTAL,
+	DRV_STABLE,
 	{ NULL }
 };
 
@@ -52,6 +55,10 @@
 uint8_t input_monophase;
 uint8_t output_monophase;
 
+/*! Time in seconds to delay before shutting down. */
+static unsigned int offdelay = DEFAULT_OFFDELAY;
+static unsigned int bootdelay = DEFAULT_BOOTDELAY; 
+
 extern uint8_t commbyte;
 extern uint8_t wait_packet;
 extern uint8_t foundnak;
@@ -929,6 +936,13 @@
 	dstate_addcmd("test.battery.start");
 	dstate_addcmd("test.panel.start");
 
+	dstate_setinfo("ups.delay.shutdown", "%d", offdelay);
+	dstate_setflags("ups.delay.shutdown", ST_FLAG_RW | ST_FLAG_STRING);
+	dstate_setaux("ups.delay.shutdown", 3);
+	dstate_setinfo("ups.delay.reboot", "%d", bootdelay);
+	dstate_setflags("ups.delay.reboot", ST_FLAG_RW | ST_FLAG_STRING);
+	dstate_setaux("ups.delay.reboot", 3);
+
 	/* install handlers */
 /*	upsh.setvar = hid_set_value; setvar; */
 
@@ -1012,9 +1026,12 @@
 	dstate_setinfo("input.bypass.frequency", "%.2f", DevData.Fbypass/10.0);
 	dstate_setinfo("output.frequency", "%.2f", DevData.Fout/10.0);
 	dstate_setinfo("battery.voltage", "%.1f", DevData.Ubat/10.0);
-	dstate_setinfo("battery.charge", "%u", DevData.BatCap);
-	dstate_setinfo("battery.runtime", "%u", DevData.BatTime*60);
-	dstate_setinfo("ups.temperature", "%u", DevData.Tsystem);
+	if ((DevData.BatCap < 0xFFFF) &&  (DevData.BatTime < 0xFFFF)) {
+		dstate_setinfo("battery.charge", "%u", DevData.BatCap);
+		dstate_setinfo("battery.runtime", "%u", DevData.BatTime*60);
+	}
+	if (DevData.Tsystem < 0xFF)
+		dstate_setinfo("ups.temperature", "%u", DevData.Tsystem);
 
 	if (input_monophase) {
 		dstate_setinfo("input.voltage", "%u", DevData.Uinp1);
```

geoghegan, 2022-02-09:
@NilByte  PR #1280 may be of help to you

jimklimov, 2022-02-10:
Thanks for the PR, commented some bits there :)

As for the original issue, I think Riello devices use a variant of Megatec Qx protocols (sources mention "cypress"), so wondering if the newer `nutdrv_qx` (which consolidates many older drivers into same architecture) might handle them better?

geoghegan, 2022-02-11:
The erroneous runtime, temp, and charge should be fixed for an IDG (1600). Shutdown and reboot delays should also be working. Both addressed in version 0.7 of the driver, merged as PR #1280 .

NilByte, 2022-02-16:
Finally I managed to get successful timed shutdown, ups powering off and powering up again when mains return.

NUT build from master driver.version: 2.7.4-4609-g61862059, driver.version.internal: 0.07 according to  https://github.com/networkupstools/nut/wiki/Building-NUT-on-Debian,-Raspbian-and-Ubuntu

Configuration that worked for me:

```
        NUT configuration  2022-02-16 16:38:28 UTC

        ########### Operating system ###########
Kernel release 5.10.63-v8+ GNU/Linux
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_CODENAME=buster
Network UPS Tools upsd 2.7.4
Bash version 5.0.3(1)-release
User root
nut-report.sh version: 2020-08-07

        ########### /etc/nut/nut.conf ###########
MODE=standalone

        ########### /etc/nut/ups.conf ###########
pollinterval = 1
maxretry = 3
[UPS-1]
	driver = riello_usb
	port = auto
        desc = "Riello IDG400"
	override.ups.delay.shutdown = 14
	override.ups.delay.reboot = 7

        ########### /etc/nut/upsd.conf ###########
LISTEN 127.0.0.1 3493
LISTEN 192.168.1.105

        ########### /etc/nut/upsd.users ###########
[upsmaster]
	upsmon master
	password =  ****m**** 
	actions = SET FSD
	instcmds = ALL

        ########### /etc/nut/upsmon.conf ###########
RUN_AS_USER nut
MONITOR UPS-1@localhost 1 upsmaster  ****m****  master
MINSUPPLIES 1
SHUTDOWNCMD "/home/admin/config.scripts/custom.shutdown.sh"
NOTIFYCMD /sbin/upssched
POLLFREQ 2
POLLFREQALERT 1
HOSTSYNC 15
DEADTIME 15
POWERDOWNFLAG /etc/killpower
NOTIFYMSG ONLINE    "UPS %s on line power"
NOTIFYMSG ONBATT    "UPS %s on battery"
NOTIFYMSG LOWBATT   "UPS %s battery is low"
NOTIFYMSG FSD       "UPS %s: forced shutdown in progress"
NOTIFYMSG COMMOK    "Communications with UPS %s established"
NOTIFYMSG COMMBAD   "Communications with UPS %s lost"
NOTIFYMSG SHUTDOWN  "Auto logout and shutdown proceeding"
NOTIFYMSG REPLBATT  "UPS %s battery needs to be replaced"
NOTIFYMSG NOCOMM    "UPS %s is unavailable"
NOTIFYMSG NOPARENT  "upsmon parent process died - shutdown impossible"
NOTIFYFLAG ONLINE   SYSLOG+WALL+EXEC
NOTIFYFLAG ONBATT   SYSLOG+WALL+EXEC
NOTIFYFLAG LOWBATT  SYSLOG+WALL+EXEC
NOTIFYFLAG FSD      SYSLOG+WALL+EXEC
NOTIFYFLAG COMMOK   SYSLOG+WALL+EXEC
NOTIFYFLAG COMMBAD  SYSLOG+WALL+EXEC
[... 85 lines cut ...]
-rw-r----- 1 root nut  1538 Jun  1  2018 /etc/nut/nut.conf.dpkg-dist
-rw-r----- 1 root nut   647 Feb 16 17:00 /etc/nut/ups.conf
-rw-r----- 1 root nut  5522 Jun  1  2018 /etc/nut/ups.conf.dpkg-dist
-rw-r----- 1 root nut    43 Feb  3 22:39 /etc/nut/upsd.conf
-rw-r----- 1 root nut  4578 Jun  1  2018 /etc/nut/upsd.conf.dpkg-dist
-rw-r----- 1 root nut   358 Feb 10 18:00 /etc/nut/upsd.users
-rw-r----- 1 root nut  2131 Jun  1  2018 /etc/nut/upsd.users.dpkg-dist
-rw-r----- 1 root nut  1378 Feb 13 14:27 /etc/nut/upsmon.conf
-rw-r----- 1 root nut 15308 Jun  1  2018 /etc/nut/upsmon.conf.dpkg-dist
-rw-r----- 1 root nut  4815 Feb 15 21:51 /etc/nut/upssched.conf
-rw-r----- 1 root nut  3887 Jun  1  2018 /etc/nut/upssched.conf.dpkg-dist

/etc/nut/upssched:
total 8
drwxr-xr-x 2 nut  nut 4096 Feb 16 17:08 ./
drwxr-xr-x 3 root nut 4096 Feb 16 17:28 ../
srw-rw---- 1 nut  nut    0 Feb 16 17:08 upssched.pipe=

        ########### Custom NOTIFYCMD scheduler /sbin/upssched ###########
Custom NOTIFYCMD scheduler /sbin/upssched is not a script

        ########### nut-journal ###########
```

I believe that the issue on not powering off has also been solved by riello_usb driver version  0.7 https://github.com/networkupstools/nut/pull/1280

ntd, 2024-03-07:
Trying to summarize the current status (nut 2.8.1, riello_usb 0.11, UPS iDialog whatever): I'm able to shutdown if (and only if) the UPS is on battery and I send a shutdown command from software. Specifically I cannot:
1. shutdown when UPS is on power: it fails with `instcmd: unknown command [shutdown.return] [(null)]`
2. shutdown autonomously when power goes off

In my use case, 2 is especially important: if for some reason the software fails to send the shutdown (e.g. the PC is off), that thing stays on battery forever.

Is this correct?