networkupstools/nut #534: Windows-Client Error ntdll.dll

Meinen, 2018-03-08:
Hello,

I am having trouble deploying the Windows-Client.
After going through the installation and configuration, the client runs into the following error on startup:

Errorcode: 0xc000007b
Erroroffset: 0x0009d4c2
Application: : C:\Program Files (x86)\NUT\sbin\upsmon.exe
faulty module: C:\Windows\SYSTEM32\ntdll.dll

I did copy the gcc and the openssl libs to their required places and the configuration I am using is working on my Linux machines... except for the shutdowncommand of course.

Meinen, 2018-03-09:
looks like somthing is broken in Version 2.6.5-6.
I got Version 2.6.5-4 from your archive and got it running with the exact same Konfiguration.

clepple, 2018-03-18:
The NUT project doesn't currently have anyone maintaining the Windows port, but you may want to take a look at the file linked at the beginning of issue #455.

asylum71, 2019-06-18:
Thanks Charles!  I downloaded the installer at the beginning of #455 and Nut is working for my Win 10 and Win Server 1012 R2 hosts.

thenktor, 2020-04-02:
Yes, installer 2.6.5-6 is broken:
`./upsc.exe
C:/Program Files (x86)/NUT/bin/upsc.exe: error while loading shared libraries: ?: cannot open shared object file: No such file or directory`