networkupstools/nut #537: PowerCool VA1500 and nutdrv_qx

deaddox, 2018-03-10:
Hi,

I believe that the UPS the PowerCool VA1500 should be supported by the nutdrv_qx driver. It detects and says its connected but all the value read as zero. Looking through the forums there seems to be a few mentions of this but tbh honest i'm getting a little confused.

My current config is 

[nutdev1]
        driver = "nutdrv_qx"
        port = "auto"
        vendorid = "0001"
        productid = "0000"
        product = "MEC0003"
        protocol="q1"

This connects but provides 0 value readouts. What would be the next steps in trying to resolve this problem and is anyone able to assist. I know these PowerCool UPS are quite popular now in the UK as they are very cheap so it would be nice to get this working if possible.

deaddox, 2018-03-11:
The debug output for the driver is here https://pastebin.com/i3ddyDkd
The debug output for the UPSmart Windows software is here https://pastebin.com/KbgAHe6u

deaddox, 2018-04-20:
I was wondering if anyone had a chance to look into this?

ecscndt, 2021-01-03:
Any luck on this?

0x0BCA, 2021-01-03:
Any luck on this?

lxgenie, 2021-01-09:
Has anyone sucessfully got a Powercool UPS to work with NUT?

0x0BCA, 2021-01-09:
Yes, I have a powercool 2000va (i also had to use a different usb cable, the supplied one did not work):

https://github.com/networkupstools/nut/issues/674

Had to compile from source and merge the changes from mariojan

For ubuntu I used (other distros may vary):

 apt-get build-dep nut

./configure --includedir=/usr/include --mandir=/usr/share/man --infodir=/usr/share/info --sysconfdir=/etc/nut --localstatedir=/var --libexecdir=/usr/lib/nut --srcdir=. --enable-maintainer-mode --disable-silent-rules --libdir=/usr/lib/`gcc -print-multiarch` --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/var/run/nut --with-altpidpath=/var/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/var/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/`gcc -print-multiarch`/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system
 1227  ./configure --includedir=/usr/include --mandir=/usr/share/man --infodir=/usr/share/info --sysconfdir=/etc/nut --localstatedir=/var --libexecdir=/usr/lib/nut --srcdir=. --enable-maintainer-mode --disable-silent-rules --libdir=/usr/lib/`gcc -print-multiarch` --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/var/run/nut --with-altpidpath=/var/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/var/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/`gcc -print-multiarch`/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system

then make and make install

/etc/nut/ups.conf:

[ups]
       driver = "nutdrv_qx"
       port = "auto"
       vendorid = "0001"
       productid = "0000"
       protocol = hunnox
       langid_fix = 0x0409
       novendor
       noscanlangid

# upsc ups
Init SSL without certificate database
battery.charge: 100
battery.voltage: 40.50
battery.voltage.high: 39.00
battery.voltage.low: 31.20
battery.voltage.nominal: 36.0
device.type: ups
driver.flag.noscanlangid: enabled
driver.flag.novendor: enabled
driver.name: nutdrv_qx
driver.parameter.langid_fix: 0x0409
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0000
driver.parameter.protocol: hunnox
driver.parameter.synchronous: no
driver.parameter.vendorid: 0001
driver.version: 2.7.4-2077-g699bf8b4
driver.version.data: Hunnox 0.01
driver.version.internal: 0.28
input.current.nominal: 6.9
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 241.0
input.voltage.fault: 0.0
input.voltage.nominal: 230
output.voltage: 241.0
ups.beeper.status: enabled
ups.delay.shutdown: 60
ups.delay.start: 0
ups.load: 12
ups.productid: 0000
ups.status: OL
ups.temperature: 28.0
ups.type: offline / line interactive
ups.vendorid: 0001

HTH

holl595, 2021-01-22:
> Yes, I have a powercool 2000va (i also had to use a different usb cable, the supplied one did not work):
> 
> #674
> 
> Had to compile from source and merge the changes from mariojan
> 
> For ubuntu I used (other distros may vary):
> 
> apt-get build-dep nut
> 
> ./configure --includedir=/usr/include --mandir=/usr/share/man --infodir=/usr/share/info --sysconfdir=/etc/nut --localstatedir=/var --libexecdir=/usr/lib/nut --srcdir=. --enable-maintainer-mode --disable-silent-rules --libdir=/usr/lib/`gcc -print-multiarch` --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/var/run/nut --with-altpidpath=/var/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/var/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/`gcc -print-multiarch`/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system
> 1227 ./configure --includedir=/usr/include --mandir=/usr/share/man --infodir=/usr/share/info --sysconfdir=/etc/nut --localstatedir=/var --libexecdir=/usr/lib/nut --srcdir=. --enable-maintainer-mode --disable-silent-rules --libdir=/usr/lib/`gcc -print-multiarch` --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/var/run/nut --with-altpidpath=/var/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/var/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/`gcc -print-multiarch`/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system
> 
> then make and make install
> 
> /etc/nut/ups.conf:
> 
> [ups]
> driver = "nutdrv_qx"
> port = "auto"
> vendorid = "0001"
> productid = "0000"
> protocol = hunnox
> langid_fix = 0x0409
> novendor
> noscanlangid
> 
> root@foo:/etc/nut# upsc ups
> Init SSL without certificate database
> battery.charge: 100
> battery.voltage: 40.50
> battery.voltage.high: 39.00
> battery.voltage.low: 31.20
> battery.voltage.nominal: 36.0
> device.type: ups
> driver.flag.noscanlangid: enabled
> driver.flag.novendor: enabled
> driver.name: nutdrv_qx
> driver.parameter.langid_fix: 0x0409
> driver.parameter.pollfreq: 30
> driver.parameter.pollinterval: 2
> driver.parameter.port: auto
> driver.parameter.productid: 0000
> driver.parameter.protocol: hunnox
> driver.parameter.synchronous: no
> driver.parameter.vendorid: 0001
> driver.version: 2.7.4-2077-g699bf8b4
> driver.version.data: Hunnox 0.01
> driver.version.internal: 0.28
> input.current.nominal: 6.9
> input.frequency: 50.0
> input.frequency.nominal: 50
> input.voltage: 241.0
> input.voltage.fault: 0.0
> input.voltage.nominal: 230
> output.voltage: 241.0
> ups.beeper.status: enabled
> ups.delay.shutdown: 60
> ups.delay.start: 0
> ups.load: 12
> ups.productid: 0000
> ups.status: OL
> ups.temperature: 28.0
> ups.type: offline / line interactive
> ups.vendorid: 0001
> 
> HTH

Did you have to do anything special to get the battery percentage to be returned?

Following your guide I managed to get it working with my Powercool 1500VA but battery percentage is not being returned:
`Init SSL without certificate database
battery.voltage: 27.20
battery.voltage.high: 0.00
battery.voltage.low: 0.00
battery.voltage.nominal: 0.0
device.type: ups
driver.flag.noscanlangid: enabled
driver.flag.novendor: enabled
driver.name: nutdrv_qx
driver.parameter.langid_fix: 0x0409
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0000
driver.parameter.protocol: hunnox
driver.parameter.synchronous: no
driver.parameter.vendorid: 0001
driver.version: 2.7.4-2120-g38b4520c
driver.version.data: Hunnox 0.01
driver.version.internal: 0.28
input.current.nominal: 0.0
input.frequency: 49.8
input.frequency.nominal: 0
input.voltage: 246.0
input.voltage.fault: 0.0
input.voltage.nominal: 0
output.voltage: 248.0
ups.beeper.status: disabled
ups.delay.shutdown: 60
ups.delay.start: 0
ups.load: 11
ups.productid: 0000
ups.status: OL
ups.temperature: 29.0
ups.type: online
ups.vendorid: 0001 `

jammmyb, 2021-04-03:
Was there any progress made on this?

just got the 1000VA model and would love to achieve connectivity of it with NUT under Unraid.

andydvsn, 2021-12-02:
I've had the RACK850VA model from CPC in the UK for some time (order code CS33117 if you want to find it). Never bothered connecting it up until now, but when I tried today the repo default version of NUT supplied with Raspbian Stretch was having nothing to do with it.

I found instructions on the [NUT hardware compatibility list](https://networkupstools.org/stable-hcl.html) which led me down the garden path with the [NUTDRV_ATCL_USB](https://networkupstools.org/docs/man/nutdrv_atcl_usb.html) driver, which would have been very limited anyway. Thankfully I found this thread instead.

Purged the repo default version and followed @0x0BCA's instructions above to compile the latest code on here. I did not need to merge changes from mariojan's repo, but did need:

```
git clone https://github.com/networkupstools/nut.git
cd nut
sudo apt build-dep nut
./autogen.sh
./configure --includedir=/usr/include --mandir=/usr/share/man --infodir=/usr/share/info --sysconfdir=/etc/nut --localstatedir=/var --libexecdir=/usr/lib/nut --srcdir=. --enable-maintainer-mode --disable-silent-rules --libdir=/usr/lib/gcc --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/var/run/nut --with-altpidpath=/var/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/var/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/gcc --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system
make
sudo checkinstall -D -y --install
```
Upon installation it would repeatedly fail to create necessary directories:

```
sudo mkdir -p /usr/local/ups/bin
sudo mkdir -p /usr/lib/cgi-bin
sudo mkdir -p /usr/share/nut/www
```
Already losing the will to live, I was now able to create my `/etc/ups/ups.conf` file:

```
[ups]
driver = nutdrv_qx
port = auto
vendorid = 0001
productid = 0000
product = "MEC0003"
protocol = hunnox
langid_fix = 0x409
novendor
noscanlangid
```

Then followed some chowning and permissions fun:

```
sudo chown -R www-data:www-data html
sudo chown -R nut:nut /var/run/nut
sudo chown -R nut:nut /etc/nut
sudo usermod -a -G nut pi
```

Eventually [the output from the following command](https://pastebin.com/raw/23xsTZVj) was my prize:

```
/lib/nut/nutdrv_qx -a ups -DDDDD
```
I'm now investigating the normal stuff such as the preferred manner of recovering this on reboot, then I'll be able to feed the data into my little home automation system to keep an eye on things.

If you would like the .deb I generated with checkinstall up above, it should be [here](https://birdslikewires.net/download/raspberrypi/nut/). Sorry, I was keyboard hammering and didn't bother to put proper versioning on it or anything fancy like that.

Meatballs1, 2021-12-08:
The above worked steps also for my PowerCool 650VA which is listed as:

```
Bus 001 Device 008: ID 0001:0000 Fry's Electronics MEC0003
```

uipavec, 2022-02-09:
@Meatballs1 Was there anything else you had to do to make it work? I also have the same device as you:
```
Bus 003 Device 010: ID 0001:0000 Fry's Electronics MEC0003
```

...I have compiled the code & installed as per instructions but when I run: 
```
sudo /lib/nut/nutdrv_qx -a greencell -DDDDD
```

...I get:
```
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.7.4-4539-gefb9ee45)
USB communication driver (libusb 0.1) 0.42
   0.000000	[D1] debug level is '5'
   0.000829	[D5] send_to_all: SETINFO device.type "ups"
   0.000854	[D1] upsdrv_initups...
   0.000883	[D2] Language ID workaround enabled (using '0x409')
   0.415368	[D3] usb_busses=0x55c19f236150
   0.415394	[D2] Checking device (1D6B/0003) (004/001)
   0.415606	[D2] - VendorID: 1d6b
   0.415626	[D2] - ProductID: 0003
   0.415630	[D2] - Manufacturer: unknown
   0.415634	[D2] - Product: unknown
   0.415638	[D2] - Serial Number: unknown
   0.415642	[D2] - Bus: 004
   0.415646	[D2] - Device: 001
   0.415650	[D2] - Device release number: 0513
   0.415654	[D2] Trying to match device
   0.415658	[D3] match_function_regex: matching a device...
   0.415667	[D2] match_function_regex: failed match of VendorID: 1d6b
   0.415676	[D2] Device does not match - skipping
   0.415692	[D2] Checking device (0001/0000) (003/011)
   0.418422	[D2] - VendorID: 0001
   0.418442	[D2] - ProductID: 0000
   0.418446	[D2] - Manufacturer: MEC
   0.418450	[D2] - Product: MEC0003
   0.418454	[D2] - Serial Number: unknown
   0.418458	[D2] - Bus: 003
   0.418462	[D2] - Device: 011
   0.418466	[D2] - Device release number: 0100
   0.418470	[D2] Trying to match device
   0.418474	[D3] match_function_regex: matching a device...
   0.418525	[D2] Device matches
   0.418540	[D2] failed to claim USB device: could not claim interface 0: Device or resource busy
   0.418779	[D2] detached kernel driver from USB device...
   0.418800	[D3] nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   0.418814	[D5] send_to_all: SETINFO ups.vendorid "0001"
   0.418822	[D5] send_to_all: SETINFO ups.productid "0000"
   0.419317	[D1] First supported language ID: 0x9 (please report to the NUT maintainer!)
   0.419339	[D2] Skipping protocol Voltronic 0.06
   0.419355	[D2] Skipping protocol Voltronic-QS 0.07
   0.419365	[D2] Skipping protocol Voltronic-QS-Hex 0.10
   0.419376	[D2] Skipping protocol Mustek 0.07
   0.419385	[D2] Skipping protocol Megatec/old 0.07
   0.419394	[D2] Skipping protocol BestUPS 0.06
   0.419407	[D2] Skipping protocol Mecer 0.07
   0.419418	[D2] Skipping protocol Megatec 0.06
   0.419428	[D2] Skipping protocol Zinto 0.06
   0.419436	[D2] Skipping protocol Masterguard 0.01
   0.419450	[D3] send: Q1
   0.419464	[D4] command index: 0x03
   0.585272	[D4] fabula_command: the reply lacks the expected terminating CR.
   0.585288	[D4] fabula_command: adding missing terminating CR.
   0.585292	[D5] read: (11 bytes) => 55 50 53 20 4e 6f 20 41 63 6b 0d
   0.585294	[D3] read: UPS No Ack
   0.585298	[D2] qx_process_answer: short reply (input.voltage)
   0.585302	[D3] send: Q1
   0.585303	[D4] command index: 0x03
   0.751253	[D4] fabula_command: the reply lacks the expected terminating CR.
   0.751286	[D4] fabula_command: adding missing terminating CR.
   0.751296	[D5] read: (11 bytes) => 55 50 53 20 4e 6f 20 41 63 6b 0d
   0.751302	[D3] read: UPS No Ack
   0.751310	[D2] qx_process_answer: short reply (input.voltage)
   0.751317	[D3] send: Q1
   0.751322	[D4] command index: 0x03
   0.917393	[D4] fabula_command: the reply lacks the expected terminating CR.
   0.917425	[D4] fabula_command: adding missing terminating CR.
   0.917434	[D5] read: (11 bytes) => 55 50 53 20 4e 6f 20 41 63 6b 0d
   0.917441	[D3] read: UPS No Ack
   0.917450	[D2] qx_process_answer: short reply (input.voltage)
   0.917457	[D2] Skipping protocol Ablerex 0.01
   0.917463	[D2] Skipping protocol Q1 0.07
   0.917467	Device not supported!
   0.917471	Device not supported!
```
My UPS is a GreenCell, Emergency Power Supply for Line Interactive 600VA. The model: Micropower 600.

jimklimov, 2022-02-10:
@uipavec : Just in case, a couple of comments above there is a configuration snippet for the device `[ups]` with many custom fields that may be critical for device detection and further dialog; is your `[greencell]` set up similarly?

Other than that, "unfortunately" many USB interface chips are sold and used in unrelated devices without reflashing to registered vendor/product IDs for the thing they are sold in, or using "surprisingly random and unique" values like 0x0000, 0xffff, 0x0001 and such. Probably those vendors did not bother to go to USB-IF to pay for that. As a result, completely unrelated devices talking arbitrary protocols may be exposed with same PID/VID combinations. NUT has a few drivers with such "conflict of interest", I think "atcl" was among those.

jimklimov, 2022-02-10:
Looking at history, check if #1080 setup for a Greencell device helps you?

darkvaderzerotwo, 2022-04-11:
I managed to get it working for me following the comments above with the Hunnox protocol on my powercool 650VA unit that came with UPSmart v1.9. Thank you all for taking the time to document your efforts!
From what I can tell it would be nice would be to have a robust way of testing and profiling these devices by unloading and reloading nearly every option under the sun, starting with the least featureful (and most likely to output working data) then going through each subdriver and protocol until something works as expected. In my case it would have been useful to have a way that I could point a profiler at my ups, because even when the driver \worked\ it still output all values as zero and wouldn't respond to any change in AC input, surely there's a way that it could be made to detect that and flag it as suspect data then try a different protocol? I suppose that's a feature request for the future.
Anyway my config and output is below for reference. Thank you everyone!

Config:
```
[ups]
        driver = nutdrv_qx
        desc = "Powercool 650VA UPS"
        port = "auto"
        vendorid = 0001
        productid = 0000
        bus = 001
        novendor
        norating
        noscanlangid
        protocol=hunnox
        langid_fix = "0x0409"

```
**Output:**

> battery.voltage: 13.80
> device.type: ups
> driver.flag.norating: enabled
> driver.flag.noscanlangid: enabled
> driver.flag.novendor: enabled
> driver.name: nutdrv_qx
> driver.parameter.bus: 001
> driver.parameter.langid_fix: 0x0409
> driver.parameter.pollfreq: 30
> driver.parameter.pollinterval: 2
> driver.parameter.port: auto
> driver.parameter.productid: 0000
> driver.parameter.protocol: hunnox
> driver.parameter.synchronous: auto
> driver.parameter.vendorid: 0001
> driver.version: 2.8.0-rc2
> driver.version.data: Hunnox 0.01
> driver.version.internal: 0.32
> driver.version.usb: libusb-1.0.24 (API: 0x1000108)
> input.frequency: 49.9
> input.voltage: 240.0
> input.voltage.fault: 0.0
> output.voltage: 241.0
> ups.beeper.status: disabled
> ups.delay.shutdown: 60
> ups.delay.start: 0
> ups.load: 18
> ups.productid: 0000
> ups.status: OL
> ups.temperature: 29.0
> ups.type: online
> ups.vendorid: 0001
>

AliMickey, 2022-04-22:
@darkvaderzerotwo 
I tried your config with my Digitech Generic UPS from a local retailer. 

lsusb: `Bus 001 Device 005: ID 0001:0000 Fry's Electronics MEC0003`

ups.conf: ```
[ups]
        driver = nutdrv_qx
        desc = "Powercool 650VA UPS"
        port = "auto"
        vendorid = 0001
        productid = 0000
        bus = 001
        novendor
        norating
        noscanlangid
        protocol=hunnox
        langid_fix = "0x0409"
```

Running it causes this error: 
``` 
Starting up the UPS drivers ...
nut_1                  | Network UPS Tools - UPS driver controller 3.8.0-3220-g866b921892
nut_1                  | Network UPS Tools - Generic Q* USB/Serial driver 0.28 (3.8.0-3220-g866b921892)
nut_1                  | USB communication driver 0.33
nut_1                  |
nut_1                  | Fatal error: 'noscanlangid' is not a valid flag for this driver.
nut_1                  |
nut_1                  | Look in the man page or call this driver with -h for a list of
nut_1                  | valid variable names and flags.
nut_1                  | Starting up the UPS daemon ...
nut_1                  | Network UPS Tools upsd 3.8.0-3220-g866b921892
```

If i remove the `noscanlangid` variable then: 
```
Starting up the UPS drivers ...
nut_1                  | Network UPS Tools - UPS driver controller 3.8.0-3220-g866b921892
nut_1                  | Network UPS Tools - Generic Q* USB/Serial driver 0.28 (3.8.0-3220-g866b921892)
nut_1                  | USB communication driver 0.33
nut_1                  | No supported devices found. Please check your device availability with 'lsusb'
nut_1                  | and make sure you have an up-to-date version of NUT. If this does not help,
nut_1                  | try running the driver with at least 'subdriver', 'vendorid' and 'productid'
nut_1                  | options specified. Please refer to the man page for details about these options
nut_1                  | (man 8 nutdrv_qx).
nut_1                  |
nut_1                  | Driver failed to start (exit status=1)
nut_1                  | ERROR on driver startup.
```

Giving me headaches.

darkvaderzerotwo, 2022-04-22:
Hi @AliMickey,
I'm not surprised you have a headache! I took over 3 hours trying different options to get it to work correctly for me. At the minute I would say that the support for these chinese devices is really in it's infancy, it's good to see that NUT is documenting it though and trying their best to get it to work but yes, mileage will vary until, i imagine, 2.8 becomes commonplace with some of the newer drivers.

Looks like that particular version of the nutdrv_qx driver your using doesn't support the noscanlangid option, to be fair I didn't know that because i built 2.8-rc2 pretty much from the getgo. this is my output from when i reload the driver:

```
Network UPS Tools - UPS driver controller 2.8.0-rc2
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0-rc2)
USB communication driver (libusb 1.0) 0.43
```

What I would suggest is, if you feel comfortable, to try building RC2 from source. I can't really help you with building it though as i did the lazy option only following the standard, ./config , make and make install once you have Libusb-dev1.0 installed (i believe that's the package name on debian). 
It wasn't ideal on what I installed it on (Raspberry Pi 3 with raspbian lite) as it took a few extra steps to get it to load configs properly but has been pretty stable since.

I hope my input helps you down the path to get your device working for you!

jimklimov, 2022-04-24:
Note: for local builds, docs/config-prereqs.txt lists dependencies and
tools used by CI farm to build everything possible on a vanilla distro.

commit f69a7d61 (2022-04-25): driver.list.in: Add HCL info for CPC, ARES and Powercool models with hunnox protocol (USB)

Closes: #537
closed 2022-04-25 by jimklimov