networkupstools/nut #542: Data stale errors with Cyberpower AVRG750U

stevegilbert23, 2018-03-17:
Running on Debian Stretch with version 2.7.4 (2.7.4-5 on Debian)

About a month ago I upgraded from an older Cyberpower unit to a AVRG750U.  Within an hour of hooking it up, I started to see "Data stale" errors in the nut log.  

`[upsmon] daemon err:  Poll UPS [ups@localhost] failed - Data stale`

I did a lot of research and messing around with the polling time values but nothing helped.

I eventually ended up building nut locally and adding debugging statements to determine the root cause.  I found two problems:

1. There appears to be a bug in the libusb 0.1 "usb_interrupt_read" function.  It will occasionally hang forever, not obeying the timeout passed as the second last parameter.  This function is called from libusb.c around line 534.  This can happen in as little as 5 minutes to a few hours with the client querying the UPS every few seconds.
2. The timeout passed to the above noted call (250 ms) is too small for the AVRG750U and even when the function operates correctly, the timeout error code is often returned.

After discovering these problems, I started researching the reported issues here and came upon #300 which ultimately led me believe the issue may be specific to libusb 0.1.  I switched to the libusb0.1+1.0 branch.  I built it locally and found that the hanging issue in libusb seems to not exist with version 1.0.  I have been running for a couple of weeks and have experienced no hangs.

The timeout value used is still 250 ms and the timeout error code is often returned with the AVRG750U.  I have determined experimentally that a timeout value of 750 ensures the call never times out with the AVRG750U.  Based on my testing, I believe it is safe to increase this timeout value since the function returns when it has received the available data, before it reaches the timeout, under normal circumstances.

I will submit a pull request with the change from 250 to 750, hoping that it will be accepted.  If changing that value is not an acceptable approach, I would be happy to attempt a change that is acceptable, perhaps using a configuration value, or whatever is suggested.  I could continue to run my modified version locally but I'd prefer to not do that and use an unmodified nut if I can.

I will reference this issue in my pull request.

clepple, 2018-03-21:
@stevegilbert23 thanks for running this to ground.

@zykh, @jimklimov or @aquette : any objections to merging #543?

Although this could push the inner polling loop closer to 3 seconds (versus 2), we should still be well below the default staleness threshold (15 seconds) for other hardware.

stevegilbert23, 2018-04-11:
@clepple Can the pull request be accepted now?  Let me know if there is anything additional you need from me.  

Thanks.

lagweasel, 2018-08-29:
You guys are awesome. I've been having the same problem with CyberPower CP1500PFCLCD since I upgraded from Ubuntu Xenial ages ago. Recently had another stab at solving it, and tried building from source using "libusb-compat-1.0" branch and problem is now fixed. Looking through the git logs lead me here.

closed 2022-01-11 by jimklimov

stevegilbert23, 2022-01-11:
@jimklimov Does your closure indicate this change has been merged somewhere and available in a release?  I am still running from the version I compiled and would like to get back to a released version if I can.

jimklimov, 2022-01-12:
@stevegilbert23 : Thanks for bringing this up. Actually, a proposed fix (with commit comment that it closes this issue) was part of the larger changes in libusb-1.0+0.1 supporting branch (more context in issue #300) that was merged recently. So GitHub closed this one automatically - thanks for making me aware of this :)

Mainstreaming libusb-1.0 support was among milestones for NUT 2.7.5 release, alas there were too many years since 2.7.4 whose source most systems package, so hopefully it would also happen soon (some more PRs remain, to clear and merge).