networkupstools/nut #549: PyNUT "ListClients" method does not work

clepple, 2018-04-13:
There seem to be a few related bugs in the PyNUT `ListClients` method, as of module version 1.3.0 (NUT 2.7.4+). The symptom is `ERR INVALID-ARGUMENT`.

- [ ] The command is `LIST CLIENT `+ups (no "S" at the end of "CLIENT")
- [ ] The "BEGIN" response includes the UPS name

Not sure if there are others.

ansred, 2022-07-25:
In the meantime I figured out away to report NUT client by simply running nestat to check connected/established connections 

**For count:**
```
root@raspberrypi:/etc/nut# netstat -anp | grep :3493 | grep ESTABLISHED | wc -l
4
```
.
.
.
.

**For details:**
```
root@raspberrypi:/etc/nut# netstat -anp | grep :3493 | grep ESTABLISHED
tcp        0      0 127.0.0.1:3493          127.0.0.1:35496         ESTABLISHED 1896/upsd
tcp        0      0 127.0.0.1:35496         127.0.0.1:3493          ESTABLISHED 1910/upsmon
tcp        0      0 10.0.3.37:3493          10.0.12.201:46964       ESTABLISHED 1896/upsd
tcp        0      0 10.0.3.37:3493          10.0.3.16:60548         ESTABLISHED 1896/upsd          
```

jimklimov, 2022-07-27:
Note that in the example above you account localhost twice (as server and as client). Technically that grep expression could also match e.g. port 34935.

Posting and testing a PR for PyNUT, to query the data server for its perspective, could be helpful though :)

commit 8645321e (2022-08-15): scripts/python/module/PyNUT.py.in: fix ListClients()

* to call the right protocol command

Closes: #549

* handle the begin/end messages correctly to work at all

* make ListClients() usable with "ups==None"

* parse "CLIENT ups host" in correct order
closed 2022-08-15 by jimklimov